gphoto/libgphoto2 #266: 2.5.17 segfault

littlexiang, 2018-04-29:
I have encountered 100% segment fault while reading file from camera, which is not with 2.5.16.
however on normal PC it won't happen, this only occurs in a LinuxDeploy environment (both ubuntu 16.04).
I am pretty sure there is something change caused this between 2.5.16.1 ~ 2.5.17

`Fatal error: unexpected signal during runtime execution
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0xe98ef954]

runtime stack:
runtime.throw(0x579962, 0x2a)
	/data/golang/src/runtime/panic.go:616 +0x60
runtime.sigpanic()
	/data/golang/src/runtime/signal_unix.go:372 +0x22c

goroutine 77 [syscall]:
runtime.cgocall(0x3fed38, 0x1134fc2c, 0x578e5c)
	/data/golang/src/runtime/cgocall.go:128 +0x64 fp=0x1134fc0c sp=0x1134fbf4 pc=0x2caa0
github.com/littlexiang/gphoto2go._Cfunc_gp_camera_file_get(0xe37004f0, 0xe3745020, 0xe3745008, 0x1, 0xe3745048, 0xe3700470, 0x0)
	_cgo_gotypes.go:319 +0x38 fp=0x1134fc28 sp=0x1134fc0c pc=0x2b43e0
github.com/littlexiang/gphoto2go.(*Camera).FileReader.func4(0xe37004f0, 0xe3745020, 0xe3745008, 0x1, 0xe3745048, 0xe3700470, 0x1130c000)
	/data/go/src/github.com/littlexiang/gphoto2go/gphoto2go.go:349 +0xbc fp=0x1134fc48 sp=0x1134fc28 pc=0x2b8ce0
github.com/littlexiang/gphoto2go.(*Camera).FileReader(0x11454228, 0x11480b20, 0x1d, 0x1147ba50, 0xc, 0x0, 0x0)
	/data/go/src/github.com/littlexiang/gphoto2go/gphoto2go.go:349 +0x174 fp=0x1134fc78 sp=0x1134fc48 pc=0x2b6fb4
gocam.(*cameraWorker).saveImage(0x8170c0, 0x11480b20, 0x1d, 0x1147ba50, 0xc, 0x100, 0x0)`

littlexiang, 2018-05-01:
another difference is when camera disconnected in the middle of event listening
2.5.17 broke down while 2.5.16 not

msmeissn, 2018-05-13:
can you share the go app? 
can you also try 2.5.18, but it probably will make no difference

littlexiang, 2018-09-21:
works agin with 2.5.19
thanks a lot

closed 2018-09-21 by littlexiang