gphoto/libgphoto2 #269: Canon 750D USB port resets in the middle of application

sepehrhm, 2018-05-05:
Hi,

I have observed a faulty behaviour on two Canon EOS 750D (aka T6i) cameras. An application captures the preview (for streaming), and while the application is running, the USB port disconnects and reconnects.

After the reconnect, the bus/dev id of the camera changes, however gphoto2 cli can detect the camera. The following is my `dmesg -T` log, in which you can see the disconnect and several attempts of reconnecting:

> [Sat May  5 10:06:06 2018] usb 1-4: new high-speed USB device number 6 using xhci_hcd
[Sat May  5 10:06:06 2018] usb 1-4: New USB device found, idVendor=04a9, idProduct=32a1
[Sat May  5 10:06:06 2018] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[Sat May  5 10:06:06 2018] usb 1-4: Product: Canon Digital Camera
[Sat May  5 10:06:06 2018] usb 1-4: Manufacturer: Canon Inc.
[Sat May  5 10:42:01 2018] usb 1-4: Process 1767 (rover-capture) called USBDEVFS_CLEAR_HALT for active endpoint 0x02
[Sat May  5 10:42:04 2018] usb 1-4: reset high-speed USB device number 6 using xhci_hcd
[Sat May  5 10:53:56 2018] perf: interrupt took too long (2511 > 2500), lowering kernel.perf_event_max_sample_rate to 79500
[Sat May  5 11:30:41 2018] perf: interrupt took too long (3143 > 3138), lowering kernel.perf_event_max_sample_rate to 63500
[Sat May  5 11:34:24 2018] usb 1-4: USB disconnect, device number 6
[Sat May  5 11:34:26 2018] usb 1-4: new high-speed USB device number 7 using xhci_hcd
[Sat May  5 11:34:26 2018] usb 1-4: New USB device found, idVendor=04a9, idProduct=32a1
[Sat May  5 11:34:26 2018] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[Sat May  5 11:34:26 2018] usb 1-4: Product: Canon Digital Camera
[Sat May  5 11:34:26 2018] usb 1-4: Manufacturer: Canon Inc.
[Sat May  5 11:34:30 2018] usb 1-4: USB disconnect, device number 7
[Sat May  5 11:34:32 2018] usb 1-4: new high-speed USB device number 8 using xhci_hcd
[Sat May  5 11:34:32 2018] usb 1-4: New USB device found, idVendor=04a9, idProduct=32a1
[Sat May  5 11:34:32 2018] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[Sat May  5 11:34:32 2018] usb 1-4: Product: Canon Digital Camera
[Sat May  5 11:34:32 2018] usb 1-4: Manufacturer: Canon Inc.
[Sat May  5 11:34:35 2018] usb 1-4: USB disconnect, device number 8
[Sat May  5 11:34:38 2018] usb 1-4: new high-speed USB device number 9 using xhci_hcd
[Sat May  5 11:34:38 2018] usb 1-4: New USB device found, idVendor=04a9, idProduct=32a1
[Sat May  5 11:34:38 2018] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[Sat May  5 11:34:38 2018] usb 1-4: Product: Canon Digital Camera
[Sat May  5 11:34:38 2018] usb 1-4: Manufacturer: Canon Inc.

I am not sure if this is a libgphoto2 issue, but I wanted to know if there is a limitation in send/receive operations on a camera port.

msmeissn, 2018-05-13:
there is no limitation on these operations. weird that it crashes. ... we might miss to download images or so

samidalati, 2019-01-02:
Having this issue as well with T7i, usb port disconnect and I would have to restart the camera to get connected again. Please advise.

msmeissn, 2019-01-04:
what commands are you running? after which time does it happen?
i see the first reporter it worked for like 1.5 hours?

msmeissn, 2019-01-04:
also please specify which libgphoto2 versions are in use. there were some older ones where the "keep device on" logic was faulty

sepehrhm, 2019-01-04:
I basically have a tethered shooting application. I grab previews with `gp_camera_capture_preview()` and display them, and then I can trigger a full shot with `gp_camera_capture()` function.

Here is the libgphoto2 I am using:
`libgphoto2      2.5.18         all camlibs, gcc, ltdl, EXIF`

Do I need to update to newer versions?