gphoto/libgphoto2 #270: Having trouble getting shutter count for Canon T6i 750D and T3i

Rosscao, 2018-05-07:
Thanks for the product and I'm new to use it, I have searched thought the similar posts but still don't have a clue how to deal with this problem. 

-------------------------------------------------------------

~$ gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Canon EOS 750D                 usb:001,006     
~$ gphoto2 --get-config /main/status/shuttercount
                                                                               
*** Error ***              
/main not found in configuration tree.
*** Error (-1: 'Unspecified error') ***       

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --get-config /main/status/shuttercount

Please make sure there is sufficient quoting around the arguments.
------------------------------------------------------------------------------------------------------

Please advise. Thank you.

msmeissn, 2018-05-08:
gphoto2 --get-config shuttercount

should work as well,.

which libgphoto2 version?

The shuttercount is only exported by some Canon EOS cameras. My own 750D does not export it.

Rosscao, 2018-05-09:
Thanks Marcus, I've tried this but it still does not work, the version is
the newest 2.5.17

Maybe 750D cannot export shutter count? I also tried it on T3i but got the
same result...

Please see below:

~$ gphoto2 --auto-detect
Model                          Port

----------------------------------------------------------
Canon EOS 750D                 usb:001,006
~$ gphoto2 --get-config shuttercount

*** Error ***
An error occurred in the io-library ('Could not claim the USB device'):
Could not claim interface 0 (Device or resource busy). Make sure no other
program (gvfs-gphoto2-volume-monitor) or kernel module (such as sdc2xx,
stv680, spca50x) is using the device and you have read/write access to the
device.

*** Error ***
An error occurred in the io-library ('Could not claim the USB device'):
Could not claim interface 0 (Device or resource busy). Make sure no other
program (gvfs-gphoto2-volume-monitor) or kernel module (such as sdc2xx,
stv680, spca50x) is using the device and you have read/write access to the
device.
*** Error (-53: 'Could not claim the USB device') ***

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --get-config
shuttercount

Please make sure there is sufficient quoting around the arguments.

~$ gphoto2 --version
gphoto2 2.5.17

Copyright (c) 2000-2018 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named
COPYING.

This version of gphoto2 is using the following software versions and
options:
gphoto2         2.5.17         gcc, popt(m), exif, no cdk, no aa, no jpeg,
no readline
libgphoto2      2.5.17         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect
usbscsi, gcc, ltdl, USB, serial without locking

ccrtnsp, 2018-05-09:
I might be wrong, but seem to recall reading that only the top-end Canons (1D variants basically) record the shutter count to EXIF data. It could be that the same applies to exporting it other ways. At least my 7Dmk2 does not export it in any way.