gphoto/libgphoto2 #276: Sony A7S not providing list of shutterspeed

astronono, 2018-05-22:
Hello,
I've been working with Jasem Mutlaq to try to solve an issue with indi sony ccd driver not working. In fact it looks like the shutterspeed exhaustive list is provided by libgphoto2 and this seems not working for the Sony A7S.

The discussion we had on this matter is there.
http://indilib.org/forum/ccds-dslrs/2964-indi-and-sony-a7s.html?start=24#26182

Jasem told me to file a bug into the project to see if that could help.

Best regards,
Bruno

msmeissn, 2018-05-23:
which libgphoto2 version is used?

it might be in a mode where shutterspeed setting is not possible, like manual .. hard to see from the logfile posted.

perhaps get
gphoto2 --list-all-config

astronono, 2018-05-23:
its true, A7S is in manual mode by habit, which mode should be used ? I shall provide the list-all-config tonight

msmeissn, 2018-05-23:
P or Tv would allow setting the shutterspeed I think.

astronono, 2018-05-23:
Will try, thanks.

knro, 2018-06-10:
```
/main/capturesettings/shutterspeed
Label: Temps de pose
Readonly: 0
Type: RADIO
Current: 300/10
END
```

This is the problem here. For Canon, you get something like this:

```
/main/capturesettings/shutterspeed
Label: Temps de pose
Readonly: 0
Type: RADIO
Current: automatique
Choice: 0 30
Choice: 1 25
Choice: 2 20
Choice: 3 15
Choice: 4 13
Choice: 5 10
Choice: 6 8
...etc
```

We have several users who are INDI gphoto driver (based on libgphoto2 library from master) reporting the same problem.

knro, 2018-06-25:
Any hints on what's going on here? Is there anyway I can help?

elijahparker, 2018-06-25:
This is normal for Sony -- in an embedded driver I wrote, I was able to get
the lists using the opcode 0x9209

Here's my implementation:
https://github.com/timelapseplus/TimelapsePlus-Firmware/blob/master/src/PTP_Codes.h#L90
https://github.com/timelapseplus/TimelapsePlus-Firmware/blob/master/src/PTP.cpp#L2555

knro, 2018-06-25:
Right thanks! But since libgphoto2 is supposed to _abstract_ these low-level details for us, I expect to use the same widget `shutterspeed` across different cameras without worrying on PTP stuff.

amuro1977, 2018-08-04:
Maybe this can help to find issue:

with Ubuntu 16.04 LTS;         Gphoto 2.5.15-2;              Libgphoto 2.5.16-2
Sony worked fine under INDI-EKOS..

Now i upgrade to Kubuntu 18.04LTS with last Gphoto and Libgphoto and Sony A7s stop to work as reported before

 Some problems appeared after recently release

knro, 2018-08-06:
I'm going to provide a workaround for this in INDI, but the long term solution is for this to be fixed in libgphoto2 of course where the RADIO control is presented instead of TEXT.

japz, 2018-12-18:
I'm running into the same issue with my Sony A7r iii, when trying to get it to work with Indigo. @knro, did you ever merge a workaround in INDI?

I looked at the source code of the ptp2 driver, but wouldn't know where to begin addressing this. Are the possible options supposed to be read from the camera, or do the txt files such as `./camlibs/ptp2/cameras/sony-a7r3.txt` have a function besides acting as reference material?

msmeissn, 2018-12-24:
I have added a fixed list of shutterspeeds to libgphoto2 git now, commit  daacc5c2f27b76943771e63f6e9dfab555ea1c97