gphoto/libgphoto2 #277: Camera Canon 50d disconect when downloading files

bordac, 2018-05-28:
Hello friends!
I tried 3 different cameras Canon 50D, result is the same.
After power on, camera detected by gphoto2:
 
_**root@raspberrypi:/camsoft# gphoto2 --auto-detect
Model                          Port
----------------------------------------------------------
Canon EOS 50D                  usb:001,047**_
 
then, when do command "gphoto2 --list-files"
output is:
 
**_root@raspberrypi:/camsoft# gphoto2 --list-files
There is no file in folder '/'.
There is no file in folder '/store_00010001'.
 
*** Error ***
PTP I/O error
*** Error (-1: 'Unspecified error') ***_**

[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/2045053/my-logfile.txt)

msmeissn, 2018-05-29:
I also emailed, please upgrade to a newer ligphoto2 and report back if that helped.

bordac, 2018-05-29:
Hello. It's helped for me. Thank You very mach.

closed 2018-05-29 by bordac