gphoto/libgphoto2 #284: Canon EOS 1300D First Image didn´t capture after init

Raggygandalf, 2018-06-22:
Hi gphoto2 Team,

i have a problem with the gphoto2 lib in my c program.

When i call 

'gp_camera_capture()' directly after the 'gp_camera_init()' it captures the picture.

But when I wait about 5 seconds or more after the init,  the first capture returns an errorcode  with 'I/O in progress'.
After that I can call capture again and there are no problem anymore.

msmeissn, 2018-09-08:
which version of libgphoto2 is in use? I did some fixes in newer versions

Raggygandalf, 2018-10-04:
Hi. sorry for the late Response.
The version was 2.5.18. I try to update it!

msmeissn, 2018-10-06:
it is quite new already. hmm

Raggygandalf, 2018-11-13:
Oh sorry. I use vs2017 over remote to compile and i have used the old Version. My fail, sorry!

msmeissn, 2018-11-13:
thx for checking :)

closed 2018-11-13 by msmeissn