gphoto/libgphoto2 #288: add Sony a700

moorjunge, 2018-07-07:
Hello,

today i tried to use my old a700 with gphoto.

In the menu i set USB connection to "Remote PC".
If i put the put in the USB Plug in the camera, on the display the message "Initializing USB Connection" apperas.

```
pi@raspberrypi:~ $  gphoto2 --auto-detect
Model                          Port
----------------------------------------------------------
Sony DSC-A900 (PTP mode)       usb:001,013
```

So it detected a wrong camera model.

After a gphoto2 --config the camera changed to -PC- mode and the terminal window freezes.
Open a new window and run the command gphoto2 --capture-image
i get this

```
pi@raspberrypi:~ $ gphoto2 --capture-image

*** Error ***
An error occurred in the io-library ('Could not claim the USB device'): Could not claim interface 0 (Device or resource busy). Make sure no other program (gvfs-gphoto2-volume-monitor) or kernel module (such as sdc2xx, stv680, spca50x) is using the device and you have read/write access to the device.
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-53: 'Could not claim the USB device') ***

For debugging messages, please use the --debug option.
Debugging messages may help finding a solution to your problem.
If you intend to send any error or debug messages to the gphoto
developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
gphoto2 as follows:

    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --capture-image

Please make sure there is sufficient quoting around the arguments.

```

So, is it possible to get the camera to work with the programm?

Camera in -PC- Mode
```
pi@raspberrypi:~ $ gphoto2 --debug
0.003954 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.005945 main                        (2): gphoto2 2.5.11
0.007882 main                        (2): gphoto2 has been compiled with the following options:
0.009713 main                        (2):  + gcc (C compiler used)
0.009926 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.010072 main                        (2):  + exif (for displaying EXIF information)
0.010219 main                        (2):  + cdk (for accessing configuration options)
0.010361 main                        (2):  + aa (for displaying live previews)
0.010502 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.010641 main                        (2):  + readline (for easy navigation in the shell)
0.011930 main                        (2): libgphoto2 2.5.12
0.012160 main                        (2): libgphoto2 has been compiled with the following options:
0.012321 main                        (2):  + all camlibs
0.012462 main                        (2):  + gcc (C compiler used)
0.012605 main                        (2):  + ltdl (for portable loading of camlibs)
0.012750 main                        (2):  + EXIF (for special handling of EXIF files)
0.012895 main                        (2): libgphoto2_port 0.12.0
0.013076 main                        (2): libgphoto2_port has been compiled with the following options:
0.014287 main                        (2):  + gcc (C compiler used)
0.014511 main                        (2):  + ltdl (for portable loading of camlibs)
0.014664 main                        (2):  + USB (libusb1, for USB cameras)
0.014813 main                        (2):  + serial (for serial cameras)
0.014958 main                        (2):  + no resmgr (serial port access and locking)
0.015101 main                        (2):  + no ttylock (serial port locking)
0.015248 main                        (2):  + no lockdev (serial port locking)
0.015393 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.015542 main                        (2): IOLIBS env var not set, using compile-time default instead
0.015687 main                        (2): invoked with following arguments:
0.015834 main                        (2):   --debug
0.017716 load_settings               (2): Creating gphoto config directory ('/home/pi/.gphoto')
0.033115 load_settings               (2): Loading settings from file '/home/pi/.gphoto/settings'.
0.034486 gp_camera_free              (2): Freeing camera...
0.034736 gp_port_free                (2): Freeing port...
0.034892 gp_filesystem_reset         (2): resetting filesystem
0.035042 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
0.035186 gp_filesystem_lru_clear     (2): fscache LRU list already empty
0.035338 delete_all_folders          (2): Internally deleting all folders from '/'...
0.036589 lookup_folder               (2): Lookup folder '/'...
0.036770 lookup_folder               (2): Found! / is 0xa38b20
0.036921 recurse_delete_folder       (2): Recurse delete folder 0xa38b20//
```

Greetings
Thiemo

moorjunge, 2018-07-07:
Hello,

i tried a gphoto2 --summary and it gives the right camera type

```
pi@raspberrypi:~ $ gphoto2 --summary
Camera summary:
Manufacturer: Sony Corporation
Model: DSLR-A700
  Version: 01.0000
Vendor Extension ID: 0x11 (1.0)
Vendor Extension Description: Sony PTP Extensions

Capture Formats:
Display Formats: JPEG, Unknown(b101)

Device Capabilities:
        File Download, No File Deletion, No File Upload
        No Image Capture, No Open Capture, Sony Capture

Storage Devices Summary:
store_00010001:
        StorageDescription: None
        VolumeLabel: None
        Storage Type: Removable RAM (memory card)
        Filesystemtype: Generic Hierarchical
        Access Capability: Read-Write
        Maximum Capability: 8017018880 (7645 MB)
        Free Space (Bytes): 3090579456 (2947 MB)
        Free Space (Images): -1

Device Property Summary:
Compression Setting(0x5004):(readwrite) (type=0x2) JPEG Fine (2)
White Balance(0x5005):(readwrite) (type=0x4) 512
F-Number(0x5007):(readwrite) (type=0x4) f/5.1 (512)
Focus Mode(0x500a):(readwrite) (type=0x4) 512
Exposure Metering Mode(0x500b):(readwrite) (type=0x4) 512
Flash Mode(0x500c):(readwrite) (type=0x4) 512
Exposure Program Mode(0x500e):(readwrite) (type=0x4) 512
Exposure Bias Compensation(0x5010):(readwrite) (type=0x3) 0.5 stops (512)
Still Capture Mode(0x5013):(readwrite) (type=0x4) 512
DOC Compensation(0xd200):(readwrite) (type=0x3) 512
DRangeOptimize(0xd201):(readwrite) (type=0x2) 2
Image size(0xd203):(readwrite) (type=0x2) 2
Shutter speed(0xd20d):(readwrite) (type=0x6) 50331647
Property 0xd20e:(readwrite) (type=0x2) 2
Color temperature(0xd20f):(readwrite) (type=0x4) 277
CC Filter(0xd210):(readwrite) (type=0x2) 1
Aspect Ratio(0xd211):(readwrite) (type=0x2) Enumeration [2] value: 2
Property 0xd212:(readwrite) (type=0x2) 1
Focus status(0xd213):(readwrite) (type=0x2) 2
Property 0xd214:(readwrite) (type=0x6) 16777216
Objects in memory(0xd215):(readwrite) (type=0x4) 256
Expose Index(0xd216):(readwrite) (type=0x4) 767
Property 0xd217:(readwrite) (type=0x2) Enumeration [2] value: 2
Battery Level(0xd218):(readwrite) (type=0x1) 1
Property 0xd2c1:(readwrite) (type=0x4) Enumeration [512] value: 512
Property 0xd2c2:(readwrite) (type=0x4) Enumeration [512] value: 512
Property 0xd2c3:(readwrite) (type=0x4) Enumeration [512] value: 512
Property 0xd2c4:(readwrite) (type=0x4) Enumeration [512] value: 512
Property 0xd2c5:(readwrite) (type=0x4) Enumeration [512] value: 512
Property 0xd2c6:(readwrite) (type=0x4) Enumeration [512] value: 512
Still Image(0xd2c7):(readwrite) (type=0x4) Enumeration [512] value: 512

```

then  a gphoto2 --capture-image is running and the camera makes one shoot, but

```
pi@raspberrypi:~ $ gphoto2 --capture-image
ERROR: Could not capture image.
ERROR: Could not capture.

```

and after that the camera lost the connection.
"-PC-" is blinking yellow

msmeissn, 2018-07-08:
The misdetection is just due to the fact that multiple cameras might share the same ID, or that someone might have misreported the name.
Having a correct name is not that important in this place.

The first error message ('Could not claim the USB device') is due to GVFS (gnome virtual filesystem) blocking the camera.

The other --summary traces show it support capture just fine.

Can you first upgrade to libgphoto2 2.5.18, as this contains more Sony bugfixes.
If this does not fix --capture-image, can you attach a debuglogfile 

gphoto2 --capture-image --debug --debug-logfile=capture.log

moorjunge, 2018-07-08:
Camera was in a wrong mode, i'll check it later again.

msmeissn, 2019-02-10:
it crashes with the 9209 command, which we assumed to be present... 
this reduces functionality, as the sony capture logic currently relies on this a bit
It might work better with current git (to be in 2.5.22), but I am not sure.