gphoto/libgphoto2 #289: Add support for Fujifilm X-H1

Firstyear, 2018-07-08:
It would be great to have support for the Fujifilm X-H1. This is (hopefully) similar to the X-T2. Thanks for your time and effort!

Firstyear, 2018-07-08:
[summary.log](https://github.com/gphoto/libgphoto2/files/2173247/summary.log)
[summary.txt](https://github.com/gphoto/libgphoto2/files/2173248/summary.txt)

Firstyear, 2018-07-08:
As a note, I can't see the listing of live view support however Fuji's manuals and software supports live view for the X-H1, X-T2 and the X-Pro2.

msmeissn, 2018-07-08:
we have the X-h1 listed as capture capable in current git, so --capture-image will just work probably already.

we do not know how to liveview capture. it likely happens via one ofthe 3 vendor opcodes though

elijahparker, 2018-07-08:
Here's the code for Fuji liveview capture:
https://github.com/timelapseplus/VIEW/blob/master/lib/libgphoto2_ptp2_updates/library.c#L2981-L3042

Basically, the camera has to first be in open image capture mode, then the
preview object is accessed as a regular object at 0x80000001

This works with the X-H1 as well (which appears to be fully identical to
the X-T2).  It also works with the X-T1.

Firstyear, 2018-07-09:
That's really cool @elijahparker. What would it take to get similar support into libgphoto2? It would be amazing to have live view and capture in darktable as a result. Thank you.

darix, 2018-07-09:
@elijahparker are there any fixes in your copy of your gphoto which you could submit back?

elijahparker, 2018-07-09:
It should be possible to copy and paste that exact code into the current
libgphoto2.  I have a lot of work yet to reconcile my copy with the
official repo, but plan to work on that as soon as I have a release
completed for my application.  The referenced copy of library.c is from
libgphoto2 and should mostly work except there's a conflict with the way
I'm handling the Sony capture (so don't copy that function).  Next month I
should be able to merge all my updates into the official libgphoto2 repo.

msmeissn, 2018-07-09:
I have actually started merging back things from your tree to libgphoto2 git.

fuji liveview is in at least, the rest partially.

I need some testing at least.

Firstyear, 2018-07-09:
Hey @msmeissn,

If you can provide me some instructions on how to test, I might be able to help out with this. 

Thanks,

msmeissn, 2018-07-10:
easiest way to build is with https://github.com/gonzalo/gphoto2-updater  
use the provided script with -d option (development)

msmeissn, 2018-07-11:
On X-T2 and X-Pro2 I have now the liveview mode running. It is still a bit buggy when mixed with regular capture.

elijahparker, 2018-07-11:
Same here, at the high level in my app I'm disabling the open capture
before triggering a photo since it fails otherwise. Same goes for changes
settings. I suppose this should be implemented within the
camera_fuji_capture function. I'll review what I'm doing to disable it and
let you know.

msmeissn, 2018-07-11:
I think also the regular capture needs some more poking , preview after a regular capture returns PTP_RC_Busy

elijahparker, 2018-07-11:
It's working for me on an X-T2.  I can alternately switch between capture
and preview, but just not both at once.  However, after stopping liveview,
it needs a couple seconds before capture, otherwise the capture fails.

To disable open capture after running liveview (so that other functions
work again), it's ptp_terminateopencapture, which is already supported
through gphoto2 --set-config movie=0

msmeissn, 2018-07-13:
So the state in libgphoto2 GIT is quite good at this time.

i found:
- the image captured needs to be downloaded so capture preview can start ... the capture seems to "RAM". I added that.
- I also added open capture termination / reinitialization on capture / camera exit 
- Added support for dual RAW+JPG capture

Only thing not working very reliably is dual capture with RAW+JPG in combination with preview.

Firstyear, 2018-07-13:
Thanks @msmeissn. @darix has said he'll get me a build of this so that I can test it and report back with the X-H1.

msmeissn, 2020-09-27:
i think solved... state is improving

closed 2020-09-27 by msmeissn