gphoto/libgphoto2 #292: gp_camera_trigger_capture not returning immediatly

doisyg, 2018-07-30:
Hello,
I am using gphoto2 with a Sony alpha 6300. I am interested in taking pictures as fast as possible and as close as possible to the time my program sends the capture command (i can take care of downloading the pictures later).
I tried to use `gp_camera_trigger_capture` but this function takes about 1.2 second to return. So it means that I cannot take pictures faster than ~1.2 times per seconds (and of course gp_camera_wait_for_event would have to be added in top of that in order not to crash the camera if no ready).
My camera was set to full manual (exposure and focus).
Any clue how to go faster?

msmeissn, 2018-08-05:
hmm, what we are doing is that we do half-press => wait for focusing event => full press

we have no manual focusing detevction, so we might run into the 1 second assigned timeout there.

doisyg, 2018-08-05:
Hello. Thanks for your answer. I am having more or less the same result with the focus set to auto. 
Would that be possible to send a trigger event independently from the focus?