gphoto/libgphoto2 #293: Canon Regression

ryanstout, 2018-07-31:
First, thanks so much to Marcus and everyone who helped develop gphoto2.  Looks like there is a regression after 2.5.18, on Canon's when in aperture priority, getting "iso" results in a value of "Unknown Value 0004" and "aperture" has a value of "Unknown 0001"   So far I've seen this on most canon's I've tried it on.  (80d, t6i, 7dii)  Worked fine on the 5diii.  Possibly an issue with the OLC stuff?  I'm not sure what got changed there.

Thanks!

msmeissn, 2018-07-31:
I changed OLC handling in 2.5.19... is this 2.5.18 exact or what version?

ryanstout, 2018-07-31:
I tested on the 5dsr, btw.  Which has the issue.

msmeissn, 2018-07-31:
can you capture a debug trace ?

gphoto2 --debug --debug-logfile=list.log --list-config

ryanstout, 2018-07-31:
No problem. Here's the list.log on the 5dsr: https://gist.github.com/ryanstout/e0a68e2d29388d8ac05de29ea913faa0

msmeissn, 2018-08-05:
I commtted a patch to GIT... can you test it?

ryanstout, 2018-08-06:
Yep, that fixed it on the 7d ii, 80d, and 5dsr that I tested on.  Thanks a bunch!

closed 2018-08-06 by ryanstout

huhelicopter, 2020-11-21:
@msmeissn could you pls to try to see my email? huzhangli@gmail.com