gphoto/libgphoto2 #294: gtkdoc-mktmpl not part of gtkdoc since

mMerlin, 2018-08-04:
I am having a issue with "Could not claim the USB device" for a couple of different Canon EOS cameras.
```
gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Canon EOS 1200D                usb:002,011     
Canon EOS 700D                 usb:001,004     
```
Researching says that the latest version of libgphoto2 might fix the problem.  Trying to build that from github sources on a Fedora 27 box:
```
4.17.7-100.fc27.x86_64 #1 SMP Tue Jul 17 16:29:47 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

I tried building it Using
```bash
autoreconf --install --symlink
./configure --prefix=/usr/local --enable-docs
make
```
figuring to do an *everything* build.  That fails with
```
Making all in doc
make[4]: Entering directory '…/libgphoto2/libgphoto2_port/doc'
cd . && gtkdoc-scan --module=gphoto2-port --source-dir=../libgphoto2_port
cd . && gtkdoc-mktmpl --module=gphoto2-port
/bin/sh: gtkdoc-mktmpl: command not found
make[4]: *** [Makefile:540: templates] Error 127
```
Research that found https://bugs.archlinux.org/task/55210 then https://github.com/GNOME/gtk-doc/commit/46df4354abed5724697fd5e39630c5bbc6637cc4

It seems that recent gtkdoc installations do not include gtkdoc-mktmpl any more.  Removing the `templates` references from doc/Makefile got past that problem.
```bash
# all-local: templates sgml html
all-local: sgml html

scan:
	cd $(srcdir) && gtkdoc-scan --module=$(DOC_MODULE) --source-dir=$(DOC_SOURCE_DIR)

# templates: scan
# 	# cd $(srcdir) && gtkdoc-mktmpl --module=$(DOC_MODULE)

# sgml: templates
sgml:
	cd $(srcdir) && gtkdoc-mkdb --module=$(DOC_MODULE) --source-dir=$(DOC_SOURCE_DIR)
```

Now get the following, so the trivial fix was not enough.
```
if ! test -d html ; then mkdir html ; fi
cd html && gtkdoc-mkhtml gphoto2-port .././gphoto2-port-api.sgml
.././gphoto2-port-api.sgml:1: parser error : SystemLiteral " or ' expected
<!DOCTYPE BOOK PUBLIC "-//Davenport//DTD DocBook V3.0//EN" [
                                                           ^
.././gphoto2-port-api.sgml:1: parser error : SYSTEM or PUBLIC, the URI is missing
<!DOCTYPE BOOK PUBLIC "-//Davenport//DTD DocBook V3.0//EN" [
                                                           ^
.././gphoto2-port-api.sgml:2: parser error : internal error: xmlParseInternalSubset: error detected in Markup declaration

<!entity hash "#">
^
.././gphoto2-port-api.sgml:2: parser error : DOCTYPE improperly terminated
<!entity hash "#">
^
.././gphoto2-port-api.sgml:2: parser error : StartTag: invalid element name
<!entity hash "#">
 ^
.././gphoto2-port-api.sgml:2: parser error : Extra content at the end of the document
<!entity hash "#">
 ^
unable to parse .././gphoto2-port-api.sgml
```

For my *real* purposes, I can just remove the "--enable-docs" option.  But this should be fixed sometime.

msmeissn, 2018-08-27:
can confirm. migrating to the new gtkdoc build magic is a bit of work :/

W-Hamra, 2018-09-01:
Just came across this same issue. For most packages, I can simple flip a switch in their configure.ac then autoreconf will magically create a compatible gtk-doc build, but in the case of libgphoto2, the build process is hardcoded in DocRules.am :/

For reference, this is what I usually do in other packages: https://developer.gnome.org/gtk-doc-manual/stable/settingup_autoconf.html.en

msmeissn, 2018-09-01:
I tried around with this already, need to find a bit more patience to fiddle this in :/