gphoto/libgphoto2 #297: Canon EOS cameras Resource busy, Could not claim interface 0

mMerlin, 2018-08-07:
On multiple platforms, multiple versions,
failed: Resource busy (-6);
Error (-53: 'Could not claim the USB device')

```
gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Canon EOS 1200D                usb:002,011     
Canon EOS 700D                 usb:001,004     
```

My environments:
Raspberry Pi B+
4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l GNU/Linux
gphoto2/stable,now 2.5.11-1 armhf [installed]
libgphoto2-6/stable,now 2.5.12-1 armhf [installed,automatic]

'env LANG=C gphoto2 --debug --debug-logfile=dbg1.txt --port=usb:001,006 --get-all-files`

```
0.000573 main                        (2): gphoto2 2.5.11
0.000614 main                        (2): gphoto2 has been compiled with the following options:
0.000644 main                        (2):  + gcc (C compiler used)
0.000672 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000699 main                        (2):  + exif (for displaying EXIF information)
0.000725 main                        (2):  + cdk (for accessing configuration options)
0.000752 main                        (2):  + aa (for displaying live previews)
0.000779 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000805 main                        (2):  + readline (for easy navigation in the shell)
0.000849 main                        (2): libgphoto2 2.5.12
0.000890 main                        (2): libgphoto2 has been compiled with the following options:
0.000917 main                        (2):  + all camlibs
0.000942 main                        (2):  + gcc (C compiler used)
0.000968 main                        (2):  + ltdl (for portable loading of camlibs)
0.000993 main                        (2):  + EXIF (for special handling of EXIF files)
0.001022 main                        (2): libgphoto2_port 0.12.0
0.001057 main                        (2): libgphoto2_port has been compiled with the following options:
0.001084 main                        (2):  + gcc (C compiler used)
0.001110 main                        (2):  + ltdl (for portable loading of camlibs)
0.001138 main                        (2):  + USB (libusb1, for USB cameras)
0.001164 main                        (2):  + serial (for serial cameras)
0.001190 main                        (2):  + no resmgr (serial port access and locking)
0.001216 main                        (2):  + no ttylock (serial port locking)
0.001241 main                        (2):  + no lockdev (serial port locking)
0.001271 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.001297 main                        (2): IOLIBS env var not set, using compile-time default instead
0.001322 main                        (2): invoked with following arguments:
0.001348 main                        (2):   --debug
0.001374 main                        (2):   --debug-logfile=dbg1.txt
0.001400 main                        (2):   --port=usb:001,006
0.001424 main                        (2):   --get-all-files
…
0.191466 gp_camera_init              (2): Loading '/usr/lib/arm-linux-gnueabihf/libgphoto2/2.5.12/ptp2'...
0.195066 gp_port_open                (2): Opening USB port...
0.195130 gp_libusb1_open             (2): ()
0.195218 gp_libusb1_open             (2): claiming interface 0
0.195277 gp_libusb1_open [libusb1.c:421](0): 'libusb_claim_interface (port->pl->dh, port->settings.usb.interface)' failed: Resource busy (-6)
0.195317 gp_port_set_error [gphoto2-port.c:1190](0): Could not claim interface 0 (Device or resource busy). Make sure no other program (gvfs-gphoto2-volume-monitor) or kernel module (such as sdc2xx, stv680, spca50x) is using the device and you have read/write access to the device.
0.195851 gp_context_error            (0): An error occurred in the io-library ('Could not claim the USB device'): Could not claim interface 0 (Device or resource busy). Make sure no other program (gvfs-gphoto2-volume-monitor) or kernel module (such as sdc2xx, stv680, spca50x) is using the device and you have read/write access to the device.
0.197031 gp_camera_free              (2): Freeing camera...
```

Fedora 27
4.17.7-100.fc27.x86_64 #1 SMP Tue Jul 17 16:29:47 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
gphoto2 2.5.15;
libgphoto2 2.5.16

```
0.000033 main                        (2): gphoto2 2.5.15
[... 76 lines cut ...]
0.062302 gp_camera_init              (2): Loading '/usr/lib64/libgphoto2/2.5.19/ptp2'...
0.062907 gp_port_open                (2): Opening USB port...
0.062929 gp_libusb1_open             (2): ()
0.062960 gp_libusb1_open             (2): claiming interface 0
0.062973 gp_libusb1_open [libusb1.c:424](0): 'libusb_claim_interface (port->pl->dh, port->settings.usb.interface)' failed: Resource busy (-6)
0.062983 gp_port_set_error [gphoto2-port.c:1190](0): Could not claim interface 0 (Device or resource busy). Make sure no other program (gvfs-gphoto2-volume-monitor) or kernel module (such as sdc2xx, stv680, spca50x) is using the device and you have read/write access to the device.
0.063064 gp_context_error            (0): An error occurred in the io-library ('Could not claim the USB device'): Could not claim interface 0 (Device or resource busy). Make sure no other program (gvfs-gphoto2-volume-monitor) or kernel module (such as sdc2xx, stv680, spca50x) is using the device and you have read/write access to the device.
0.063505 gp_camera_free              (2): Freeing camera...
…
```

I only included log extracts here because I do not think the gphoto2 or libgphoto2 code is directly the problem.  After finding https://github.com/gphoto/gphoto2/issues/154 I was able to get gphoto2 to work on fedora by killing the process running `/usr/libexec/gvfs-gphoto2-volume-monitor`.  Found that using `ps ax | grep gphoto2`

On Raspbian Stretch, the process to kill was running `/usr/lib/gvfs-gphoto2-volume-monitor`.  In both cases, the process needed to be killed before accessing the cameras.  If the cameras were accessed, and the 'could not claim interface message' was seen, then killing the process did not get things to work until the cameras were power cycled.

As a test on Raspbian Stretch, I uninstalled the gvfs package `sudo apt remove gvfs`.  After a reboot, the volume monitor was gone, and gphoto2 worked as expected.

Moving the testing over to Raspbian Stretch lite, and gphoto2 worked without any fudges.  The gvfs package is not installed on the base Raspbian Stretch lite image.
gphoto2         2.5.11         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.12         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         gcc, ltdl, USB, serial without locking

From symptoms, gvfs-gphoto2-volume-monitor is watching the usb ports for cameras to be attached.  Even when it 'decides' to not mount the camera, the port and/or camera is left in a state where libgphoto2 can not access it.  Looks like either it is not fully releasing the usb port or (more likely), it sent some sort of query to the camera that left it in a state that gphoto could not connect.  The pattern says after gvfs-gphoto2-volume-monitor got the results it needed to say 'dont mount', it dropped the usb port, but should have first sent something to the camera to properly close/reset it.  Just 'hanging up' like that left the camera waiting for something, and not responding to anything else.  Power cycle clears it, but then gvfs-gphoto2-volume-monitor check again if it should mount the camera.

Raspbian Stretch lite is likely to be the working solution for me.  Reporting the research, in the hope it points to pieces that can used to develop a more general solution.

msmeissn, 2018-08-07:
you can move away or remove the gvfs-gphoto2-volume-monitor binary. this way it will not try to connect to the camera

mMerlin, 2018-08-07:
Yes.  Saw that on the gphoto2 issue.  Again, only a work around, since it disables other potentially useful functionality.  gphoto, at least for the EOS cameras, does not need it, but other things might.

dschneiderch, 2020-05-21:
also ran into this.  i was able to get access to my canon by killing the volume-monitor. thanks @mMerlin 
```
couguser@cppc2:~$ gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Canon EOS 1200D                usb:001,006     
couguser@cppc2:~$ ps ax | grep gphoto2
  699 pts/2    S+     0:00 grep --color=auto gphoto2
 1718 ?        Ssl    0:00 /usr/lib/gvfs/gvfs-gphoto2-volume-monitor
couguser@cppc2:~$ sudo kill 1718
couguser@cppc2:~$ ps ax | grep gphoto2
  703 pts/2    S+     0:00 grep --color=auto gphoto2
```

then I ran this so hopefully it persists and doesn't screw too much else up
```
sudo chmod -x /usr/lib/gvfs/gvfs-gphoto2-volume-monitor
sudo chmod -x /usr/lib/gvfs/gvfsd-gphoto2
```

msmeissn, 2020-06-14:
seems solved

closed 2020-06-14 by msmeissn