gphoto/libgphoto2 #300: Wait for event while in capture mode?

sukoy, 2018-08-08:
Hi,

Is possible to wait for and event while in capture mode?

Capture preview thread

`while(1){
		retval = gp_file_new(&file);
		retval = gp_camera_capture_preview(canon, file, canoncontext);
		retval = gp_file_save(file, output_file);
		gp_file_unref(file);
	}`

Capture trigger image
`
while(1)
{
gp_camera_wait_for_event(camera, waittime - timediff, &evtype, &data, context);
//wait for GP_EVENT_FILE_ADDED
//if file added
retval = gp_camera_file_get(camera, queue[0].path.folder, queue[0].path.name,
			GP_FILE_TYPE_NORMAL, file, context);
retval = gp_file_get_data_and_size (file, &buffer, &size);
}
` 

Thanks

msmeissn, 2018-08-08:
the USB cameras can only have 1 thread accessing it, they have no concept of different sessions.

You need to do this in a loop, and interleave the capture preview , trigger capture calls and wait events (just wait for a small amount of time).

msmeissn, 2018-08-09:
hopefully solved :)

sukoy, 2018-08-09:
mmmh maybe I miss something? I'm trying in this way

`res = gp_camera_init(camera, context);
CAMERA_CHECK_GP(res, "gp_camera_init");
//prepare to capture on trigger
res = gp_camera_trigger_capture (camera, context);
CAMERA_CHECK_GP(res, "gp_camera_trigger_capture");
while(1)
{
res = gp_camera_wait_for_event(camera, waittime, &evtype, &data, context);
if (res != GP_OK) {
	res = gp_camera_capture_preview(camera, file, context);
	CAMERA_CHECK_GP(res, "gp_camera_capture_preview");
}	
}`

gp_camera_capture_preview return #define GP_ERROR_CAMERA_BUSY    -110

There is a way to stop the trigger command?  
Maybe I have to deinit the camera? in this case it will be to slow poll camera for events while streaming the preview. Maybe poll the file system for new file is a better way?

Thanks

msmeissn, 2018-08-09:
From the trigger command up to the captured image being available the preview will return busy ...

after the wait event reports the newly captured image being added the preview should get back to working again

sukoy, 2018-08-10:
What I need is quite simple. While in preview mode,check if a new file has been added and download it. After some test, the trigger command lock the camera until a new file has been added, after that the camera is again available for preview mode. In few words capture trigger and preview can't work together and I have to find an other way to do that.

sukoy, 2018-08-10:
OK i just did, if gp_camera_capture_preview return busy, drain events queue with gp_camera_wait_for_event with a small amount of timeout. It works well, now the question is simple what gp_camera_trigger_capture does? honestly, read from code, I still don't understand

sukoy, 2018-08-11:
yes it does, Thanks for your great help

closed 2018-10-14 by msmeissn