gphoto/libgphoto2 #301: Cannot --set-config on Canon EOS M10

brivbl, 2018-08-10:
Hi,

I have a problem trying to configure my capture and image parameters with gphoto2 on my Canon EOS M10. 

I can get parameters and take captures  without any problem. The only command I cannot use is : gphoto2 --set-config

I tried using --set-config, --set-config-value and --set-config-index on parameters like iso or aperture, but it always make my camera crash, I get the error : "PTP I/O error"

here is my logfile running : env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --set-config-value /main/imgsettings/iso=100

0.000020 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000084 main                        (2): gphoto2 2.5.17.1
0.000101 main                        (2): gphoto2 has been compiled with the following options:
0.000113 main                        (2):  + gcc (C compiler used)
0.000123 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000133 main                        (2):  + exif (for displaying EXIF information)
0.000143 main                        (2):  + no cdk (for accessing configuration options)
0.000153 main                        (2):  + no aa (for displaying live previews)
0.000163 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000172 main                        (2):  + no readline (for easy navigation in the shell)
0.000183 main                        (2): libgphoto2 2.5.19.1
0.000194 main                        (2): libgphoto2 has been compiled with the following options:
0.000204 main                        (2):  + all camlibs
0.000214 main                        (2):  + gcc (C compiler used)
0.000224 main                        (2):  + ltdl (for portable loading of camlibs)
0.000234 main                        (2):  + EXIF (for special handling of EXIF files)
0.000244 main                        (2): libgphoto2_port 0.12.0
0.000255 main                        (2): libgphoto2_port has been compiled with the following options:
0.000264 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.000274 main                        (2):  + gcc (C compiler used)
0.000284 main                        (2):  + ltdl (for portable loading of iolibs)
0.000294 main                        (2):  + USB (libusb1, for USB cameras)
0.000305 main                        (2):  + serial (for serial cameras)
0.000314 main                        (2):  + no resmgr (serial port access and locking)
0.000324 main                        (2):  + no ttylock (serial port locking)
0.000333 main                        (2):  + no lockdev (serial port locking)
0.000344 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000354 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000363 main                        (2): invoked with following arguments:
0.000374 main                        (2):   --debug
0.000384 main                        (2):   --debug-logfile=my-logfile.txt
0.000395 main                        (2):   --set-config-value
0.000406 main                        (2):   /main/imgsettings/iso=100
0.000457 load_settings               (2): Creating gphoto config directory ('/home/brieuc/.gphoto')
0.000544 load_settings               (2): Loading settings from file '/home/brieuc/.gphoto/settings'.
0.000865 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.000891 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/local/lib/libgphoto2_port/0.12.0'...
0.001001 foreach_func                (2): Called for filename '/usr/local/lib/libgphoto2_port/0.12.0/disk'.
0.001205 gp_port_library_list        (2): found fstab fsname UUID=45927b37-7d78-48ba-ba19-d9eb065e5a99
0.001229 gp_port_library_list        (2): found fstab fsname /swapfile
0.001323 gp_port_library_list        (2): found mtab fsname sysfs
0.001336 gp_port_library_list        (2): found mtab fsname proc
0.001345 gp_port_library_list        (2): found mtab fsname udev
0.001364 gp_port_library_list        (2): found mtab fsname devpts
0.001373 gp_port_library_list        (2): found mtab fsname tmpfs
0.001389 gp_port_library_list        (2): found mtab fsname /dev/sda3
0.001402 gp_port_library_list        (2): found mtab fsname securityfs
0.001411 gp_port_library_list        (2): found mtab fsname tmpfs
0.001427 gp_port_library_list        (2): found mtab fsname tmpfs
0.001443 gp_port_library_list        (2): found mtab fsname tmpfs
0.001463 gp_port_library_list        (2): found mtab fsname cgroup
0.001486 gp_port_library_list        (2): found mtab fsname cgroup
0.001495 gp_port_library_list        (2): found mtab fsname pstore
0.001512 gp_port_library_list        (2): found mtab fsname cgroup
0.001561 gp_port_library_list        (2): found mtab fsname cgroup
0.001571 gp_port_library_list        (2): found mtab fsname cgroup
0.001579 gp_port_library_list        (2): found mtab fsname cgroup
0.001588 gp_port_library_list        (2): found mtab fsname cgroup
0.001596 gp_port_library_list        (2): found mtab fsname cgroup
0.001605 gp_port_library_list        (2): found mtab fsname cgroup
[... 2568 lines cut ...]

1.158511 ptp_usb_getdata             (2): Reading PTP_OC 0x1014 (Get device property description) data...
1.158523 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
1.162055 gp_port_read                (3): Read    26 = 0x1a out of 1024 bytes from port: (hexdump of 26 bytes)
0000  1a 00 00 00 02 00 14 10-26 00 00 00 2e d0 06 00  ........&.......
0010  00 00 00 08 00 00 00 08-00 00                    ..........      

1.162107 ptp_usb_getresp             (2): Reading PTP_OC 0x1014 (Get device property description) response...
1.162117 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
1.189210 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 01 20-26 00 00 00              ....... &...    

1.189270 gp_widget_set_value         (2): Setting value of widget 'Size of Output Data from Camera' to '524288'...
1.189287 ptp_usb_sendreq             (2): Sending PTP_OC 0x1014 (Get device property description) (0xd02f) request...
1.189296 gp_port_write               (3): Writing 16 = 0x10 bytes to port...
1.189610 gp_port_write               (3): Wrote   16 = 0x10 bytes to port: (hexdump of 16 bytes)
0000  10 00 00 00 01 00 14 10-27 00 00 00 2f d0 00 00  ........'.../...

1.189637 ptp_usb_getdata             (2): Reading PTP_OC 0x1014 (Get device property description) data...
1.189668 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
1.191358 gp_port_read                (3): Read    26 = 0x1a out of 1024 bytes from port: (hexdump of 26 bytes)
0000  1a 00 00 00 02 00 14 10-27 00 00 00 2f d0 06 00  ........'.../...
0010  00 00 00 08 00 00 00 08-00 00                    ..........      

1.191404 ptp_usb_getresp             (2): Reading PTP_OC 0x1014 (Get device property descriptio

argosvr, 2018-08-14:
the problem seams to ne there 
41.317050 gp_libusb1_read libusb1.c:603: 'libusb_bulk_transfer (port->pl->dh, port->settings.usb.inep, (unsigned char*)bytes, size, &curread, port->timeout)' failed: Operation timed out (-7)

How can we wolve it?

msmeissn, 2019-02-10:
persistent configuration for EOS M is not yet working ... I dont know why or how to do it yet :/