gphoto/libgphoto2 #305: gphoto2 commandline errors on second time call onwards

tinkyawoo, 2018-09-07:
Hi 

I can give any command for the first time
gphoto2 --summary (OK)
from then on.  whatever command I gave throws error "Unexpected"...

So, I am trying to use gphoto2 --shell mode.
The problem is, i need to retrieve camera model and storage info for Model, Free Space on SD card etc, but --shell mode do not have command to get --summary and --storage-info command.

Any help?

tinkyawoo, 2018-09-07:
I am already rewriting my code to use gphoto2 --shell mode. 
Is there any way I could get camera Make, Model and StorageInfo in --shell mode?

msmeissn, 2018-09-07:
currently thats not possible, it would need to be added to shell mode

tinkyawoo, 2018-09-07:
hmm.... anyway thanks, I am writing a wrapper to get storage-info before taking capture, and get image size from exif and deduct it from initial storage-info. If we could get that available in shell mode would be really handy for interactive scripting without leaving shell and avoid usb connection lockup issue like in my initial problem which work first time and failed for sub-sequence gphoto2 cli commands