gphoto/libgphoto2 #306: "heartbeat" signal/command/function

define-private-public, 2018-09-11:
Say if I've already established a connection to a camera, is there a function I can to to check that the connection to the camera is still good?  Would `gp_camera_wait_for_event()` solve that issue?

msmeissn, 2018-09-11:
yes, wait for event would solve this, it would return io errors if the camera is gone

define-private-public, 2018-09-11:
How threadsafe is communication with a single camera object?  Say if I call that `gp_camera_wait_for_event()`, then in another thread, a few seconds later I call `gp_camera_full()` in another thread.  What will happen?  Or do you not know?

define-private-public, 2018-09-13:
For the `timeout` parameter if I set it to `0`, will it wait indefinitely (as I've seen with many other timeouts in other libraries)?  Or am I almost guaranteed a `GP_EVENT_TIMEOUT`?

Also, if I do get a `GP_EVENT_TIMEOUT` or a `GP_EVENT_UNKNOWN`, will the function still return `GP_OK`?

define-private-public, 2018-09-13:
Okay, from some initial testing, I was getting `GP_ERROR_IO_USB_FIND` when I disconnected my camera and was calling `gp_camera_wait_for_event()`.  Is that the only IO error I should check for (if there is a disconnected), or are there others I should test for too?

msmeissn, 2018-09-14:
I would just handle all errors and only accept GP_OK as "okay".

msmeissn, 2018-09-14:
timeout will also report GP_OK and listed GP_EVENT_TIMEOUT in the eventcode

define-private-public, 2018-09-14:
Okay, that seems to be what I'm getting right now, and it seems to be working for me.

If there was a more formalized function for doing this (e.g. `bool gp_camera_check_connection();`) that would be nice.  If you want to turn this ticket into something track that feature request go head.  If not, feel free to close this.

Thanks again.