gphoto/libgphoto2 #308: Canon 6D Mk2 "shuttercount not found in configuration tree."

gambitaw, 2018-09-17:
I couldn't find any other app that can find shutter count that is free (most of the paid can not as well).
The number is not in the Exif as well. 

used: gphoto2 --get-config shuttercount
*** Error ***              
shuttercount not found in configuration tree.
*** Error (-1: 'Unspecified error') ***  

Tried grep and no results, and in the man page no reference for this option.
Any idea ?

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.17         clang, popt(m), no exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.19         all camlibs, clang, ltdl, no EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, clang, ltdl, USB, serial without locking

msmeissn, 2018-09-17:
The EOS are only rarely exposing this value for some reason :/

baem2, 2021-01-20:
Is it possible that availableshots is mixed up with the shuttercount on the 6D 2?