gphoto/libgphoto2 #309: How to capture image quicker ?

gachiemchiep, 2018-09-19:
Hello @msmeissn 

At first thank you for your great work.

As the title said, do you know how to decrease the time which cost for capturing image ?

Currently, we are using python-gphoto2 and gphoto2 cli to capture image from Nikon cameras. The capturing work great without any error. But the speed is very slow. It took us 1.5 second to capture-image-and-download . Is there any approach that could help us decrease the capture-time from 1.5 second into 0.5 second.

[libgphoto2.log](https://github.com/gphoto/libgphoto2/files/2395859/libgphoto2.log)

Digging deeper into the log, i found that the first 1 second is used to initialize gphoto2 and camera connection. So an idea came into my mind is that if we could keep the camera connection alive forever, 
then the capture-image-and-download time should decease from 1.5 second into 0.5 second.

Is this approach possible ? And if this is possible, would you mind lend me a hand by sharing example code or manual ? 

Thank you very much

msmeissn, 2018-09-22:
we are getting some event info timeouts, but not more than 0.5 seconds though.
at 0.094593 we trigger the capture
at 0.25 and 0.61 we try reading events but do not get confirmation yet
at 1.01 we receive confirmation capture was done and download the image
at 1.17 we have finished reading the image
there is a final wait up to 1.31 seconds.

you can use --shell mode to keep it alive longer or e.g. use interval capture with -F frames -I seconds   if you have specific timeframes.

there seems some potential in optimizing the event checking to shave of some 0.x seconds

msmeissn, 2018-09-22:
i added a test patch that might shave off 0.2 or 0,.3 seconds.

pdeman, 2019-03-08:
Hi, I have similar problem using a canon 5DSR.
using sample-trigger-capture.c I can't get faster than 1.3 s while using interval capture with -F frames -I seconds I know I can go at least at 1s. how can I get faster ? 
PS: how do you get the log file ?