gphoto/libgphoto2 #310: void* data passed to gp_context_set_error_func is different from void* data received when capturing in a new thread

clarkmatt, 2018-09-25:
Hi,

I adapted the sample-capture.c example to capture images in a new thread. At the beginning of main() I call gp_context_set_error_func() but the pointer pass as void* data is different from what is received by errordumper(). This prevents me from logging information relevant about the object in which the error first occured. I looked at the source code and it seems gp_context_set_error_func simply sets context->error_func_data = data so I'm not sure where the ptr address is changing. Any ideas? Here is my code and output, I didn't include the original includes or capture_to_file() and capture_to_memory() as they were unchanged:

**Code:**
`
/*
 * threaded-capture.c is adapted from sample-capture.c to capture images in a new thread.
 */

#include <unistd.h>
#include <pthread.h>

class myClass{
  public:
    int value;
    Camera *camera;
    GPContext *context;
};

static void errordumper(GPLogLevel level, const char *domain, const char *str,
                 void *data) {
  // BUG: I expect this pointer to be the same as what is passed as the last argument to 
  // gp_context_set_error_func() but it is not.
  std::cerr << "error_func_ptr: " << data << std::endl;
  std::cerr << "error_func_val: " << ((myClass*)data)->value << std::endl;
}

void * camcap(void *ptr){
    int	retval;
    FILE 	*f;
    char	*data;
    unsigned long size;

    std::cerr << "new_thread_ptr: " << ptr << std::endl;
    std::cerr << "new_thread_val: " << ((myClass*)ptr)->value << std::endl;

    Camera *camera = ((myClass*)ptr)->camera;
    GPContext *context = ((myClass*)ptr)->context;

    printf("Camera init.  Takes about 10 seconds.\n");
    retval = gp_camera_init(camera, context);
    std::cout << "Camera Initialized" << std::endl;
    if (retval != GP_OK) {
            printf("  Retval of capture_to_file: %d\n", retval);
            exit (1);
    }

    std::cout << "Sleeping..." << std::endl;
    sleep(3);

    capture_to_file(camera, context, "foo.jpg");

    capture_to_memory(camera, context, (const char**)&data, &size);

    f = fopen("foo2.jpg", "wb");
    if (f) {
            retval = fwrite (data, size, 1, f);
            if (retval != size) {
                    printf("  fwrite size %ld, written %d\n", size, retval);
            }
            fclose(f);
    } else
            printf("  fopen foo2.jpg failed.\n");
    gp_camera_exit(camera, context);
    
    return (void*)0;
}

int
main(int argc, char **argv) {
        myClass test;
        test.context = sample_create_context();
        test.value = 28;
        void *ptr = &test;

        // Print value and address of instantiated myClass object
        std::cout << "original_ptr: " << ptr << std::endl;
        std::cout << "original_val: " << test.value << std::endl;
        gp_context_set_error_func(test.context, (GPContextErrorFunc)errordumper, ptr);
	//gp_log_add_func(GP_LOG_ERROR, errordumper, NULL);
	gp_camera_new(&test.camera);

        // Pass myClass object to new thread so it can be passed to camcap()
        void *retval;
        pthread_t handle;
        pthread_create(&handle, NULL, camcap, ptr);
        pthread_join(handle, &retval);
        return 0;
}`

**Output:**
original_ptr: 0x7ffd76171bb0
original_val: 28
new_thread_ptr: 0x7ffd76171bb0
new_thread_val: 28
Camera init.  Takes about 10 seconds.
Camera Initialized
Sleeping...
Capturing.
error_func_ptr: 0x7f5797566f80
error_func_val: -1755940336
  Retval: -7
Pathname on the camera: //foo.jpg
  Retval: 0
  Retval: -108
Deleting.
  Retval: -108
Capturing.
error_func_ptr: 0x7f5797566f80
error_func_val: -1755940336
  Retval: -7
Pathname on the camera: //foo.jpg
  Retval: 0
  Retval: -108
Deleting.
  Retval: -108

msmeissn, 2018-09-30:
The arguments are shifted for some reason, level has the content of domain, domain has the content of str and str has data.

Most obvious thing would be a C++ calling conveention mismatch here.

errordumper needs to be extern "C", as it should be compiled as a C function and not a C++ one, but changing that does not seem to help here, the arguments are still shifted

clarkmatt, 2018-10-03:
That was it, thank you!

closed 2018-10-03 by clarkmatt