gphoto/libgphoto2 #311: Focus issue with Nikon DSC D3500

esavolle, 2018-10-03:
Hi,

I'm testing the new Nikon DSC D3500 with gphoto2 2.5.17 / libgphoto2 2.5.17

When using the focus mode AF-A or AF-S, the camera freezes with message on the display "Connected to smart device".
It is working with the DSC D3400.

I have made several tests with capture-and-download and trigger-capture, varying the focus mode and the capture target (RAM or internal SDCard) for both the Nikon DSC D3500 and D3400.

The logs are quite big due to capture-and-download so i join them with links

[d3500.zip](https://mega.nz/#!SVtmRIAC!BWgtYwDOTdnMnXyUWjeLnKn8_1lNNMWymamCzgBYUn0)
[d3400.zip](https://mega.nz/#!3It0SCAB!QMvQweCwniePfXwO9_9K1v2n9cNpVjGkYSzuXLvVYBQ)

[gphoto2-version.txt](https://github.com/gphoto/libgphoto2/files/2442668/gphoto2-version.txt)

If more tests are needed please just ask.

Btw the USB id for the Nikon DSC D3500 is:
# Nikon DSC D3500
usb:v04B0p0445*
 GPHOTO2_DRIVER=PTP
 ID_GPHOTO2=1

cwester, 2019-01-30:
Can Confirm this problem with the D3500. Any focus mode other than manual causes the camera to lockup.

Craigson, 2019-03-20:
Any resolution on this? I'm having the same issue running Raspbian Jessie Lite on a Raspberry Pi, except my camera is definitely on Manual Focus and I'm still experiencing the issue.

avisecag, 2019-04-05:
Confirming the issue as well, just ran over it. Here's a fun little thing I found:

* Set any auto focus, Open console, run gphoto2 --capture-image
Process will freeze, camera display will show "Connected to smart device". Camera will focus, but it will never capture.
* Open new console, run any command (like reading a config). Command will not succeed, because PTP device is busy.
* Close the old (frozen) console. Camera display will still be showing "Connected to smart device".
* Run a command like reading a config in the new one, command will succeed. 
The device is no longer busy, display message on camera goes away.
* Run --capture image again, same thing will happen.

But here comes the really weird part:
If the first command you run in the new console after closing the frozen one is --capture-image, The camera will immediately respond... and capture *two* images!

msmeissn, 2019-04-07:
we are basically doing the exact same thing as on the D3400, just the cameras never goes "non busy".
In current git I abort this busy wait after 5 seconds, but I dont think it will help much.

Mahmood-Hussain, 2019-04-28:
@msmeissn is this problem fixed now? or still persists

Izzvoli, 2019-06-05:
Facing the same issue (first no capture at all, then 2 at the same time), I found a method to capture photos correctly.

- Once the D3500 is powered, run gphoto2 --capture-image. The focus is made, and the capturing process is running but it never ends.

- Kill the process twice (by pressing  "Ctrl+c" twice or by serching and killing twice the gphoto2 process' pid in another console (pgrep gphoto2, then kill )).

- The capturing process should be over. Now run gphoto2 --reset. (I lack informations on what the command actually resets but hey, it works.)

- Run gphoto2 --capture-image, it should work fine and capture a single photo every time the command is used until the camera is powered off, then you will have to repeat the process.

Reed920, 2021-06-04:
I have the same issue as mentioned above. Is this something I'm doing wrong, something gphoto can do or a problem on Nikon's end? Any updates on the progress of a fix for this or an alternative software I can use?

RaingodSpires, 2022-08-07:
Try --trigger-capture command instead. It works on a D3500.

mloodberg, 2024-06-27:
> Try --trigger-capture command instead. It works on a D3500.

This actually seems to work with Nikon D3500, with continuous captures

gphoto2 --trigger-capture --wait-event-and-download=CAPTURECOMPLETE