gphoto/libgphoto2 #315: ios 10.3.1 doesn't work with libgphoto2 2.5.17 and later

michael-skynorth, 2018-10-17:
My ipad running iOS 10.3.1 doesn't work with libgphoto2 2.5.17 and 2.5.19, it works with 2.5.16, looks like the iOS changes in 2.5.17 is not compatitable with iOS 10.3.1,  the error log is:
0.368554 ptp                         (2): adding new objectid 0x00000054 (nrofobs=1,j=1)
0.368565 ptp                         (2): adding new objectid 0x0045004c (nrofobs=2,j=2)
0.368575 ptp                         (2): adding new objectid 0x00040000 (nrofobs=3,j=3)
0.368585 ptp                         (2): adding new objectid 0x00440055 (nrofobs=4,j=4)
0.368594 ptp                         (2): adding new objectid 0x0030002e (nrofobs=5,j=5)
0.368603 ptp                         (2): adding new objectid 0x00000000 (nrofobs=6,j=6)
0.368614 file_list_func              (2): after list folder
0.368623 ptp                         (2): ptp_object_want: querying handle 0?

Hi @msmeissn ,
any ideas ?

michael-skynorth, 2018-10-17:
Not sure how to attach log here, just put a little more logs, if you need more, let me know:
`0.368248 ptp_usb_getresp             (2): Reading PTP_OC 0x1023 (Get Filesystem Manifest) response...
0.368260 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
0.368531 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 01 20-10 00 00 00              ....... ....

0.368554 ptp                         (2): adding new objectid 0x00000054 (nrofobs=1,j=1)
0.368565 ptp                         (2): adding new objectid 0x0045004c (nrofobs=2,j=2)
0.368575 ptp                         (2): adding new objectid 0x00040000 (nrofobs=3,j=3)
0.368585 ptp                         (2): adding new objectid 0x00440055 (nrofobs=4,j=4)
0.368594 ptp                         (2): adding new objectid 0x0030002e (nrofobs=5,j=5)
0.368603 ptp                         (2): adding new objectid 0x00000000 (nrofobs=6,j=6)
0.368614 file_list_func              (2): after list folder
0.368623 ptp                         (2): ptp_object_want: querying handle 0?

0.368635 file_list_func [library.c:6558](0): 'ptp_object_want (params, params->objects[i].oid, PTPOBJECT_PARENTOBJECT_LOADED|PTPOBJECT_STORAGEID_LOADED, &ob)' failed: 'PTP General Error' (0x2002)
0.368647 gp_context_error            (0): PTP General Error
0.368842 gp_camera_folder_list_files [gphoto2-camera.c:1477](0): 'gp_filesystem_list_files (camera->fs, folder, list, context)' failed: -1
0.369524 gp_camera_free              (2): Freeing camera...
0.369538 gp_camera_exit              (2): Exiting camera ('Apple iPad Air')...`

msmeissn, 2018-10-17:
a bit more debug output (or full one of -L) would help.

michael-skynorth, 2018-10-19:
Hi,  
 I am thinking if we can just skip the handling 0?

msmeissn, 2018-10-19:
i think this is just papering over a problem.
The data seems to get decoded incorrectly and the other objectids are probably also invalid (as they seem UNICODE letters of the filenames) ... I need to check if i missed an offset or so

msmeissn, 2018-10-21:
I disabled manifest read in commit 4375e9c00dd0ffb745d85cd2fc0b26afbea0c0bf as what the standard describes and what apple does does not seem to match

michael-skynorth, 2018-10-23:
thanks,  I tried the patch, it works.

msmeissn, 2018-10-29:
i wil revisit manifest reading at a later time, for now this is fixed

closed 2018-10-29 by msmeissn