gphoto/libgphoto2 #316: HelpCanon PowerShot SX720H Zoom Control

luffecn, 2018-10-24:
I use gphoto2 to control

msmeissn, 2018-10-27:
how exactly do you try it?

what is reported by:
 gphoto2 --summary
 gphoto2 --get-config d055 
 
if --get-config d055 returns something, you can also try gphoto2 --set-config d055=value from d055 range  to see if zoom happens.

The option is currently bnot yet hooked up for configuration, but with known values we can hook it up.

msmeissn, 2018-10-29:
The camera is using Canon EOS M style capture, so it should be possible to hook everything in.

I have adjusted current libgphoto2 GIT to include checks that will do steps 1-4 (as they are the same as EOS M)

luffecn, 2018-10-29:
I have updated my library files, but when I capture a image,  the camera seems to be crashed.
Details at [capture_image.log](https://github.com/gphoto/libgphoto2/files/2524721/capture_image.log)

msmeissn, 2018-10-29:
and your progrqms set evt outputdevice does not crash? exactly what bytes do you send?

luffecn, 2018-10-31:
yes，my program is not crashed.
Below is the correspondence between command and  hexadecimal.
1.opensession
HEX:10000000010002100000000001000000
2.set PTP_OC_CANON_EOS_SetRemoteMode(0x9114) = 0x15
HEX:10000000010014910100000015000000
set PTP_OC_CANON_EOS_SetEventMode(0x9115) = 0x02
HEX:10000000010015910200000002000000
set PTP_DPC_CANON_EOS_EVFOutputDevice(0xD1b0) = 0x08
HEX:0c0000000100109103000000
HEX:1800000002001091030000000c000000b0d1000008000000

msmeissn, 2018-11-03:
i changed a bit to make it align more with your camera (we did a setremotemode 0x1 instead of 0x15)

I also ordered a SX720HS from ebay for tsting :)

msmeissn, 2018-11-03:
so if you can test current git, and see if it helps :/

luffecn, 2018-11-05:
yes, it can capture a image:)
[capture_image.log](https://github.com/gphoto/libgphoto2/files/2547557/capture_image.log)
I will try other features.

luffecn, 2018-11-09:
Hi msmeissn,
I still can't find a way to control SX720HS's zoom through usb. Do you have any suggestions?
Thanks.
[get_config_d055.log](https://github.com/gphoto/libgphoto2/files/2564815/get_config_d055.log)
[set_config_d055.log](https://github.com/gphoto/libgphoto2/files/2564814/set_config_d055.log)

msmeissn, 2018-11-12:
so, current git will see the powerzoom as "zoom", but currently setting it seems not to work :/

luffecn, 2018-11-13:
yes, it does not work on ptp/usb, but it works on ptp/ip. I don't know why.

msmeissn, 2018-11-13:
perhaps a camera limitation itself. there is some more weirdness, like it does not show the option on the second gphoto2 call and so :/
but at least other configurations work now with libgphoto2.
were you able to get a liveview over ptp/iop?

luffecn, 2018-11-14:
I only get a liveview over ptp/usb:)  The capture data over ptp/ip from camera's  mobile app.

msmeissn, 2018-12-13:
the speed might be the speed of the zooming... what values did you try setting?

do you have the id of the 740HS and debugging info so I can add it?

msmeissn, 2018-12-14:
are these the ones you tried for powerzoomposition or speed?
is the value of position related to the focal length that results out of it?

luffecn, 2018-12-15:
These ones I tried for speed.
The position values range from 0d to 127d.
Position command works but speed not.

Johnnygde, 2020-11-02:
After some hours of testing, I would like to know if anybody managed to use the zoom function on either the SX720 or the SX740 yet. 
If it would definately work with a 740, I am willing o buy one, but I like my 720 very much.

What I need would be the following:
- lens moved out all the time (prevent from moving in after a while)
- capture an image and download it every 10 minutes
- using the zoom function like within the Canon connect App. (increasing or decreasing an int  value in this case)

The only thing I achieved was capturing an image.
Once I tried the wifi ptpip connection, using Entangle, it was possible to set the zoom to 127 (full zoom), but nothing in between or no option to set it back to 0 again.

I am using the latest version available with the gphoto updater ( gphoto2 2.5.23.1 - libgphoto2 2.5.26 )

If there is any option to use at least the zoom function like within the App, please let me know.
In case that is not possible (yet) with a 720 or not guaranteed that it works with the 740, maybe someone could suggest an alternative compact superzoom camera that definately works with gphoto2 ?

many thanks in advance

Regards,
John

gabrielesartor, 2021-09-09:
Hi everyone,
I'd like to buy the sx720hs or sx740hs, but I'd like to understand which one is more compatible with libgphoto2.
I just need to set the zoom, capture the image and download it on Linux via USB.
Any news on this topic?
Thank you in advance!