gphoto/libgphoto2 #317: How to trigger burst

kevinkingo, 2018-10-24:
Hi everyone,

I have a Canon EOS 1DX. When I turn the mode into "High speed continuous", and hold the shutter button, the camera just take 4~5 images every second. 
However, when I use the code libgphoto2/examples/sample-capture.c to capture the image, it can only achieve 1 image per second (or less). Even though I setup gphoto2 so that the captured image only reside in the camera RAM rather than the SD card, it's still slow.
Could anyone tell me how to rapidly trigger the burst capture using gphoto2? Maybe disable the auto-focus?

Thank you very much!

Sincerely,
Kevin

dkpho, 2022-08-30:
Hi Kevin, just came across your comment. Did you ever get a solution to this? I'm facing the same issue. Thanks

msmeissn, 2022-09-19:
the sample capture also tries to download the image , so capture+download.

look at sample-trigger-capture.c   for just triggering capture without download.

closed 2022-09-19 by msmeissn