gphoto/libgphoto2 #318: Canon Powershot A460 - Error (-6: „Unsupported operation”)

zodoczi, 2018-10-26:
Hi Guys,

I'm trying to add a substitute camere while reparing my Canon S5 IS, which was my weather cam for 3 years (shuttet died after 1.5 million exposes :) ) I put the script here which change to long time exposure during the night and auto mode for the day FYI.
idokep.hu/webkamera/dz99
https://github.com/rfsparkling/gphoto2-RaspberryPi-weather-cam-script

I'm trying to use a Canon Powershot A460, which was in the supported camera list. However having difficulties to test (under Linux Mint 64bit now). Am I missing something? Might this camare is not supported? 

When I use the "image capture" mode in the camera knob, linux not recognize it only in "image review" mode in the camera selector.

Thanks,
Zoltan

when trying to capture (while the camera mode select knob in image review mode):
```
sudo gphoto2 --capture-image
                                                                               
*** Error ***               
Sorry, your camera does not support generic capture
ERROR: Not possible to capture image.
ERROR: Not possible to capture
*** Error (-6: „Unsupported operation”) ***    
```

image review mode set in camera:
```
gphoto2 --auto-detect 
Modell                         Port                                            
----------------------------------------------------------
Canon PowerShot A460 (PTP mode) usb:001,020 
```

capture mode set in camera:
```
gphoto2 --auto-detect 
Modell                         Port                                            
----------------------------------------------------------

```

msmeissn, 2018-10-27:
Supported means "able to get accessed and download images", i tried clarifying this on the support page. 

Canon stopped offering remote control / capture support for their PowerShot series several years ago , as I tried to document on http://gphoto.org/doc/remote/

One thing that might work with the A460  is using "CHDK", the third party firmware addon, which we support to some degree.

zodoczi, 2018-10-29:
"able to get accessed and download images" oh I see... now I'm understand everything... thank you! 

CHDK is a good idea, a compromise could be to make a CHDK local script which shoots an image every minuite and then gpoto2 would download that image periodically? Makes sense to me at least.

msmeissn, 2018-10-29:
FWIW libgphoto2 supports CHDK in a away that in chdk operation it can trigger capture , get liveview and get/set some values

zodoczi, 2018-10-29:
Amazing! Thank you Marcus, I will give it a try!
Danke schön

zodoczi, 2018-10-31:
tried CHDK but linux still not recognize any USB device in "capture mode" of the camera knob, but its detects (and able to download) in the "review mode" of the camera knob.

tried to reset the camera via gphoto2 (only in preview mode) but that doesnt help either.

even looked into the menu of the A460 but there is no USB connection mode settings etc. (I was thinking might that causing that I cant reach that via USB)

msmeissn, 2019-01-02:
sorry for missing this update. the preview mode is the mode where the powershots are usually accessed.
on activating capture they would extract the lens.
I have the A460 listed as being capture capable... can you capture a --summary output?

msmeissn, 2019-01-02:
hmm. sorry, in the canon SDK notes its a460 is not able

closed 2019-02-10 by msmeissn