gphoto/libgphoto2 #321: Canon EOS 70D - iso and aperture unknown values

nowrep, 2018-10-30:
Log: https://gist.github.com/nowrep/3c0a5842734d7a5c1881d0d36e7e34be

It seems that even though 70D reports OLC version 8, the offset for shutterspeed is 6 and not 7.

This patch works for me:

```diff
diff --git a/camlibs/ptp2/ptp-pack.c b/camlibs/ptp2/ptp-pack.c
index 5319525fc..df28f4e5a 100644
--- a/camlibs/ptp2/ptp-pack.c
+++ b/camlibs/ptp2/ptp-pack.c
@@ -2481,7 +2481,9 @@ ptp_unpack_CANON_changes (PTPParams *params, unsigned char* data, int datasize,
 				ce[i].u.propid = proptype;
 				/* hack to differ between older EOS and EOS 200D newer */
 				switch (olcver) {
-				case 0x8:
+				case 0x8:	/* EOS 70D */
+					curoff += 6;
+					break;
 				case 0xf:
 					curoff += 7;	/* f (200D), 8 (M10) */
 					break;
```

msmeissn, 2018-10-30:
i did it a bit differently, please cross check current git

nowrep, 2018-10-30:
Sure, that works too. You might want to recheck if it still works with M10 or update that comment.