gphoto/libgphoto2 #322: Nikon Coolpix S3300 extremely long capture time (~45secs)

bp100a, 2018-11-05:
I'm running gphoto2 (v2.5.17, libgphoto2 v 2.5.19, libgphoto2_port v0.12.0) on a Raspberry Pi Model 3B with a Nikon S3300 camera. The gphoto2 command line works, but takes 45 seconds to take  picture and a total of 1 minute to download to the usb drive (3.5MB image files). The camera "works" as in the pictures look fine.

The debug trace shows escalating timeouts leading up to back-to-back 20 second timeouts. Any ideas on how I can diagnose this issue?

See debug trace log below,
======================================================
New file is in location /store_00010001/DSCN0030.JPG on the camera
Saving file as /mnt/usb/pic4.jpg
Deleting file /store_00010001/DSCN0030.JPG on the camera
0.000710 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000913 main                        (2): gphoto2 2.5.17
0.000992 main                        (2): gphoto2 has been compiled with the following options:
0.001028 main                        (2):  + gcc (C compiler used)
0.001058 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.001088 main                        (2):  + exif (for displaying EXIF information)
0.001117 main                        (2):  + no cdk (for accessing configuration options)
0.001145 main                        (2):  + no aa (for displaying live previews)
0.001173 main                        (2):  + no jpeg (for displaying live previews in JPEG format)
0.001201 main                        (2):  + no readline (for easy navigation in the shell)
0.001238 main                        (2): libgphoto2 2.5.19
0.001278 main                        (2): libgphoto2 has been compiled with the following options:
0.001308 main                        (2):  + all camlibs
0.001337 main                        (2):  + gcc (C compiler used)
0.001365 main                        (2):  + ltdl (for portable loading of camlibs)
0.001392 main                        (2):  + EXIF (for special handling of EXIF files)
0.001428 main                        (2): libgphoto2_port 0.12.0
0.001468 main                        (2): libgphoto2_port has been compiled with the following options:
0.001499 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.001527 main                        (2):  + gcc (C compiler used)
0.001555 main                        (2):  + ltdl (for portable loading of iolibs)
0.001586 main                        (2):  + USB (libusb1, for USB cameras)
0.001614 main                        (2):  + serial (for serial cameras)
0.001643 main                        (2):  + no resmgr (serial port access and locking)
0.001671 main                        (2):  + no ttylock (serial port locking)
0.001698 main                        (2):  + no lockdev (serial port locking)
0.001727 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.001754 main                        (2): IOLIBS env var not set, using compile-time default instead
0.001781 main                        (2): invoked with following arguments:
0.001809 main                        (2):   --debug
0.001836 main                        (2):   --debug-loglevel=all
0.001864 main                        (2):   --debug-logfile=gphoto2.log
0.001891 main                        (2):   --capture-image-and-download
0.001918 main                        (2):   --force-overwrite
0.001945 main                        (2):   --filename
0.001971 main                        (2):   /mnt/usb/pic5.jpg
0.002250 load_settings               (2): Creating gphoto config directory ('/home/pi/.gphoto')
0.002474 load_settings               (2): Loading settings from file '/home/pi/.gphoto/settings'.
0.002898 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.002970 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/local/lib/libgphoto2_port/0.12.0'...
0.003417 foreach_func                (2): Called for filename '/usr/local/lib/libgphoto2_port/0.12.0/disk'.
0.004431 gp_port_library_list        (2): found fstab fsname proc
0.004509 gp_port_library_list        (2): found fstab fsname PARTUUID=2e63d0fe-01
0.004604 gp_port_library_list        (2): found fstab fsname PARTUUID=2e63d0fe-02
0.004971 gp_port_library_list        (2): found mtab fsname /dev/root
0.005045 gp_port_library_list        (2): found mtab fsname devtmpfs
0.005095 gp_port_library_list        (2): found mtab fsname sysfs
0.005139 gp_port_library_list        (2): found mtab fsname proc
0.005180 gp_port_library_list        (2): found mtab fsname tmpfs
0.005329 gp_port_library_list        (2): found mtab fsname devpts
0.005373 gp_port_library_list        (2): found mtab fsname tmpfs
0.005449 gp_port_library_list        (2): found mtab fsname tmpfs
0.005530 gp_port_library_list        (2): found mtab fsname tmpfs
0.005631 gp_port_library_list        (2): found mtab fsname cgroup
0.005673 gp_port_library_list        (2): found mtab fsname cgroup
0.005713 gp_port_library_list        (2): found mtab fsname cgroup
0.005752 gp_port_library_list        (2): found mtab fsname cgroup
0.005791 gp_port_library_list        (2): found mtab fsname cgroup
0.005950 gp_port_library_list        (2): found mtab fsname cgroup
0.005992 gp_port_library_list        (2): found mtab fsname cgroup
0.006032 gp_port_library_list        (2): found mtab fsname systemd-1
[... 459 lines cut ...]
54.652332 debug_objectinfo            (2):   AssociationDesc: 0x00000000
54.652359 debug_objectinfo            (2):   SequenceNumber: 0x00000000
54.652386 debug_objectinfo            (2):   ModificationDate: 0x5be0daa6
54.652413 debug_objectinfo            (2):   CaptureDate: 0x5be0daa6
54.652451 gp_filesystem_list_files    (2): Added 'DSCN0031.JPG'
54.652481 internal_append             (2): Internal append DSCN0031.JPG to folder store_00010001
54.652513 gp_filesystem_list_files    (2): Listed 'DSCN0031.JPG'
54.652544 internal_append             (2): Internal append DSCN0031.JPG to folder store_00010001
54.652760 main                        (2): Getting 'DSCN0031.JPG'...
54.653172 save_file_to_file           (2): using fd method
54.653231 gp_camera_file_get          (2): Getting file 'DSCN0031.JPG' in folder '/store_00010001'...
54.653278 gp_filesystem_get_file_impl (2): Getting file 'DSCN0031.JPG' from folder '/store_00010001' (type 1)...
54.653323 lookup_folder_file          (2): Lookup folder /store_00010001 file DSCN0031.JPG
54.653353 lookup_folder               (2): Lookup folder '/store_00010001'...
54.653384 gp_filesystem_get_file_impl (2): Downloading 'DSCN0031.JPG' from folder '/store_00010001'...
54.653429 ptp                         (2): (storage=0x00010001, handle=0x00000000)
54.653467 get_file_func               (2): Getting file 'DSCN0031.JPG'.
54.653530 ptp_usb_sendreq             (2): Sending PTP_OC 0x1009 (Get object) (0xb) request...
54.653563 gp_port_write               (3): Writing 16 = 0x10 bytes to port...
54.653817 gp_port_write               (3): Wrote   16 = 0x10 bytes to port: (hexdump of 16 bytes)
0000  10 00 00 00 01 00 09 10-08 00 00 00 0b 00 00 00  ................

54.653874 ptp_usb_getdata             (2): Reading PTP_OC 0x1009 (Get object) data...
54.653910 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
54.661505 gp_port_read                (3): Read    1024 = 0x400 bytes from port: (hexdump of 1024 bytes)

msmeissn, 2018-11-06:
it issues a filesystem mnanifest command which times out.

good thing is that i disabled that in 2.5.20 again, so you can try libgphoto2 2.5.20

bp100a, 2018-11-06:
That did the trick! Now I can take a picture and write to a USB thumbdrive in about 15 seconds, which is good enough for this project.

for anyone who stumbles on this thread, here's a short script (for a Python running stretch-lite) to install the 2.5.20 version:

bp100a, 2018-11-06:
oops! Here's the script:
==========================
# build & install needed libaries
# libgphoto2 & gphoto2 used for camera control
# need v 2.5.20 to avoid long timeout
cd ~
wget http://downloads.sourceforge.net/project/gphoto/gphoto/2.5.20/gphoto2-2.5.20.tar.bz2
tar xjvf gphoto2–2.5.20.tar.bz2
cd gphoto2–2.5.20/
./configure
make
sudo make install
cd ..

# Please also check that PKG_CONFIG_PATH contains
# ${libdir}/pkgconfig
# before compiling any libgphoto2 frontend
export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig

# now gphoto2 (v2.5.20)
wget http://downloads.sourceforge.net/project/gphoto/gphoto/2.5.20/gphoto2-2.5.20.tar.bz2
tar xjvf gphoto2–2.5.20.tar.bz2
cd gphoto2–2.5.20/
./configure
make
sudo make install
cd ..

closed 2018-11-06 by bp100a