gphoto/libgphoto2 #323: Regression: gphoto2 shuts down Sony ILCE-5000

wferi, 2018-11-07:
Hi,

The Debian jessie version of gphoto2 could download images from my Sony ILCE-5000 via USB in MTP mode:
~~~
gphoto2         2.5.4          gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.4          all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.10.0         gcc, ltdl, no USB, serial without locking
~~~
However, the stretch version:
~~~
gphoto2         2.5.11         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.12         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         gcc, ltdl, USB, serial without locking
~~~
immediately shuts down the camera (and it starts charging as if switched off) on any operation I tried, including `--summary`, which reports (omitting storage details):
~~~
Camera summary:                                                                
Manufacturer: Sony Corporation
Model: ILCE-5000
  Version: 1.0
  Serial Number: 00000000000000003282653003868103
Vendor Extension ID: 0x11 (1.0)

Capture Formats: 
Display Formats: Association/Directory, MPEG, JPEG, Unknown(b982), Unknown(b301), Undefined Type, Unknown(b101), Unknown(b310)

Device Capabilities:
	File Download, No File Deletion, No File Upload
	No Image Capture, No Open Capture, No vendor specific capture

Device Property Summary:
Battery Level(0x5001):(read only) (type=0x2) Enumeration [0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,100] value: 100% (100)
Property 0xd402:(read only) (type=0xffff) 'ILCE-5000'
Property 0xd406:(readwrite) (type=0xffff) '(null)'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd303:(read only) (type=0x2) 1
~~~
while 2.5.4 reported some fields differently:
~~~
Vendor Extension ID: 0xffffffff (1.0)
Display Formats: Association/Directory, MPEG, JPEG, MP4, Unknown(b301), Undefined Type, Unknown(b101), Unknown(b310)
Friendly Device Name(0xd402):(read only) (type=0xffff) ILCE-5000 ('ILCE-5000')
Session Initiator Info(0xd406):(readwrite) (type=0xffff) '(null)'
Perceived Device Type(0xd407):(read only) (type=0x6) 1
Property 0xd303:(read only) (type=0x2) 1
~~~
2.5.20 behaves much like 2.5.11, it also shuts down the camera immediately, but forgets MP4 and recognizes ARW and fails to read some properties:
~~~
Display Formats: Association/Directory, MPEG, JPEG, Unknown(b982), Unknown(b301), Undefined Type, ARW, Unknown(b310)
Property 0xd402:(read only) (type=0x0) Undefined
Property 0xd406:(readwrite) (type=0x0) Undefined
~~~
(If I put the camera in "PC Remote" mode, `gphoto2 --capture-image-and-download` works, though.)
I'm attaching the result of `env LANG=C gphoto2 --debug --debug-logfile=`[my-logfile.txt](https://github.com/gphoto/gphoto2/files/2552952/my-logfile.txt)` --summary`.
Any idea what could have gone wrong? Please don't hesitate to ask for any information still missing.
Thanks.

msmeissn, 2018-11-07:
moved it.

the camera might just switch to a different mode after a single access, as windows usually opens just 1 session and keeps it open.

this might be tricky to approach

closed 2019-02-10 by msmeissn

msmeissn, 2019-02-10:
I removed a command call on close which caused it to swirtch to charger mode.... Need to reverse engineer the exact arguments there.

wferi, 2019-02-15:
I confirm that this patch (applied on top of 2.5.22) fixes the regression for me.
Thanks much!
Your work on free software is most appreciated.