gphoto/libgphoto2 #325: Canon Powershot SX540 HS support

mbolis, 2018-11-13:
I personally own a Canon Powershot SX540 HS, which is not currently supported.
I hoped it could work nonetheless, since other SX5xx cameras are supported. Unfortunately, I can't get it to capture any image.
Can / will this model be supported in a future release?
Thanks

msmeissn, 2018-11-14:
can you capture:

gphoto2 --summary --debug --debug-logfile=summary-debug.txt > summary.txt
gphoto2 --list-all-config > list-all-config.txt

and attach the 3 txt files?

mbolis, 2018-11-15:
gphoto2 --summary --debug --debug-logfile=summary-debug.txt > summary.txt
[summary.txt](https://github.com/gphoto/libgphoto2/files/2587145/summary.txt)
[summary-debug.txt](https://github.com/gphoto/libgphoto2/files/2587146/summary-debug.txt)

gphoto2 --list-all-config > list-all-config.txt
output:
```
*** Error (-6: 'Unsupported operation') ***
```
[list-all-config.txt](https://github.com/gphoto/libgphoto2/files/2587144/list-all-config.txt)

msmeissn, 2018-11-16:
can you update to current git of libgphoto2? 

some changes for the new SX firmware is added and remoter capture will start to work

mbolis, 2018-11-16:
Done, I post the new results below.

$ gphoto2 --summary --debug --debug-logfile=summary-debug.txt > summary.txt
[summary.txt](https://github.com/gphoto/libgphoto2/files/2591108/summary.txt)
[summary-debug.txt](https://github.com/gphoto/libgphoto2/files/2591109/summary-debug.txt)

$ gphoto2 --list-all-config > list-all-config.txt
Mighty hangs. Like forever. And the camera gets ugly stuck, not responding to any button press. I have to remove the battery to unstuck it.

I also tried:
$ gphoto2 --debug --debug-logfile=list-all-config-debug.txt --list-all-config > list-all-config.txt
[list-all-config.txt](https://github.com/gphoto/libgphoto2/files/2591116/list-all-config.txt)
[list-all-config-debug.txt](https://github.com/gphoto/libgphoto2/files/2591117/list-all-config-debug.txt)
After a while I hit Ctrl+C a couple times to get it out of its stupor... there seems to be an IO error in list-all-config-debug.txt, maybe it can help understand what's going on?

(BTW, thanks, I appreciate your helpfulness very much)

msmeissn, 2018-12-13:
can you perhaps also try current libgphoto2 git?

I fixed a bug in the code there

kleinerstromer1986, 2018-12-21:
Hello, I own a brandnew Canon Powershot SX540 HS too, which can take only the first photo. :( 
At the first photo: gphoto2 capture the image with autofocus in a resolution of 20M, download this file and delete the file on the camera. All fine. 

But at the second photo, gphoto2 can't capture a new photo, because the camera is in use,  I/O Error -110, Error: EOS M Full-Press failed (0x2019) PTP Device busy.
In the screen on the camera after the first capture is this message: "Use target device" 

When i  turn out the usb cable and pull it in again, i can capture a new picture. 
The same workaround works with power off and power on the camera before i want to capture a second photo. 

I think my raspbian (Stretch) with gphoto2 (version 2.5.20, libgphoto2 2.5.21, libgphoto2_port 0.12.0) can't close the usb port(connection) correctly, so i can't trigger the camera again to capture a photo. 

How can I fix this problem? Can I do any config settings over gphoto2 on the camera to fix this problem? 

for any help, i will be very thanksful :)

mbolis, 2018-12-26:
@msmeissn, sorry for taking so long.
For me latest git version worked fine, thank you very much!

msmeissn, 2018-12-26:
thanks for checking back!

closed 2018-12-26 by msmeissn

kleinerstromer1986, 2018-12-26:
@mbolis: which is your working camera configuration? I have problems with the second photo. 
Can you send me informations, how you use your camera. 

thank you for your help. :)

mbolis, 2018-12-27:
@kleinerstromer1986, my camera has stock settings, I'm running gphoto2 on a Fedora 29 machine.
As you can evince from previous messages, I'm using the bleeding edge development version of libgphoto2, which I got from this repository and built following the instructions provided.
I'm afraid you'll have to fetch yourself the latest version and build it your Raspbian...