gphoto/libgphoto2 #329: LGPL-2.1 License is too restrictive

cam3ran3rd, 2018-11-28:
If the goal is to further this library's adoption and the contributions to it, the current license is far too restrictive. As I understand it, the LGPL-2.1 license means that any application using this library must disclose it's source code, which means that no business applications unwilling to do this will be able to use this library. This will limit the contributions to and adoption of this library. I suggest that license is updated to something less restrictive such as GNU or Apache.

msmeissn, 2018-11-29:
Applications can (dynamically) link against LGPL libraries and do not need to disclose their source code. If someone is doing modifications to the library they need to be published.

see e.g.
https://softwareengineering.stackexchange.com/questions/86142/what-exactly-do-i-need-to-do-if-i-use-a-lgpl-licensed-library
or
http://answers.google.com/answers/threadview/id/439136.html

I have not heard of anybody having problems with the license before, so I do not see a need.

closed 2018-11-29 by msmeissn

cam3ran3rd, 2018-11-29:
@msmeissn how do you propose dynamically linking if someone wanted to use this in a mobile or any other client application? Or even in some type of IOT hardware device? This seems like an impossible requirement to meet, thus the library becomes unusable.
Is there any downside to using a license that just requires just attribution as opposed to dynamic linking? This would increase the reach and usefulness of the library, which would also encourage further development.