gphoto/libgphoto2 #331: Sony A6300

Edhyx, 2018-12-10:
In spite of my last modification in the sony-a6300.txt camlibs file, when I execute the gphoto --list-all-config command, on line F-number, I get this:

/main/capturesettings/f-number
Label: F-Number
Readonly: 0
Type: RANGE
Current: 0
Bottom: 0
Top: 655.35
Step: 0.01
END

Instead of this (written is the sony-a6300.txt file)

/main/capturesettings/f-number
Label: F-Number
Type: RADIO
Current: 0
Choice : 0 3.5
Choice: 1 4
Choice: 2 4.5
Choice: 3 5
Choice: 4 5.6
Choice: 5 6.3
Choice: 6 7.1
Choice: 7 8
Choice: 8 9
Choice: 9 10
Choice: 10 11
Choice: 11 13
Choice: 12 14
Choice: 13 16
Choice: 14 18
Choice: 15 20
Choice: 16 22
END

Have you got an explanation ? Thanks a lot for your help

msmeissn, 2019-02-10:
This is what the camera reports to us currently. Not sure if there is ever a fixed list of F-Numbers we could hardcode and add :/

notnot, 2019-02-10:
I am about to set the a6300 to work. Am i right that this issue won't prevent setting the diaphragm, but just not with the convenient numbers we are used to?

msmeissn, 2019-02-11:
yes, just tried with a a7mr2