gphoto/libgphoto2 #334: Canon EOS 2000D PTP Device Busy on second capture on slow hardware

rgenoud, 2018-12-17:
Hi,
I've been playing with Canon EOS 2000D and it works like a charm on a debian PC.
(I'm doing a simple --capture-image-and-download)
However, on a slower hardware (an ARM board at91sam9g35-cm), it works perfectly the first time, and then it fails with :
 *** Error ***              
Canon EOS Full-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-110: 'I/O in progress') ***

Here are a log file with a success, and a log file with a failure

[logfile-fail.txt](https://github.com/gphoto/libgphoto2/files/2686928/logfile-fail.txt)
[logfile-success-data-removed.txt](https://github.com/gphoto/libgphoto2/files/2686929/logfile-success-data-removed.txt)
It may be related to issue #30 , but I'm not sure.

Regards,
Richard

sasa00, 2020-02-26:
Hi,
I tried it with the latest libghoto2: 2.5.24.1 and gphoto2: 2.5.23.1. 
My hw is a RPi4 (4GB RAM). The first picture works and all tries afterwards fail like the description above. I thought perhaps might a log from another hw help:

[notworking.txt](https://github.com/gphoto/libgphoto2/files/4257398/notworking.txt)

Kind regards,
Sascha

lagrandeours, 2021-03-13:
hello does anyone found a solution, i have the same issue at the second shoot
best regards,
Hyd

PJF16, 2023-03-14:
Hey! I found a solution for this. Add --wait-event=500ms before the capture command. So for example the command would look like: 
`gphoto2 --wait-event=500ms --capture-image-and-download --filename=%s`

brleme, 2025-02-08:
Hi all,

Anybody has any progress? I have a Nikon D3500 using a Jetson Xavier and it get stuck (PTP Device Busy) after the first capture. Then, after aborting and trying to capture again, it takes two frames (without problem).

I tried the --wait-event=500ms(and 2000ms) but didn't work

I am using:
gphoto2         2.5.28    
libgphoto2      2.5.31   
libgphoto2_port 0.12.2

Thank you,

knro, 2025-02-26:
Confirmed from my side as well. Several INDI users also reported this issue with their camera. PTP Busy. I'm attaching debug log.

[gphoto2.txt](https://github.com/user-attachments/files/18978889/gphoto2.txt)