gphoto/libgphoto2 #335: Official Android port of this library?

laheller, 2018-12-18:
An unofficial but quite old port exists, maybe it can help to make an official Android port:
https://github.com/JohnChu/Snoopy

msmeissn, 2018-12-20:
(FWIW) I currently lack the android knowledge

laheller, 2019-01-24:
@msmeissn 

Maybe this way it might work:
https://github.com/libusb/libusb/tree/master/android

As you can see, they just prepared jni/*.mk files (basically a tiny [build files](https://developer.android.com/ndk/guides/android_mk) for Android builds).
Everything else is described in their _README_. Using that I was able to build _libusb_ for _Android_ with just one single command from cmdline.

I will try the same or similar on libgphoto2 and will let you know.

Best Regards,

Ladislav

laheller, 2019-02-07:
@msmeissn  @hfiguiere  @ndim 

Update: As I wrote in first comment, there is one old unofficial port.
It has an already prepared Android.mk build file:
https://github.com/JohnChu/Snoopy/blob/master/jni/libgphoto2/Android.mk

Using the above build file an Android NDK I was able to build it successfully!

Now I am trying to bring it somehow to this repository.

laheller, 2019-06-24:
@msmeissn

Can u pls send me the list off **all** external libraries/dependencies of gphoto2/libgphoto2 ?
I mean like libusb, libjpeg, etc.

Thank you.

Ladislav

msmeissn, 2019-06-24:
libusb1 - mandatory for USB access
libtool libltdl - mandatory to load camera drivers

libjpeg - optional, only used by some drivers
libexif - optional, used for EXIF extraction purposes
libxml2 - optional, for Olympus E series
libgd - optional for picture frame support

laheller, 2019-07-11:
@msmeissn 

Porting to Android seems a little bit more complex, because of the libusb1 dependency and Android's security. The libusb1 library however **exists** for Android, but because of Android's security mechanism the user must grant permission for each connected USB device manually. When this permission is not granted, the application can't access the usb device at _/dev/bus/usb/*_.
There is a [libusb1 port](https://github.com/martinmarinov/rtl_tcp_andro-/blob/master/libusb/src/main/jni/README) for Android, which workarounds the above problem by providing 2 more functions:

- [libusb_get_device2( )](https://github.com/martinmarinov/rtl_tcp_andro-/blob/eb0b251c6f337f62529105ae12c1e5c4ef525222/libusb/src/main/jni/libusb/core.c#L1497)
- [libusb_open2( )](https://github.com/martinmarinov/rtl_tcp_andro-/blob/eb0b251c6f337f62529105ae12c1e5c4ef525222/libusb/src/main/jni/libusb/core.c#L1272)

Once the permission on attached USB device was granted by user, the app can open the device, get the file descriptor and pass it to the libusb_open2( ) function.

So to summarize, I think to successfully port libgphoto2 to Android the USB-related source code might be slightly modified.

BR,

Ladislav

ndim, 2019-07-12:
Have you checked whether loading `*.so` files via `libltdl` works on Android? If not, you have a significant amount of rewriting to do, both for libgphoto2 loading camera drivers and libgphoto2_port loading the port driver.

laheller, 2019-07-12:
> Have you checked whether loading `*.so` files via `libltdl` works on Android? If not, you have a significant amount of rewriting to do, both for libgphoto2 loading camera drivers and libgphoto2_port loading the port driver.

@ndim 
Not yet, this is the next step.
But the libltdl is part of android source here, so I expect that it will work:
https://android.googlesource.com/toolchain/gcc/+/refs/heads/master/gcc-4.9/libjava/libltdl/

BR,

Ladislav

Nailik, 2019-07-13:
@laheller  
How far did you go yet?
I tried to communicate with a Sony Camera over USB with the default USB Host implementation but didn't got a good result yet so i'm very interested if it would work with a libgphoto2 android port.

lzyang2000, 2019-07-23:
> @ndim
> Not yet, this is the next step.
> But the libltdl is part of android source here, so I expect that it will work: [...]

hello,
has any progress been made on the porting? Trying to hook my Ricoh camera onto an android phone. On linux systems it seems that the ghphoto library is the most painless way and I have gotten it working, but I can't find much info about it on Android. Thanks!

laheller, 2019-08-01:
> hello,
> has any progress been made on the porting? Trying to hook my Ricoh camera onto an android phone. On linux systems it seems that the ghphoto library is the most painless way and I have gotten [...]

Fighting with libltdl, which is required...

lzyang2000, 2019-08-01:
> Fighting with libltdl, which is required...

Sad. I resorted to switching the wifi when connecting to the camera.
Wish you luck on your work!

laheller, 2019-09-04:
An intermediate status report:
I was able to build libltdl for Android.
Now I am going to try to modify libgphoto2 to be able to work with the above mentioned usb library.

laheller, 2019-10-14:
@ndim @msmeissn 

Hello

Guys, maybe a dumb question but just to be sure:
To get the **libgphoto2.so** shared library I need to put together and build **all** the *.c sorce files from this repository (except maybe the ones under examples folder) ?

BR,

Ladislav

ndim, 2019-10-14:
No.

Not having found such a description, I have added it to  [`HACKING.md`](https://github.com/gphoto/libgphoto2/blob/master/HACKING.md#source-repository-directory-structure).

It certainly is impossible to just link all camlibs and iolibs into one library. Every camlib uses the same camlib set of symbol names, and all iolibs use the iolib set of symbol names. The dynamic loading of the correct camlib and iolib will make sure that libgphoto2 and libgphoto2_port call the proper implementation.

laheller, 2019-10-15:
> No.
>
> Not having found such a description, I have added it to [`HACKING.md`](https://github.com/gphoto/libgphoto2/blob/master/HACKING.md#source-repository-directory-structure).
>
> It certainly is [...]

@ndim thanks, now it is clear for me the whole structure!

ndim, 2019-10-15:
@laheller  There should not have been significant changes to the architecture graphics at http://gphoto.github.io/doc/manual/developer-docs.html

laheller, 2019-10-29:
@ndim @msmeissn 

Slowly, but it goes well. Now I was able to build **libgphoto2-port** library.
What is sometimes really hard to solve is the autoconf/automake which is not supported in Android NDK build system, so I need to manually create/edit config.h headers for some libraries to build them successfully.

This problem however can be [workarounded](https://gist.github.com/nddrylliog/4688209) in some but not all cases. For example this way I was able to **automatically** create config.h for _ltdl_, _usb1_, _usb-compat_, but it did not work for _gphoto2-port_.

I'll keep you updated on any progress/issues.

laheller, 2019-12-09:
Update

As I told above, it goes slowly. Currently I am fighting with camera libraries.

daedeloth, 2020-01-16:
Going to follow this as well :-)

kiwileaks78, 2020-01-25:
Very interested too :)

Morreski, 2020-01-31:
Hi, Very interested by the android port. Will follow this with great interest.

thebino, 2020-02-04:
@laheller Could you share your fork or project? I started building my own android integration und would like to merge our efforts.
You can find mine here: https://github.com/thebino/libgphoto2android

laheller, 2020-02-05:
> 
> 
> @laheller Could you share your fork or project? I started building my own android integration und would like to merge our efforts.
> You can find mine here: https://github.com/thebino/libgphoto2android

Once I finish the camlibs, I will publish/share everything.
Unfortunately I am working on another high prio project therefore I have less time for libgphoto2.

thebino, 2020-02-05:
@laheller Okay, I'm also fine with a work-in-progress to help with the cam libs.
I also invited you already to my project if this helps

DiegoBM, 2020-04-15:
Any new updates? Thank you for the great work!

laheller, 2020-04-22:
> Any new updates? Thank you for the great work!

Temporarily stopped due to coronavirus and some personal things.

But in the meantime a new library for Android developers was released, which is available at:
https://github.com/rupiapps/ptplibrary

I asked the author and he states it is compatible with the most Canon, Nikon and Sony cameras.

DiegoBM, 2020-04-22:
Thank you for the information @laheller, that's a fantastic hint. I hope you (all of you in fact) are staying safe

develop-mbb, 2020-06-09:
link is not working
https://github.com/rupiapps/ptplibrary

hfiguiere, 2020-06-09:
it seems to have moved there:
https://github.com/laheller/ptplibrary

develop-mbb, 2020-06-09:
I would like to control my Canon DSLR over the webbrowser from everywhere.
I plan to connect the DSLR over USB to a android smartphone. Smartphone should connect to the internet over LTE.
Which existing project would be the best for my setup?

Nailik, 2020-06-09:
@develop-mbb maybe teamviewer and just remote control the smartphone would be the easiest way.

laheller, 2020-06-10:
> I would like to control my Canon DSLR over the webbrowser from everywhere.
> I plan to connect the DSLR over USB to a android smartphone. Smartphone should connect to the internet over LTE.
> Which existing project would be the best for my setup?

I would buy a Raspberry Pi 4, install Raspbian, TeamViewer and gphoto2 on it. Then connect DSLR via USB cable to RPI. Finally remote control RPI over TeamViewer. Easiest solution.

develop-mbb, 2020-06-10:
> I would buy a Raspberry Pi 4, install Raspbian, TeamViewer and gphoto2 on it. Then connect DSLR via USB cable to RPI. Finally remote control RPI over TeamViewer. Easiest solution.

Do you have a suggestion for a LTE-module for the Raspberry Pi 4? The performance of the internet connection will be important, because I want to transfer all the Photo-files in RAW-format.

daedeloth, 2020-06-10:
I have built a rather simple shell that exposes a photo:takePicture over a websocket connection, I built it to achieve a similar setup (raspberry pi connected to DSLR) to create a photobooth.

There isn't much documentation as I've hacked the code together over a few days, but if you have experience with nodejs I bet you can get it up and running in a couple of hours.

Nodejs websocket - gphoto2 listener
https://github.com/CatLabInteractive/gphoto2-socketio

Photobooth app (php - laravel)
https://github.com/CatLabInteractive/catlab-photobooth

Photobooth online version:
http://photobooth.catlab.eu/photobooth (I was using an NFC scanner to link photos to attendees)

laheller, 2020-06-11:
> Do you have a suggestion for a LTE-module for the Raspberry Pi 4? The performance of the internet connection will be important, because I want to transfer all the Photo-files in RAW-format.

Why an LTE module? Share your smartphone's internet connection and that's it.

develop-mbb, 2020-06-11:
> Why an LTE module? Share your smartphone's internet connection and that's it.

The DSLR should be at a different place than the controller. At the location of the camera there will be no WLAN, only LTE/3G.

I think I found a good existing solution which I could upgrade to my needs.

Check out the following link to GPhoto2 Server:
https://github.com/mvmn/gp2srv

laheller, 2020-09-20:
Hi!

Update after a long time ago:
I was able to build everything for Android (for the following ABIs: arm, aarch64), except some camera libraries.
In the meantime I figured out that we don't need to write any special Android build scripts (MK files), becaues all libgphoto2 related projects, including dependencies like libltdl, libusb are GNU Autotools project and their configure script can be relatively easily re-used for Android build. More info here:
https://developer.android.com/ndk/guides/other_build_systems#autoconf

Now I am thinking about a magic script which:
1. Installs all possibly required packages, including android-ndk for Linux (Ubuntu)
2. Clones all required git repositories
3. Creates the required script in each repository which calls the corresponding autotools/configure script
4. Starts build for each repository (libtool/libltdl first, libusb next, etc.)
5. Collects all the generated shared library (.so files) and moves them into one output folder.

Any ideas are appreciated for better automation of the above.

huangmingyou, 2020-09-23:
> Hi!
> 
> Update after a long time:
> I was able to build everything for Android (for the following ABIs: arm, aarch64), except some camera libraries.
> In the meantime I figured out that we don't need to write any special Android build scripts (MK files), becaues all libgphoto2 related projects, including dependencies like libltdl, libusb are GNU Autotools project and their configure script can be relatively easily re-used for Android build. More info here:
> https://developer.android.com/ndk/guides/other_build_systems#autoconf
> 
> Now I am thinking about a magic script which:
> 
> 1. Installs all possibly required packages, including android-ndk for Linux (Ubuntu)
> 2. Clones all required git repositories
> 3. Creates the required script in each repository which calls the corresponding autotools/configure script
> 4. Starts build for each repository (libtool/libltdl first, libusb next, etc.)
> 5. Collects all the generated shared library (.so files) and moves them into one output folder.
> 
> Any ideas are appreciated for better automation of the above.

can share some documents.

websocketing, 2021-04-16:
> 你好！
> 
> 经过很长时间的更新：
> 除了一些相机库之外，我能够为Android（针对以下ABI：arm，aarch64）构建所有内容。
> 同时，我发现我们不需要编写任何特殊的Android构建脚本（MK文件），因为所有与libgphoto2相关的项目（包括libltdl，libusb等依赖项）都是GNU Autotools项目，并且它们的configure脚本可以相对轻松地重新用于Android构建。此处提供更多信息：[https](https://developer.android.com/ndk/guides/other_build_systems#autoconf) :
> [//developer.android.com/ndk/guides/other_build_systems#autoconf](https://developer.android.com/ndk/guides/other_build_systems#autoconf)
> 
> 现在，我正在考虑一个魔术脚本，其中：
> 
> 1. 安装所有可能需要的软件包，包括适用于Linux的android-ndk（Ubuntu）
> 2. 克隆所有必需的git存储库
> 3. 在每个存储库中创建所需的脚本，该脚本调用相应的autotools / configure脚本
> 4. 为每个存储库启动构建（首先为libtool / libltdl，其次为libusb，等等）
> 5. 收集所有生成的共享库（.so文件），并将它们移动到一个输出文件夹中。
> 
> 为了更好地实现上述目的，任何想法都应受到赞赏。

Is Android running normally now? Can we share open source materials?

thebino, 2021-04-16:
@websocketing Your "magic" script could be solved by a cmake script running inside a github workflow. I've done it years ago but sadly not saved the results. Some parts can still be found in my github archive.

websocketing, 2021-04-17:
> @websocketing您的“魔术”脚本可以通过在github工作流中运行的cmake脚本来解决。我几年前就做过，但可惜没有保存结果。某些部分仍然可以在我的github档案中找到。

You're right https://github.com/thebino/libgphoto2android Unfortunately, there is no way to compile this open source. I hope you can give me some help. Thank you

websocketing, 2021-04-17:
> @websocketing Your "magic" script could be solved by a cmake script running inside a github workflow. I've done it years ago but sadly not saved the results. Some parts can still be found in my github archive.

Can your Facebook tell me, I hope I can communicate with you online, if there are other contact information

laheller, 2021-05-22:
Sharing some screenshots from the first tests after successful build of gphoto2 and libgphoto2 for Android from sources using Android NDK tools on Ubuntu.

Some info:
- gphoto2 has a lot of dependencies, therefore I had also build all of then (libltdl, libusb, libexif, etc)
- so far tested in Android Terminal app on my rooted Samsung Galaxy S20+ smartphone
- I had to configure LD_LIBRARY_PATH, CAMLIBS and IOLIBS environment variables
- my Canon EOS 1300D was connected to the smartphone via USB cable

[image]

[image]

[image]

laheller, 2021-07-17:
Hi community!

Almost 3 years ago I opened this issue, now after a lot of learning and experimenting I can tell that build libgphoto2 for Android is possible, but not the easisest thing. It has a lot of dependencies and of course we need to build all of them before even start to build libgphoto2.

Since the minimal required dependencies for libgphoto2 are among others:
- libltdl
- libusb1.0
- libcurl
- libxml2
- libjpeg
- libgd
- libexif,

I am not going to describe the whole process, because this is not the right place for that. Instead I will provide a sample, how to build the first dependency - libltdl - from sources for Android.

First of all we need to install some required packages, which on Ubuntu can be done by:
`sudo apt install git make google-android-ndk-installer libtool help2man texi2html autoconf automake texinfo`

The next steps are similar to when we build an autoconf project from sources on Ubuntu:

- run the _configure_ script -> checks the system, generates input for build, here is the main difference on Android!
- _make_ -> builds the project
- _make install_ -> installs the project

Probably the easisest is when we put everything together in a single shell script and just run it:
```
#! /bin/sh

# in this example we are going to clone and build libtool/libltdl for Android
# this is the most important dependency for libgphoto2 and libgphoto2_port
REPO_NAME=libtool
REPO_URL=https://git.savannah.gnu.org/git/libtool.git

git clone --recursive ${REPO_URL} ${REPO_NAME}
cd ./${REPO_NAME}

# needed for some autoconf projects, might not work for some others
if [ -f ./bootstrap ]; then
    ./bootstrap
else
    autoreconf -i
fi

# build will happen in this dir
mkdir _build
cd _build

# by default the Android NDK root is at /usr/lib/android-ndk
export ANDROID_NDK=/usr/lib/android-ndk

# valid values: arm-linux-androideabi-4.9, aarch64-linux-android-4.9, x86-4.9, x86_64-4.9
export TOOLCHAIN=aarch64-linux-android-4.9

# valid values: arm-linux-androideabi, aarch64-linux-android, i686-linux-android, x86_64-linux-android
export CROSS_COMPILE=aarch64-linux-android

# valid value: android-XX, where XX is 9-24
export API=android-24

# valid values: arch-arm, arch-arm64, arch-x86, arch-x86_64
export ARCH=arch-arm64

export ANDROID_PREFIX=${ANDROID_NDK}/toolchains/${TOOLCHAIN}/prebuilt/linux-x86_64
export SYSROOT=${ANDROID_NDK}/platforms/${API}/${ARCH}
export CROSS_PATH=${ANDROID_PREFIX}/bin/${CROSS_COMPILE}

# settings for compiler
export AR=${CROSS_PATH}-ar
export AS=${CROSS_PATH}-as
export CC=${CROSS_PATH}-gcc
export CPP=${CROSS_PATH}-cpp
export CXX=${CROSS_PATH}-g++
export LD=${CROSS_PATH}-ld
export NM=${CROSS_PATH}-nm
export OBJDUMP=${CROSS_PATH}-objdump
export RANLIB=${CROSS_PATH}-ranlib
export READELF=${CROSS_PATH}-readelf
export STRIP=${CROSS_PATH}-strip

# set whatever target dir for installation, but do NOT set to /usr
export PREFIX=$(pwd)/../_deploy

# additional settings for compiler
# we can also specify additional compiler flags
export CFLAGS="${CFLAGS} --sysroot=${SYSROOT} -I${SYSROOT}/usr/include -fPIE -std=c99"
export CPPFLAGS="${CFLAGS}"

# we can also specify additional linker flags
export LDFLAGS="${LDFLAGS} --sysroot=${SYSROOT} -L${SYSROOT}/usr/lib -pie"

# run "../configure --help" to see all available command line switches, sometimes they must be tweaked
# here is the main difference when we build a project from sources for Android platform
../configure --host=${CROSS_COMPILE} --with-sysroot=${SYSROOT} --prefix=$(pwd)/../_deploy

make

make install

echo "Architecture of the built library:"
readelf -a ${PREFIX}/lib/libltdl.so | grep Machine:

echo "Finished!"

```
The above script clones the source repository into _./libtool_ subfolder, switches to that folder, builds the library from sources and finally installs it to **./libtool/_deploy**. In this example it's built for Android aarch64 platform.

With no or only minimal changes on the above script we can build all dependencies and libgphoto2 itself. Mostly we need to change CFLAGS, LDFLAGS, add new enviroment variables and call the configure script with various switches to enable/disable some features for given library. Before the build running simply _./configre --help_ might give some more info.

Happy compiling/linking/installing!

@msmeissn @ndim - we can close this issue

BR,

Ladislav

fahrion2600, 2021-11-05:
Hi @laheller!

Thank you for your efforts! Using your most recent comment, I have finally been able to build libgphoto2 and gphoto2 for Android as well. It was indeed tricky, but your script provided the key (as I'm a newbie to NDK myself). For now, I'm building just libtool, libusb-1.0, libexif, and libpopt dependencies. The rest should hopefully be straightforward.

I hope you don't mind, but I expanded on your script and encapsulated it along with the cross-compiling tools into a container image. It makes it much easier for me to keep things clean on my dev machine. If anybody is interested, I uploaded the scripts here: https://github.com/fahrion2600/gphoto2-build-scripts/tree/main/android

Along the way, I found there was a problem in how the NDK compiler treats empty variadic arguments which affected the ptp2 driver. If anybody else hits it, I have a potential fix here: https://github.com/fahrion2600/libgphoto2/commit/914553d71f02515b3c12acdca760aef2c92d0591

I hope this helps someone!

laheller, 2021-11-09:
@fahrion2600 

Great job!
Unfortunately still the biggest problem is for Android its weird USB security. I already started to study the libgphoto2's source code but I am quite newbie to C language so for now have no idea, how to change it to make it properly work on Android (without root).
Anyway it is possible to do that, because the latest libusb1.0 fully supports the Android USB workflow (see the issue #683 ).

fahrion2600, 2021-11-11:
Agreed, I have not had success yet connecting via USB without root. I tried following the instructions [here](https://github.com/libusb/libusb/blob/master/android/README), such as setting LIBUSB_OPTION_NO_DEVICE_DISCOVERY before libusb_init. However, I suspect disabling device discovery prevents USB device listing, which libgphoto appears to rely on in several places.

I am keeping an eye on https://github.com/libusb/libusb/pull/874 which might be an alternative, as it supposedly supports device listing. However, it is not yet merged.

Phobetor, 2023-10-22:
@laheller 
> Unfortunately still the biggest problem is for Android its weird USB security.

Not sure if my question is out of scope: Is it possible to use WIFI to connect to the camera with this library on android?

laheller, 2023-10-22:
> @laheller 
> > Unfortunately still the biggest problem is for Android its weird USB security.
> 
> Not sure if my question is out of scope: Is it possible to use WIFI to connect to the camera with this library on android?

@Phobetor 
Yes, it should be possible.
But I would wait for the official build:
https://github.com/gphoto/libgphoto2/pull/918

laheller, 2023-10-24:
Official Android support now exists, see
https://github.com/gphoto/libgphoto2/pull/918

Closing this thread.

closed 2023-10-24 by laheller

akshayrgund2, 2025-03-05:
Hi @laheller,

Thank you for all your hard work on this project. I'm relatively new to both building libraries and Android development, so please forgive any rookie mistakes.

I'm encountering an issue integrating libgphoto2 into my Android app. Specifically, I've been trying to build libgphoto2 using the script below:

```

#!/bin/bash

# Set up environment variables
export TARGET="aarch64-linux-android"
export API="21"
export ANDROID_NDK_HOME="/opt/homebrew/share/android-ndk"
export TOOLCHAIN="$ANDROID_NDK_HOME/toolchains/llvm/prebuilt/darwin-x86_64"
export PATH="$TOOLCHAIN/bin:$PATH"
export AR="llvm-ar"
export CC="$TARGET$API-clang"
export CXX="$TARGET$API-clang++"
export LD="ld.lld"
export RANLIB="llvm-ranlib"
export STRIP="llvm-strip"

# Create build directory
mkdir -p ~/android-build
cd ~/android-build

# Build libtool
echo "Building libtool..."
wget https://ftp.gnu.org/gnu/libtool/libtool-2.4.7.tar.gz
tar -xvzf libtool-2.4.7.tar.gz
cd libtool-2.4.7
rm -rf install
mkdir install
./configure CFLAGS="-fPIC" --host=$TARGET --prefix=$PWD/install --enable-shared --disable-static --enable-ltdl-install
make
make install
cd ..

# Build libusb
echo "Building libusb..."
wget https://github.com/libusb/libusb/releases/download/v1.0.26/libusb-1.0.26.tar.bz2
tar -xvjf libusb-1.0.26.tar.bz2
cd libusb-1.0.26
rm -rf install
mkdir install
./configure CFLAGS="-fPIC" --host=$TARGET --prefix=$PWD/install --disable-shared --enable-static --disable-udev --disable-ltdl
make
make install
cd ..

# Build libxml2
echo "Building libxml2..."
wget https://download.gnome.org/sources/libxml2/2.13/libxml2-2.13.6.tar.xz
tar -xvf libxml2-2.13.6.tar.xz
cd libxml2-2.13.6
rm -rf install
mkdir install
./configure CFLAGS="-fPIC" --host=$TARGET --prefix=$PWD/install --enable-shared --disable-static --without-python --without-zlib --without-lzma
make
make install
cd ..

# Clone libgphoto2
echo "Cloning libgphoto2..."
git clone https://github.com/gphoto/libgphoto2.git
cd libgphoto2

# there  is install doc text, terminal confuses it with install folder
rm INSTALL

# Build libgphoto2_port
echo "Building libgphoto2_port..."
cd libgphoto2_port
rm -rf install

mkdir -p ~/android-build/libgphoto2/install
autoreconf -i
./configure CFLAGS="-fPIC" --host=$TARGET --prefix=$HOME/android-build/libgphoto2/install --enable-shared --disable-static \
    LTDLINCL="-I$HOME/android-build/libtool-2.4.7/install/include" \
    LIBLTDL="-L$HOME/android-build/libtool-2.4.7/install/lib -lltdl"
make
make install

# Build libgphoto2
echo "Building libgphoto2..."
rm -rf install
mkdir install
autoreconf -i
export PKG_CONFIG_PATH="$HOME/android-build/libusb-1.0.26/install/lib/pkgconfig:$HOME/android-build/libxml2-2.13.6/install/lib/pkgconfig:$HOME/android-build/libgphoto2/install/lib/pkgconfig"
./configure CFLAGS="-fPIC" --host=$TARGET --prefix=$PWD/install --enable-shared --disable-static \
    --with-libusb=$HOME/android-build/libusb-1.0.26/install \
    --with-ltdl=$HOME/android-build/libtool-2.4.7/install \
    --with-libgphoto2-port=$HOME/android-build/libgphoto2/install \
    LTDLINCL="-I$HOME/android-build/libtool-2.4.7/install/include" \
    LIBLTDL="-L$HOME/android-build/libtool-2.4.7/install/lib -lltdl" \
    --disable-libusb-1.0-compat \
    --disable-ltdl
make
make install

echo "Build completed successfully!"`

```

I have also tried the solution suggested by @fahrion2600 using a cross-compiling container image (see [here](https://github.com/fahrion2600/gphoto2-build-scripts/tree/main/android)). In both cases, the shared object (.so) files are successfully generated. However, when I include them in my Android project, I encounter the following error at runtime:

Could not load library — error code: -4

I suspect one of two possible issues:

The library might not be built correctly.
There may be a problem with how it's integrated into Android Studio.
I would greatly appreciate any insights or suggestions on how to resolve this issue. Please let me know if you need any additional details.

Thank you for your help!

laheller, 2025-03-05:
Hi @akshayrgund2 

Library load errors could cause the following things:

- The output .so file has an incorrect format. This one you can check using the `readelf` Linux tool:
`readelf -h <path_to_so_file> | grep Machine`
The output should be something like
`Machine: arm64`
or
`Machine: aarch64`
- As you mentioned, the integration into Android Studio is not correct. This is described, how to do correctly at the Android Developers site, google for it.
- The `LD_LIBRARY_PATH` environment variable might be set to the path where your .so files are located.
- The android app manifest XML should be configured correctly as well. I think the attribute [ExtractNativeLibs](https://developer.android.com/guide/topics/manifest/application-element#extractNativeLibs) should be set to true.

BR,

Ladislav