gphoto/libgphoto2 #336: Add Sony DSC-RX0

romainreignier, 2018-12-18:
I have tested the Sony DSC-RX0 camera with libgphoto2 in PTP mode and it works so I would like to add it to the known devices list.

I have added the USB Ids in the list, see the [commit](https://github.com/romainreignier/libgphoto2/commit/8ce4d8040b9a28148930ee14cc1cb715ef071a93) on my fork.

But I did not opened a pull request (yet), because, I am not sure of the capabilities to chose.

For know, I have only set `PTP_CAP|PTP_CAP_PREVIEW` because that is what most other cameras have. How to chose which capabilities to set?

The same for the description string, I have set `(PTP mode)` but other Sony cameras have `(Control)`, which one to chose?

Do I have to put some data in `camlibs/ptp2/cameras`?

I have attached the output of several commands (`lsusb`, `auto-detect`, `summary`, `list-all-config`).

Feel free to add the camera yourself if you prefer.

[lsusb.txt](https://github.com/gphoto/libgphoto2/files/2691545/lsusb.txt)
[gphoto_version.txt](https://github.com/gphoto/libgphoto2/files/2691546/gphoto_version.txt)
[gphoto_autodetect.txt](https://github.com/gphoto/libgphoto2/files/2691560/gphoto_autodetect.txt)
[gphoto_summary.txt](https://github.com/gphoto/libgphoto2/files/2691547/gphoto_summary.txt)
[summary-debug.txt](https://github.com/gphoto/libgphoto2/files/2691549/summary-debug.txt)
[gphoto_config.txt](https://github.com/gphoto/libgphoto2/files/2691550/gphoto_config.txt)

msmeissn, 2018-12-19:
the question is foremost, does it support capture  in this mode, and which mode is shown in the camera USB menu?

PTP_CAP is set when capture is supported.
PTP_CAP_PREVIEW is set when liveview / preview capture is supported by the camera.

The other Sonys have selections like "MTP", "PC ControL" "Mass Storage", from this kind of mode we derive the ones in brackets.

The cameras/ subdirectory is used for debugging purposes, its not included in the code.

romainreignier, 2018-12-19:
> PTP_CAP is set when capture is supported.
> PTP_CAP_PREVIEW is set when liveview / preview capture is supported by the camera.

Both functions works, but maybe more should be added.

> The other Sonys have selections like "MTP", "PC Control" "Mass Storage", from this kind of mode we derive the ones in brackets.

Indeed, the 3 modes are listed in the Camera Settings menu. So I should use `(PC Control)`.

>  The cameras/ subdirectory is used for debugging purposes, its not included in the code.

Ok, so do you want me to add a some files here?

msmeissn, 2018-12-19:
Yes please use (PC Control) for the pc control mode. 

You can also add the MTP mode one, but capture will not work in this mode likely, so leave the flags to 0.

You can create the cameras/ debug file on your own, usually I do that:
--summary outpout, serialnr blanked
--list-all-config output, serialnr blanked
--debug of --summary , extracting the DeviceInfo block so we see all ptp opcodes/etc.

romainreignier, 2018-12-19:
> You can also add the MTP mode one, but capture will not work in this mode likely, so leave the flags to 0.

I did not know that I could use the PTP in MTP mode.

I have added the MTP mode in [my fork](https://github.com/romainreignier/libgphoto2/commit/9971186c35e35e99eb61070bfa27c37e45b8b973).

closed 2018-12-27 by romainreignier