gphoto/libgphoto2 #340: Olympus OM-D E-M10 Mark III

techexo, 2018-12-31:
Hi everyone,

This issue is probably very related to https://github.com/gphoto/libgphoto2/issues/229, but as it is not the same model, I preferred opening a new issue.

I am in possession of the [Olympus OM-D E-M10 Mark III](https://www.getolympus.com/us/en/digitalcameras/omd/e-m10-mark-iii.html) and I would very much like to help adding support to this camera.

Is there a place where the action to make or the information to give you are listed? I am running Windows, but I can also use Linux if needed (WSL). Please tell me how I can help :smile:.

msmeissn, 2018-12-31:
It might work without specific listing. what we need to get it added is the USB vendor/product ID. 
also confirmation if it works , not sure if this is possible with WSL

techexo, 2019-01-01:
Hi @msmeissn. In _Normal_ mode, vendor ID is `07B4` and the product ID is `012E`. In _MTP_ mode, the vendor ID is `07B4` and the product ID is `012F`.

I confirm that I cannot make it work on WLS (no libusb support yet), but I would be happy to try it on a Debian Live, for example; if you can tell me what to test.

msmeissn, 2019-01-01:
can you run and capture output of:
gphoto2 --summary
gphoto2 --list-all-config

and also check if those work:
gphoto2 --capture-image-and-download
gphoto2 --capture-preview

techexo, 2019-01-01:
@msmeissn Here are the desired outputs: https://gist.github.com/techexo/641bf2a6c1494a7aa7e2810ab31a4275 . I also included the debug outputs for the last commands (capture), as they did not work out of the box.

They were all done in MTP mode. _Normal_ mode just leads to a `No camera found` error.

msmeissn, 2019-01-02:
Hmm. can you try with at least libgphoto2 2.5.19 ? (you currently seem to have 2.5.12)

techexo, 2019-01-03:
Hi. I am really frustrated because on a Debian Live testing, with gphoto2 2.5.20 and libgphoto2 2.5.21:
1. the camera is not supported in _Normal_ mode (`Error (-6: 'Unsupported operation')`)
2. the MTP mode is broken.

I suppose that the break of MTP is not due to libgphoto2, cause my file explorer doesn't detect the MTP "storage" either (which was done with Debian stable). I will try to fix that and come back with the desired command.

techexo, 2019-01-04:
Hi again. I finally compiled gphoto2 and libgphoto2 to avoid issues and make a sensible report.
In MTP mode, `gphoto2 --auto-detect` ouputs that my E-M10 is considered as an `Olympus E-M5`.

After launching `gvfs-mount -s gphoto2` to get rid of the `Could not lock the device issue`, I ran into the following.

1. `gphoto2 --summary` fails, with the following error:
> *** Error ***              
> PTP General Error
> 
> *** Error ***              
> An error occurred in the io-library ('Unspecified error'): No error description available
> *** Error (-1: 'Unspecified error') ***

[Debug file for summary](https://github.com/gphoto/libgphoto2/files/2725736/summary.txt)

2. `gphoto2 --list-all-config` fails with the same `PTP General Error` ([Debug file](https://github.com/gphoto/libgphoto2/files/2725739/listallconfig.txt)).

In the two cases, it seems that the issue comes from that:
```
8.018768 gp_port_write               (3): Writing 16 = 0x10 bytes to port...
16.019308 gp_port_write [gphoto2-port.c:406](0): Writing 16 = 0x10 bytes to port failed: Error writing to the port (-35)
16.019353 ptp_usb_sendreq [usb.c:109] (0): PTP_OC 0x1002 sending req failed: Error writing to the port (-35)
16.019384 camera_init [library.c:8254](0): 'ptp_opensession (params, sessionid)' failed: PTP General Error (0x2002)
16.019393 ptp                         (2): PTP: Opening session
16.019408 ptp_usb_sendreq             (2): Sending PTP_OC 0x1002 (Open session) (0x1) request...
16.019416 gp_port_write               (3): Writing 16 = 0x10 bytes to port...
24.019665 gp_port_write [gphoto2-port.c:406](0): Writing 16 = 0x10 bytes to port failed: Error writing to the port (-35)
24.019711 ptp_usb_sendreq [usb.c:109] (0): PTP_OC 0x1002 sending req failed: Error writing to the port (-35)
```

Further notes to finish: I used [this script](https://github.com/gonzalo/gphoto2-updater/blob/master/gphoto2-updater.sh) to compile and install gphoto2 and libgphoto2.
The following versions are consequently installed:
```
gphoto2         2.5.20         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.20         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking
```

techexo, 2019-01-10:
@msmeissn Anything I can do to help going forward with this issue? :smile:

msmeissn, 2019-01-10:
It cannot open communicaiton with the camera at all.

can you check if gvfs gphoto2 slaves are running and remove them?

techexo, 2019-01-14:
Indeed this doesn't exist on my camera ([manual](http://download.aws.olympus.eu/consumer/manuals/cameras/E-M10_Mark_III_MANUAL_EN.pdf)). Just for information, I would like to emphasize that the one in my possession is the **E-M10 Mark III**, not the **E-M5**. I thought that gphoto2 was using the MTP connection to do everything, I didn't get that there was supposed to be a special PC/camera mode.

I confirmed on various websites that the tethering capabilities are only for E-M1 and E-M5 cameras, not the E-M10. However, I reckon that it could be possible at least to access the pictures in MTP mode?

My main interest in gphoto2 supporting the E-M10 (restricted) capabilities was to detect and access _on-camera_ pictures with darktable, not doing capture and such. Do you think it could be possible?

techexo, 2019-01-26:
> My main interest in gphoto2 supporting the E-M10 (restricted) capabilities was to detect and access on-camera pictures with darktable, not doing capture and such. Do you think it could be possible?

@msmeissn Any idea about that ?

techexo, 2019-04-07:
@msmeissn I take the liberty to do a small ping on this issue regarding my last question: 
> My main interest in gphoto2 supporting the E-M10 (restricted) capabilities was to detect and access on-camera pictures with darktable, not doing capture and such. Do you think it could be possible?

If it's not possible even to only access on-camera pictures, then this issue can be closed.

msmeissn, 2019-04-08:
I looked over all the logfiles again and its not establishing contact with the camera in either...

To avoid any conflict with other programs using libgphoto2, can you not start the desktop and login over a textconsole and try running gphoto2 --sumary there ?

techexo, 2019-04-08:
Thanks @msmeissn for getting back on the issue.

Here are the logs, launched in a root textconsole with X killed (note that I am anyway on a minimalist Debian installation, without any file manager, automount capabilities and so on).
When the **E-M10 III** is put in Normal mode: [normalmode.txt](https://github.com/gphoto/libgphoto2/files/3053948/normalmode.txt)
When it is put in MTP mode: [mtpmode.txt](https://github.com/gphoto/libgphoto2/files/3053950/mtpmode.txt)

Versions `gphoto2 2.5.20`, `libgphoto2 2.5.22` & `libgphoto2_port 0.12.0` 

Having a quick look at the log, it seems that the camera is detected as an **E-M5** instead of an **E-M10**, and I am not sure they share their communications protocols? (at least they don't share the capabilities, as seen previously).

pluszakk, 2020-04-15:
Hi @techexo !

Have you solved this problem? I found the same problems connecting my camera (also Oly Em10m3).

Thanks in advance!

andyrtr, 2020-05-04:
I'm now having the same camera and have run into this issue here as well. I'm using shotwell and can use camera "mass storage" mode to import photos as a workaround. But getting the device to work in mtp would be much better. I'm running Arch Linux and may be able to compile recent git snapshots of required libraries.

What can I do to help here?

hfiguiere, 2020-05-04:
To import files from the camera, PTP/MTP doesn't provide any benefits.

DavidCussans, 2020-05-04:
A "me too" comment ( I have not been able to get Gphoto2 to work with my Olympus OM-D M10 mk3 ) but with a more recent version of Gphoto2 ( 2.5.23 on Fedora 31 )

Output of `gphoto2 --summary --debug` and `gphoto2 --list-all-config --debug`

[gphoto2_summary_error.txt](https://github.com/gphoto/libgphoto2/files/4577110/gphoto2_summary_error.txt)
[gphoto2_list-all-config_error.txt](https://github.com/gphoto/libgphoto2/files/4577111/gphoto2_list-all-config_error.txt)

tecunited, 2020-06-02:
I am using the latest libgphoto2 & gphoto2 installed from Github; I just need to Capture functionally, is there any workaround or something to do to make it work?

```
gphoto2         2.5.23.1       gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.25.1       standard camlibs (SKIPPING lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

```
```
Camera summary:                                                                
Manufacturer: OLYMPUS
Model: E-M5MarkIII
  Version: 1.00
  Serial Number: BJ9A03134
Vendor Extension ID: 0xfffd (1.0)

Capture Formats: MS Wave Unknown(b980) Defined Type JPEG Unknown(b101)
Display Formats: Undefined Type, Association/Directory, DPOF, MS Wave, Unknown(b980), Defined Type, JPEG, Unknown(b101)

Device Capabilities:
	File Download, File Deletion, No File Upload
	No Image Capture, No Open Capture, No vendor specific capture

Storage Devices Summary:
store_00010001:
	StorageDescription: None
	VolumeLabel: None
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Digital Camera Layout (DCIM)
	Access Capability: Read-Write
	Maximum Capability: 15923150848 (15185 MB)
	Free Space (Bytes): 12224430080 (11658 MB)
	Free Space (Images): -1

Device Property Summary:
Battery Level(0x5001):(read only) (type=0x2) Range [1 - 100, step 1] value: 100% (100)
Date & Time(0x5011):(readwrite) (type=0xffff) '20200602T210608'
Property 0xd405: error 201b on query.
Property 0xd406:(readwrite) (type=0xffff) 'Windows/6.0.5330.0 MTPClassDriver/6.0.5330.0'
Property 0xd407:(read only) (type=0x6) 1
```

yankeppey, 2026-10-04:
Hello, just to add my data point:
1. Import works, E-M10 Mark IV, firmware 1.00, 07b4:012f, USB Mode = MTP. Interface is plain PTP (class 6 / subclass 1 / protocol 1, PIMA 15470)
2. Capture does not work, and is not supposed to work:
- https://explore.omsystem.com/c/en/om-d-e-m10-mark-iv - the specs say "Tethering shooting: No"
- the camera's own DeviceInfo: 20 operations, all standard PTP 0x1001–0x101b plus MTP object-property ops 0x9801/9802/9803/9805. No 0x100E InitiateCapture, no 0x101C InitiateOpenCapture, no Olympus 0x94xx.
3. The camera is misidentified as Olympus E-M5, but that's probably fine. 07b4:012f is shared with the E-M5, E-PL5 and SP-720UZ entries, the behavior seems to be the same.