gphoto/libgphoto2 #341: Is there a persistent way of selecting a PTP camera?

romainreignier, 2019-01-02:
I have several USB PTP cameras connected to a computer and each camera is used by a separate process.

Is there a persistent way of selecting which camera to open?
Until now I use the USB *port* but it changes every time I plug the camera.
So I open every camera and read the serial number but it is not very convenient.

msmeissn, 2019-01-02:
this would be also my recommended method ...
it is not really possible to find out otherwise

msmeissn, 2019-01-02:
to elaborate ... bus position is hard in usb, the hub are in nonpredictable order and the devices on the hubs are enumerated as they come usually.

so it is not detectable from usb:XXX,YYY 

the only good way is using serial numbers :/

romainreignier, 2019-01-02:
Thanks for your quick answer.

> the only good way is using serial numbers :/

Ok, thanks.

It seems that serial number reported by `lsusb -v` is different than the one queried by ligphoto2 through PTP so I have to open all detected cameras in each process at startup causing race conditions.

Did you ever heard of any way to convert from one serial number to the other?

msmeissn, 2019-01-02:
i so far did not investigate their relation. the usb protocol level one might be easier to access... hmm, so far I am not exposing it over the api

romainreignier, 2019-01-02:
Sorry, I am not familiar with low-level USB, what do you mean by
> usb protocol level one might be easier to access