gphoto/libgphoto2 #342: 300 ms spent in gp_camera_wait_for_event() with 0 ms timeout

romainreignier, 2019-01-03:
I am in the same situation than #300 
I am capturing preview with `gp_camera_capture_preview()` in a thread every 100 ms and in another thread, I have to take pictures.
This was working well with a fluid liveview only slowed down during capture which is normal.
But I have decided to switch from capturing pictures from ` gp_camera_capture()` to `gp_camera_trigger_capture()` and a third thread polling the events to retrieve several files in case of JPEG+RAW and detect camera disconnection as advised in #306 
To avoid concurrent access to the camera, my `Camera` object is of course protected by a mutex and I have set the `timeout` argument of `gp_camera_wait_for_event()` to 0 to let the preview thread running.
But I have noticed a really slow liveview. I have timed the operations and if there is no new event, the `gp_camera_wait_for_event()` function takes 310 ms (only 10 ms if an event occurred).
I have found that this 300 ms comes from `PTP2_FAST_TIMEOUT` (= 150) being called twice in `ptp_usb_event()`.

Do you think of any solution to check the events while producing a liveview at a decent rate (at least 5 Hz) with the current libgphoto2?

I am using a Sony DSC-RX0 camera.

msmeissn, 2019-01-03:
can you capture a debug trace of this to see where exactly it happens?
we do some additional work beside checking for events, perhaps this can be opmtized.

romainreignier, 2019-01-03:
Of course. This is the output of my application for an iteration of the wait for events loop using `gp_log_add_func(GP_LOG_DATA, errordumper, nullptr);`.

[events_loop_log.txt](https://github.com/gphoto/libgphoto2/files/2723416/events_loop_log.txt)

As a side question, is there a mean to print logs with an environment variable, without registering a log handler?

romainreignier, 2019-01-03:
As you can see, all the time is wasted in the last lines:

```
             d_rx0_1: [2.384037470] Current port timeout is 20000 milliseconds.
             d_rx0_1: [2.384043232] Setting port timeout to 150 milliseconds.
             d_rx0_1: [2.384049094] Reading 24 = 0x18 bytes from interrupt endpoint...
             d_rx0_1: [2.534544565] Reading 24 = 0x18 bytes from interrupt endpoint...
             d_rx0_1: [2.685003218] Setting port timeout to 20000 milliseconds.
```

romainreignier, 2019-01-03:
If by debug trace, you mean the callstack:

```
1  ptp_usb_event              usb.c            569  0x7fffd8ad03a7
2  ptp_usb_event_check        usb.c            637  0x7fffd8ad07a4
3  ptp_check_event            ptp.c            2858 0x7fffd8aa32d8
4  camera_wait_for_event      library.c        5884 0x7fffd8ac0367
5  gp_camera_wait_for_event   gphoto2-camera.c 1449 0x7ffff68d9fa3
```

msmeissn, 2019-01-03:
I added a patch to git that does not use check_event when there are small timeouts.

can you test it?

romainreignier, 2019-01-03:
Sure. I was trying to propagate the 0 timeout to the libusb and it was working but was a dirty patch.

romainreignier, 2019-01-03:
Thanks for making a patch so quickly!

I have tested it and the result is very good.

But I have noticed that once in two, the result is twice as long because `PTP_OC 0x9209` is read twice. Is this normal?

```
d_rx0_1: [47.448355106] waiting for events timeout 0 ms
d_rx0_1: [47.448404639] Sending PTP_OC 0x9209 (PTP_OC_SONY_GetAllDevicePropData) request...
d_rx0_1: [47.448507709] Reading PTP_OC 0x9209 (PTP_OC_SONY_GetAllDevicePropData) data...
d_rx0_1: [47.452453206] Reading PTP_OC 0x9209 (PTP_OC_SONY_GetAllDevicePropData) response...
d_rx0_1: [47.452821215] DEBUG== 0xd215 after capture = 0
d_rx0_1: [47.452832842] Current port timeout is 20000 milliseconds.
d_rx0_1: [47.452858104] Setting port timeout to 0 milliseconds.
d_rx0_1: [47.452862641] Setting port timeout to 20000 milliseconds.
d_rx0_1: [47.452890088] Reading PTP event failed: Timeout reading from or writing to the port (-10)
d_rx0_1: Events loop in: 4 ms
d_rx0_1: [48.453123688] waiting for events timeout 0 ms
d_rx0_1: [48.453170951] Sending PTP_OC 0x9209 (PTP_OC_SONY_GetAllDevicePropData) request...
d_rx0_1: [48.453423627] Reading PTP_OC 0x9209 (PTP_OC_SONY_GetAllDevicePropData) data...
d_rx0_1: [48.457757252] Reading PTP_OC 0x9209 (PTP_OC_SONY_GetAllDevicePropData) response...
d_rx0_1: [48.458245676] Sending PTP_OC 0x9209 (PTP_OC_SONY_GetAllDevicePropData) request...
d_rx0_1: [48.458423143] Reading PTP_OC 0x9209 (PTP_OC_SONY_GetAllDevicePropData) data...
d_rx0_1: [48.465158880] Reading PTP_OC 0x9209 (PTP_OC_SONY_GetAllDevicePropData) response...
d_rx0_1: [48.465509506] DEBUG== 0xd215 after capture = 0
d_rx0_1: [48.465521242] Current port timeout is 20000 milliseconds.
d_rx0_1: [48.465525580] Setting port timeout to 0 milliseconds.
d_rx0_1: [48.465550875] Setting port timeout to 20000 milliseconds.
d_rx0_1: [48.465556605] Reading PTP event failed: Timeout reading from or writing to the port (-10)
d_rx0_1: Events loop in: 12 ms
d_rx0_1: [49.465811044] waiting for events timeout 0 ms
d_rx0_1: [49.465858340] Sending PTP_OC 0x9209 (PTP_OC_SONY_GetAllDevicePropData) request...
d_rx0_1: [49.465940591] Reading PTP_OC 0x9209 (PTP_OC_SONY_GetAllDevicePropData) data...
d_rx0_1: [49.472167431] Reading PTP_OC 0x9209 (PTP_OC_SONY_GetAllDevicePropData) response...
d_rx0_1: [49.472559073] DEBUG== 0xd215 after capture = 0
d_rx0_1: [49.472570842] Current port timeout is 20000 milliseconds.
d_rx0_1: [49.472596133] Setting port timeout to 0 milliseconds.
d_rx0_1: [49.472600664] Setting port timeout to 20000 milliseconds.
d_rx0_1: [49.472628025] Reading PTP event failed: Timeout reading from or writing to the port (-10)
d_rx0_1: Events loop in: 6 ms
d_rx0_1: [50.472862077] waiting for events timeout 0 ms
d_rx0_1: [50.472909789] Sending PTP_OC 0x9209 (PTP_OC_SONY_GetAllDevicePropData) request...
d_rx0_1: [50.473043620] Reading PTP_OC 0x9209 (PTP_OC_SONY_GetAllDevicePropData) data...
d_rx0_1: [50.478859670] Reading PTP_OC 0x9209 (PTP_OC_SONY_GetAllDevicePropData) response...
d_rx0_1: [50.480110249] Sending PTP_OC 0x9209 (PTP_OC_SONY_GetAllDevicePropData) request...
d_rx0_1: [50.480363960] Reading PTP_OC 0x9209 (PTP_OC_SONY_GetAllDevicePropData) data...
d_rx0_1: [50.484452853] Reading PTP_OC 0x9209 (PTP_OC_SONY_GetAllDevicePropData) response...
d_rx0_1: [50.484792036] DEBUG== 0xd215 after capture = 0
d_rx0_1: [50.484803724] Current port timeout is 20000 milliseconds.
d_rx0_1: [50.484828861] Setting port timeout to 0 milliseconds.
d_rx0_1: [50.484834227] Setting port timeout to 20000 milliseconds.
d_rx0_1: [50.484861833] Reading PTP event failed: Timeout reading from or writing to the port (-10)
d_rx0_1: Events loop in: 12 ms
```

msmeissn, 2019-01-03:
I added a bit "timed out" checking to waiting_for_timeout helper, this should avoid the loop running twice I think.