gphoto/libgphoto2 #343: Sony A7M3 support

firemeteorxx, 2019-01-06:
I get a Sony A7M3 which does not work through libgphoto2 backend version 2.5.21(e.g. shotwell).

To the first order, it does not regonize the device even though the model was referenced by the v2.5.19 release note. I really doubt if this ID in the code base is correct:
> {"Sony:Alpha-A7III (Control)",  0x054c, 0x0994, PTP_CAP|PTP_CAP_PREVIEW},

From what I see, relevant devices are:
0x0994: USB charger
0x096f: Control
0x0c02: Mass Storage
0x0c03: MTP

After adding 0x0c03 into the device list and rebuild, libgphoto2 does not work any better with my device. Before my hack when it was recognized as a general MTP device, the Shotwell can list the photos thumbnails but once the list is complete the device disconnects and falls back to charger mode. It's not any better with the hack. Note the diagnose command like 'gphoto2 --summary' as suggested in the project page also put the device into disconnect. The output looks kind of weird so I have no idea if it was a sucess output.

According to issue #323, similar problem was observed on another Sony model and it was a regression. There was mention about 'single session' limitation in that issue so I tried to add PTP_DONT_CLOSE_SESSION flag to my hack but it does not seem to help.

I also tried to access the device in MTP mode through libmtp. I also need to manually add the device ID and rebuild. But for libmtp, it just works with the hack. There is no 'single session' issue here. I read from libmtp README.gz about the disconnect issue:

> Generic MTP/PTP disconnect misbehaviour: we have noticed that
  Windows Media Player apparently never close the session to an MTP
  device. There is a daemon in Windows that "hooks" the device
  by opening a PTP session to any MTP device, whenever it is
  plugged in. This daemon proxies any subsequent transactions
  to/from the device and will never close the session, thus
  Windows simply does not close sessions at all.

>  For example this means that a device may work the first time
  you run some command-line example like "mtp-detect" while
  subsequent runs fail.
  
>  Typical sign of this illness: broken pipes on closing sessions,
  on the main transfer pipes(s) or the interrupt pipe:
  
  >  Closing session
    usb_clear_halt() on INTERRUPT endpoint: Broken pipe                                                                                                                                       
    OK.

However, I was actually able to run mtp-detect several times successfully. The output from mtp-detect is much better than 'gphoto2 --summary' so I also attach it here for reference.
[lsusb_a7m3_charge.txt](https://github.com/gphoto/libgphoto2/files/2730400/lsusb_a7m3_charge.txt)
[lsusb_a7m3_control.txt](https://github.com/gphoto/libgphoto2/files/2730402/lsusb_a7m3_control.txt)
[lsusb_a7m3_mass.txt](https://github.com/gphoto/libgphoto2/files/2730403/lsusb_a7m3_mass.txt)
[lsusb_a7m3_mtp.txt](https://github.com/gphoto/libgphoto2/files/2730404/lsusb_a7m3_mtp.txt)
[auto-detect.txt](https://github.com/gphoto/libgphoto2/files/2730405/auto-detect.txt)
[abilities.txt](https://github.com/gphoto/libgphoto2/files/2730406/abilities.txt)
[config.txt](https://github.com/gphoto/libgphoto2/files/2730407/config.txt)
[summary.txt](https://github.com/gphoto/libgphoto2/files/2730408/summary.txt)
[mtp-detect.txt](https://github.com/gphoto/libgphoto2/files/2730409/mtp-detect.txt)

msmeissn, 2019-01-06:
can you adjust the reconnect behaviour in the camera perhaps?
other sonys are not reporting it.

I adjusted the IDs in git.

firemeteorxx, 2019-01-07:
I can try if 'gphoto2 --reset' helps getting the connection back. 
If it works maybe we can fix the issue by applying model specific reset-on-close?
I'll report back later.

>perhaps its configurable in the camera menu

But I can't figure out which menu item could possibly affect the reconnect behavior.
There are three menus that appears to control the USB behavior: (type from memory as Camera not at hand)
1. Connection mode: Auto / Mass storage / MTP / Control 
2. Number LUN: one / Multi //Don't know what it means, I leave it as multi
3. USB power: On / Off

For the first bullet, I try to settle on MTP as i provides better organization (group by dates instead of a big pile of files in the same directory).

> as I did not get this reported by other Sony camera users

So do you think the issue #323 represents a different problem?
To be honest, I find the 'gphoto2 --summary' output from my attachment looks quite similar to the one quoted on the other issue. And the 'mtp-detect' output is much more verbose which makes me wonder if the action was unexpectedly interrupted somehow.
As a reference, the A7R3 model summary in the git is much more verbose (even though mostly for capture). Is PTP && control making so much difference?

msmeissn, 2019-01-08:
Control would be the one that can remote control the camera for capture.
MTP for downloading images
the issue 323 might be the same issue yes

firemeteorxx, 2019-01-08:
Dropping PTP_DONT_CLOSE_SESSION does not make the 'gphoto2 --summary --reset' work any better. The camera still disconnects while the program stall and report Error -7 finally.

BTW, probably not relevant, but it just come to my mind that when I was on old udev version, my camera used to automatically disconnect without any chance for me to operate. According to some comments around the web, it may have something to do with the MTP probe rule in udev. Somebody fixed his issue with a customized rule to disable the probe. In my case upgrading to latest udev version solved the automatic disconnect problem and I see the following line in the rule file:

> #Sony ILCE-A7M3
ATTR{idVendor}=="054c", ATTR{idProduct}=="0c03", SYMLINK+="libmtp-%k", MODE="660", GROUP="audio", ENV{ID_MTP_DEVICE}="1", ENV{ID_MEDIA_PLAYER}="1" 

So this essentially disabled the probe with a known device entry...

closed 2019-02-10 by msmeissn

msmeissn, 2019-02-10:
I removed a command call on close which caused it to swirtch to charger mode.... Need to reverse engineer the exact arguments there.