gphoto/libgphoto2 #347: Nikon Broken

under-score, 2019-01-11:
As reported already on thread [https://github.com/gphoto/libgphoto2/issues/218](https://github.com/gphoto/libgphoto2/issues/218) the most recent update of gphoto2 has broken also my second D4 leaving it in an unresponsive state after the first few shots. So I have to switch back to the built-in http server solution which is a pain too.

msmeissn, 2019-01-11:
can you try current libgphoto2 git? it might be transfering too big NEF files crashing the camera

under-score, 2019-01-11:
thank so much, really appreciate your response. I only download small jpegs but keep the big NEFs in camera. What I did is
`brew upgrade gphoto2`
which resulted in gphoto2 2.5.20. Tried to switch back but [brew](https://stackoverflow.com/questions/3987683/homebrew-install-specific-version-of-formula) did not find any earlier version.