gphoto/libgphoto2 #350: Nikon D3500 fails every other capture

Daniel-Reynolds, 2019-01-12:
Hi

Im trying to work with libgphoto2 and a Nikon D3500, it worked fine with a D3100 but now I've upgraded to the 3500 every other image is hanging and not taking. This is on a Raspberry Pi 3 running Raspbian Stretch. I have found multiple references to a problem like this in the github issues, but I haven't found a fix/workaround.

Whether I use gphoto2 cli or libgphoto2 through the python bindings it is capturing an image fine, and then when I try to capture another one it is pulsing the flash and then stalling. If I then interrupt the command, and 'sudo kill -9' the process then the subsequent capture will work again, until the next one which hangs alternatedly.

I have done this with the most stable release (2.5.22) and have cloned the git repo and compiled that this morning (attached dbug files are from todays git clone). I don't know if it's at all relevent, but when the capture works it seems to take 2 images, which mu previous camera did not.

I've attached the debug logs for one working capture and one failing one.

- note: I killed the failing capture after about 20secs or it would go on forever
- note: I removed tons of data from the middle of the working logfile, looked like the actual image data being recorded which pushed the file to 200MB and unable to link

[logfile-working.log](https://github.com/gphoto/libgphoto2/files/2752192/logfile-working.log)
[logfile-not_working.log](https://github.com/gphoto/libgphoto2/files/2752189/logfile-not_working.log)

Stepcam, 2019-03-12:
I have the same problem with the D3500. No problems with the D3200, D3300, D3400 with the same script.

Stepcam, 2020-02-26:
It look the last gphoto update fixe this bug

logock, 2020-03-25:
I'm unfortunately having the same issue with my D3500. I've used the gphoto2-updater script for installation of the current development branch:

gphoto2         2.5.23.1       gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.24.1       standard camlibs (SKIPPING lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

Please advise if there is the possiblity to workaround the issue somehow.
In my current python script I'm using something like this:

`    if call(["gphoto2","--capture-image-and-download","--filename","/mnt/storage/"+TIMESTAMP+".NEF"]) != 0:
        logging.error('Could not perform capture and download')`

However, the return code during the lockup is still 0, therefore my script cannot stop or handle the issue well.

gaphill, 2020-06-15:
Is there any news regarding this "every other" behaviour? It's still there.

I think there might be some changes in in behaviour with latest releases, so here is an  update how it works now. I tried to minimize the commands to bare minimum still revealing the errors. 

These are my software versions reported by `gphoto2 --version`. Platform is Arch Linux and I use packaged versions. To my eye they are the latest. Lens is the kit 18-55mm.
```
gphoto2         2.5.23         gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.25         standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

I start with camera on manual focus mode
```
% env LANG=C gphoto2 --get-config focusmode2 | grep Current
Current: MF (selection)
```
Then I run the first try of preview:
```
% env LANG=C gphoto2 --debug --debug-logfile=preview_first_.txt --force-overwrite --set-config-index focusmode2=0 --capture-preview

                                                                               
*** Error ***              
Nikon enable liveview failed (0x2019: PTP Device Busy)
*** Error (-110: 'I/O in progress') ***       
```

So it fails but was able to change the focus mode
```
% env LANG=C gphoto2 --get-config focusmode2 | grep Current
Current: AF-S
```
At this phase there is "Connected to smart device" at the camera screen.

The second preview with focus setting then works. (In principle setting focusmode2 is not needed at this phase, however I left it in place since it produces a weird error which may tell something to someone) 
```
% env LANG=C gphoto2 --debug --debug-logfile=preview_second_.txt --force-overwrite --set-config-index focusmode2=0 --capture-preview --set-config manualfocusdrive=-32767 --set-config manualfocusdrive=800
                                                                               
*** Error ***              
The property 'Focus Mode 2' / 0xd161 was not set (0x200f: PTP Access Denied)

*** Error ***              
Failed to set new configuration value 0 for configuration entry focusmode2.
Saving file as capture_preview.jpg
```
Now the camera screen is normal.

After this I am able to take pictures with `gphoto2 --set-config focusmode2=4 --force-overwrite --capture-image-and-download` endlessly.

Here are the logs from above

[preview_first_.txt](https://github.com/gphoto/libgphoto2/files/4779642/preview_first_.txt)
[preview_second_.txt](https://github.com/gphoto/libgphoto2/files/4779643/preview_second_.txt)

Reed920, 2021-06-04:
I have the same issue as mentioned above. Is this something I'm doing wrong, something gphoto can do or a problem on Nikon's end? Any updates on the progress of a fix for this?

RaingodSpires, 2022-08-07:
Try --trigger-capture command instead. It works on a D3500.