gphoto/libgphoto2 #351: Use PTP API of macOS (ImageCaptureCore) as backend

jannschu, 2019-01-13:
Hi,

as you might know macOS has its own implemetation of PTP (in the ImageCaptureCore framework), but it only covers basic features. The API though provides a way to send custom mesages.

I am interested in implementing this API as a PTP transport backend for libgphoto2 on macOS. The single advantage would be, that it is no longer necessary to kill the PTPCamera processes. According to the API documentation of Canon, this is also the way EDSDK is implemented.

What do you think? I would try to bridge the current libgphoto2 PTP API to ImageCaptureCore. Do you support this idea? How can I start?

msmeissn, 2019-01-13:
As begin,
You could hook it as a replacement for usb.c or ptpip.c in camlibs/ptp2/ , or as a ptp_transaction replacement , depending on how lowlevel the macos version goes.

jannschu, 2019-01-15:
A little update. I looked through the code. It seems to me a new port type (added to libgphoto2_port) for ImageCaptureCore (ICC) seems more appropiate. The reason is that in order to communicate with the camera with ICC one has to connect to the camera through the API. You can not specify a USB port or something like this. Then, in `camera_init` of `ptp2` one checks for the ICC port and sets up the low level IO function pointers to use ICC, which I would add in a new file, similar to ptp2/usb.c.

I will also enable the port by default on macOS and prefer it over USB if available. It seems the USB port can not be deactivated.

I am not sure if I made this clear in my issue: Are you interested in adding the code to libgphoto2? It will involve a few lines of Objective-C code.

hmeine, 2024-01-01:
I just wanted to file a similar issue, but found that this might be more or less a duplicate: I just wanted to add that on recent MacOS version, there is a `/usr/libexec/ptpcamerad` running through launchctl that **cannot be killed anymore** (it will automatically re-spawn, and at least with my limited launchctl knowledge, it did not seem to be stoppable while my camera was connected). Googling for workarounds, I found the explicit statement that one should access PTP cameras through IOKit, which is probably not only the cleaner solution, but nowadays the only available one.

msmeissn, 2024-01-02:
so far I have no access to a Mac , so its hard to do for me.

hfiguiere, 2024-01-02:
I don't think it's in scope for libgphoto2 anyway.