gphoto/libgphoto2 #352: 'ptp_canon_eos_resetuilock (params)' failed on camera.exit()

mihailtudoran, 2019-01-15:
Hi,

I'am using the swig python binding for ligphoto2 and I encounter this warning each time I do a camera.exit() ( gp_camera_exit ):

WARNING:gphoto2:(ptp_usb_getresp [usb.c:514]) PTP_OC 0x911c receiving resp failed: PTP Device Busy (0x2019)
WARNING:gphoto2:(camera_unprepare_canon_eos_capture [config.c:545]) 'ptp_canon_eos_resetuilock (params)' failed: PTP Device Busy (0x2019)
WARNING:gphoto2:(ptp_usb_getresp [usb.c:514]) PTP_OC 0x911c receiving resp failed: PTP Device Busy (0x2019)
WARNING:gphoto2:(camera_unprepare_canon_eos_capture [config.c:545]) 'ptp_canon_eos_resetuilock (params)' failed: PTP Device Busy (0x2019)
WARNING:gphoto2:(ptp_usb_getresp [usb.c:514]) PTP_OC 0x911c receiving resp failed: PTP Device Busy (0x2019)
WARNING:gphoto2:(camera_unprepare_canon_eos_capture [config.c:545]) 'ptp_canon_eos_resetuilock (params)' failed: PTP Device Busy (0x2019)
WARNING:gphoto2:(ptp_usb_getresp [usb.c:514]) PTP_OC 0x911c receiving resp failed: PTP Device Busy (0x2019)
WARNING:gphoto2:(camera_unprepare_canon_eos_capture [config.c:545]) 'ptp_canon_eos_resetuilock (params)' failed: PTP Device Busy (0x2019)

The main problem is that each time I do gp_camera_exit() and after that gp_camera_init() this warning doubles. For example if I exit() the camera 10 times, then the warning will show 10 times, this leads to a slow down when initializing the camera.

Normaly gp_camera_init() takes like 1 seconds, after doing exit()/init() 10 times, the init will take about 11 seconds.

I am looking for the correct way to exit the camera and avoid the warning ( if related ) and most important the slower init time.

Thanks!

mihailtudoran, 2019-01-15:
libgphoto2      2.5.22         all camlibs, gcc, ltdl, no EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

jim-easterbrook, 2020-02-25:
Is there anything I might be getting wrong in the Python interface that could cause these errors? The increasing slowness suggests there is a real problem here, not just over-verbose messages as I've tended to assume. I get the same messages with my Canon camera and libgphoto2 v2.5.18.

msmeissn, 2020-02-25:
This happens if the camera is in a state where it does not want to unlock for some reason.

like if it has some operations buffered still (images not yet downloaded) or similar.

This is hard to tackle :/ usually "Busy" is more a protocol operation flow issue, and not a bug

jim-easterbrook, 2020-02-25:
I've just noticed that I don't always get them, but always do if I've called `gp_camera_get_config` at some point. The Python interface is supposed to call `gp_widget_unref` when the Python object is destroyed, but this might not be happening as intended. It's also quite likely not to happen in a user program until after the camera has been exited.

msmeissn, 2020-02-25:
get-config enables the capture mode if not yet set, so it changes the state of the camera.
(the gp_widget can be left referenced, it has no camera state in itself.)

jim-easterbrook, 2020-02-25:
OK, nothing to do with CameraWidget freeing. Thanks.

jim-easterbrook, 2021-09-29:
> get-config enables the capture mode if not yet set, so it changes the state of the camera. (the gp_widget can be left referenced, it has no camera state in itself.)

I've noticed that when I get the `ptp_canon_eos_resetuilock` errors on camera exit the camera's LCD display still shows the computer icon. This can be cleared by a camera init then exit cycle. I assume this icon shows the camera is in "capture mode". Is there a way to unset "capture mode" before calling camera exit?

jim-easterbrook, 2021-09-29:
PS I've revisited this bug because I've got a python-gphoto2 user who's getting a `-1 Unspecified error` from camera_exit with a Canon 250D.
https://github.com/jim-easterbrook/python-gphoto2/issues/128

jim-easterbrook, 2021-10-04:
Just to prove this doesn't only happen with the Python interface, I wrote a short C program that demonstrates the problem (with my Canon 100D).
```
#include <stdio.h>
#include "gphoto2/gphoto2.h"
#include "gphoto2/gphoto2-version.h"

static void log_func(GPLogLevel level, const char *domain,
                     const char *str, void *data) {
    printf("log: %d: %s: %s\n", level, domain, str);
}

int test() {
    Camera* camera = NULL;
    CameraWidget* config = NULL;
    int error = GP_OK;
    int id = GP_OK;
    id = gp_log_add_func(GP_LOG_ERROR, log_func, NULL);
    if (id < 0) return id;
    error = gp_camera_new(&camera);
    if (error) return error;
    error = gp_camera_init(camera, NULL);
    if (error) return error;
    error = gp_camera_get_config(camera, &config, NULL);
    if (error) return error;
    error = gp_camera_exit(camera, NULL);
    if (error) return error;
    error = gp_camera_unref(camera);
    return error;
}

int main() {
    printf("libgphoto2 version %s\n", *gp_library_version(0));
    int result = test();
    printf("result: %d: %s\n", result, gp_result_as_string(result));
    return 0;
}
```

msmeissn, 2021-10-04:
so jim, your case was a bit different, there was a bug introduced during 2.5.27.1 development due to mismatched return type handling.

msmeissn, 2021-10-04:
that the ui does not switch back on is still there

jim-easterbrook, 2021-10-04:
I should have mentioned the `ptp_canon_eos_resetuilock` error occurs with 2.5.18 and 2.5.27 (and probably every one in between). I can't remember a time when it didn't happen.

msmeissn, 2021-10-04:
i think this debug message happens when there is no setuilock before.  (in camera_exit path it will not lead to an error, just a debuglog message).

jim-easterbrook, 2021-10-04:
The multiple occurrences and slowing down reported in the first post in this thread looks like a problem to me.  It also leaves the camera display showing the computer icon.

msmeissn, 2021-10-04:
i also fixed the appearance of this error now by tracking the uilocked state.

still after a config operation the camera shows the computer icon ... lets see how this goes away

jim-easterbrook, 2022-01-04:
My `ptp_canon_eos_resetuilock` problem is solved with the 2.5.28 release of libgphoto2.

msmeissn, 2022-01-04:
-> lets close

closed 2022-01-04 by msmeissn