gphoto/libgphoto2 #353: can't compile using gcc

pdeman, 2019-01-21:
Hi,

I have installed gphoto2 and libgphoto2-dev.

I want to compile the example sample-trigger-capture.c for example.
I tried:
`gcc -Wall -o sampleTrigger -lgphoto2 sample-trigger-capture.c`
and I get the error:

> error: GP_EVENT_FILE_CHANGED undecleared etc

how do I know if -lgphoto2 exists ? 
how can I solve this error ?
thanks

msmeissn, 2019-01-21:
This is a new enum only in the last releases of libgphoto2.

Is your system libgphoto2 up to date?

You could also remove this specific switch case from the example

pdeman, 2019-01-21:
when I do
`apt-get install libgphoto2-dev`
I have 
libgphoto2-dev is already the newest version (2.5.12-1)

msmeissn, 2019-01-21:
the new event GP_EVENT_FILE_CHANGED was added in 2.5.17 ... 
again, you can just remove it from the sample to build it ;)

pdeman, 2019-01-21:
ok thanks. it's compiling.
and sample trigger is running.
I have a lot of failed (see below) and the time between to capture is random but it's compiling.

> *** Contexterror ***              
> Canon EOS Half-Press failed (0x2019: PTP Device Busy)
> 'camera->functions->trigger_capture (camera, context)' failed: -110
> done triggering
> loop is at 1548085421.512523
> triggering capture 11
> PTP_OC 0x9128 receiving resp failed: PTP Device Busy (0x2019)
> 'ptp_canon_eos_remotereleaseon (params, 1, 0)' failed: Canon EOS Half-Press failed (0x2019: PTP Device Busy)
> Canon EOS Half-Press failed (0x2019: PTP Device Busy)

msmeissn, 2019-01-21:
which EOS is it?

pdeman, 2019-01-21:
it s a eos50D.

i am trying to maintain the canon in "half press" forcing the autofocus to run and do some capture time to time

msmeissn, 2019-01-22:
hmm, should even work in the older libphhoto2 then.

is it happening often, or only occasionaly?

pdeman, 2019-01-22:
quite a lot.
time between two shoots is then very variable.
[error_sample_trigger.txt](https://github.com/gphoto/libgphoto2/files/2783078/error_sample_trigger.txt)

pdeman, 2019-03-07:
Him I am trying to compile again, on a new setup.
I did the full install specify here: https://github.com/gphoto/libgphoto2/blob/master/INSTALL
and when I try to compile:
` gcc -W -o sampleTrigger sample-trigger-capture.c -L/usr/local/lib -lgphoto2 -lgphoto2_port`
I get this error:

> sample-trigger-capture.c: In function ‘wait_event_and_download’:
> sample-trigger-capture.c:49:16: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
>    if (timediff >= waittime) 
>                 ^
> sample-trigger-capture.c:127:13: warning: implicit declaration of function ‘lseek’ [-Wimplicit-function-declaration]
>    if (-1 == lseek(fd, 0, SEEK_SET))
>              ^
> sample-trigger-capture.c:129:13: warning: implicit declaration of function ‘write’ [-Wimplicit-function-declaration]
>    if (-1 == write (fd, buffer, size)) 
>              ^
> sample-trigger-capture.c:131:3: warning: implicit declaration of function ‘close’ [-Wimplicit-function-declaration]
>    close (fd);
>    ^
> **/tmp/ccIFRT8y.o: In function `main':
> sample-trigger-capture.c:(.text+0x720): undefined reference to `sample_create_context'
> collect2: error: ld returned 1 exit status**
>

msmeissn, 2019-03-09:
Use:
 gcc -O2 -Wall -o sampleTrigger sample-trigger-capture.c context.c config.c -L/usr/local/lib -lgphoto2 -lgphoto2_port

pdeman, 2019-03-19:
ok thanks.

closed 2019-03-19 by pdeman

cxd-nanning, 2019-05-29:
Hi all,
I cross-compile the example and download it to the ARM development board to run, but the compilation error:
root@ubuntu:/home/cxd/app/liu/gphoto/libgphoto2-2.5.4/examples# arm-hisiv300-linux-gcc sample-capture.c -o sample-capture2 -I/home/cxd/app/arm-install/include/ -L/home/cxd/app/arm-install/lib/
/tmp/ccVQeMBw.o: In function `capture_to_memory':
sample-capture.c:(.text+0xb4): undefined reference to `gp_camera_capture'
sample-capture.c:(.text+0xfc): undefined reference to `gp_file_new'
sample-capture.c:(.text+0x14c): undefined reference to `gp_camera_file_get'
sample-capture.c:(.text+0x170): undefined reference to `gp_file_get_data_and_size'
sample-capture.c:(.text+0x1a8): undefined reference to `gp_camera_file_delete'
/tmp/ccVQeMBw.o: In function `capture_to_file':
sample-capture.c:(.text+0x248): undefined reference to `gp_camera_capture'
sample-capture.c:(.text+0x2a8): undefined reference to `gp_file_new_from_fd'
sample-capture.c:(.text+0x2f8): undefined reference to `gp_camera_file_get'
sample-capture.c:(.text+0x340): undefined reference to `gp_camera_file_delete'
sample-capture.c:(.text+0x35c): undefined reference to `gp_file_free'
/tmp/ccVQeMBw.o: In function `main':
sample-capture.c:(.text+0x390): undefined reference to `sample_create_context'
sample-capture.c:(.text+0x3a4): undefined reference to `gp_log_add_func'
sample-capture.c:(.text+0x3b0): undefined reference to `gp_camera_new'
sample-capture.c:(.text+0x3c8): undefined reference to `gp_camera_init'
sample-capture.c:(.text+0x4a0): undefined reference to `gp_camera_exit'
collect2: error: ld returned 1 exit status

what should I do?
thanks!

msmeissn, 2019-05-29:
add 
-lgphoto2 -lgphoto2_port 
to the commandline.