gphoto/libgphoto2 #356: Sony Alpha 5100 controllable zoom

Raggygandalf, 2019-01-24:
Hi, 

i try to Control the zoom of the sony Alpha 5100.
i get the list with "list-config" and try out the property "d214" with the range Option. 
But Nothing happens. 
Is there no possibility to Change the zoom with gphoto2 ?

thx for help

msmeissn, 2019-02-12:
we do not know it yet ... it might be d214 ... hmm. I will need to find a sony with powerzoom

TripackMcLovin, 2020-12-23:
Hi Marcus,
Im trying to dig into libgphoto and scramble through the code, need it to run a RX100M4
Having also trouble to adjust the zoom-level:
With the summary query i can see d214:
Property 0xd214:(readwrite) (type=0x6) Range [0 - 4294967295, step 1] value: 32053248
and you guessed that it might be f-value by 1.000.000, which won't make sense since the f-value for the Mark IV starts at 24mm.

Problem: the value is not changing when changing zoom manually.
So I doubt the d214. But there is no other suitable candidate.

Also: all type=0x6 values won't stick at all.

Any recommendation on how to get into sonys ptp definitions? Is it maybe possible that they aren't supporting zoom Properties on this Model? Are the defitions/Keys consistent along different sony models? I try to avoid their http API but currently I have to make a design decission.

TripackMcLovin, 2020-12-26:
Hi Marcus,
Im trying to dig into libgphoto and scramble through the code, need it to run a RX100M4
Having also trouble to adjust the zoom-level:
With the summary query i can see d214:
Property 0xd214:(readwrite) (type=0x6) Range [0 - 4294967295, step 1] value: 32053248
and you guessed that it might be f-value by 1.000.000, which won't make sense since the f-value for the Mark IV starts at 24mm.

Problem: the value is not changing when change the zoom manually.
So I doubt the d214. But there is no other suitable candidate.

Also: all type=0x6 values won't stick at all.

Any recommendation on how to get into sonys ptp definitions? Is it maybe possible that they aren't supporting zoom Properties on this Model? Are the defitions/Keys consistent along different sony models? I try to avoid their http API but currently I have to make a design decission.

msmeissn, 2020-12-26:
i reversed most PTP sony properties, see camlibs/ptp2/ptp.h   

can you add a debuglogfile of 
gphoto2 --list-all-config --debug --debug-logfile=listall.log

attach listall.log here.

particular interested in these output lines:

prop 0x%04x, datatype 0x%04x, changemethod %d getset %d

(Sonys have a weird property setting ... some props can be set directly, some need to be set by a "stepping up / down " method.)

Currently "gphoto2 --set-config zoom=value" should do the stepping method.

TripackMcLovin, 2020-12-28:
[listall.log](https://github.com/gphoto/libgphoto2/files/5747344/listall.log)

Its a little outdated (2.5.23), 
[listall_set.log](https://github.com/gphoto/libgphoto2/files/5747424/listall_set.log)
don't state any datatype and changemethod

attached also the attempt to just set. (as done with "gphoto2 --set-config zoom=4000.0")

TripackMcLovin, 2021-01-08:
OK, tried to disassembled it and I think it fits not properly yet.

Property d214 (LSB "14 D2") has ptp type 6 (as in the hex dump) and the type of value don't seem to have consistent length (don't know where this is translated to Widget Type RANGE (3))
Have you a hint on how to disassemble the 25 byte inside the dump (OK, the first one is the type) or can I just the ptp reference?

14 d2 06 00 00-01 ff ff ff ff 00 18 e9
01 01 00 00 00 00 ff ff-ff ff 01 00 00 00 15 

Even in 2.5.23 is was treated as "Unknown DPC code" and the current version has the PTP_DPC_SONY_Zoom Property, there is this issue that the value of the Range is not changing at all even if it is changed the zoom manually.

Should I try to see if there are some official information in the vendor extension by Sony? The guessing is exhausting. Btw, their 0x00000011 is not appearing to be registered properly.

Thanks for the good work on that monster of a library!

msmeissn, 2021-01-10:
There is at this time no official documentation from Sony to my knowledge.

We try to handle the Zoom code with the "zoom" option as UINT32:

#define PTP_DPC_SONY_Zoom                               0xD214 /* might be focal length * 1.000.000 */

        { N_("Zoom"),                           "zoom",                     PTP_DPC_SONY_Zoom,                      PTP_VENDOR_SONY,    PTP_DTC_UINT32, _get_Sony_Zoom,                     _put_Sony_Zoom },

In 2.5.26 we have some more debug... At offset 4 of the block is the "change method" (which seems to be 00), meaning the incremental method.

Your set debuglog looks good, we do the expected things (SetControlDeviceB for the stepping method, then wait for change), and also a SetControlDeviceA in case this is a direct settable thing.

If the zoom does not change, we either need to do something else too, or its not possible.

Does the windows software do it? If yes, we can try to catch a USB trace of that.