gphoto/libgphoto2 #360: Canon Powershot SX540 HS Timout while capturing

Creat1veM1nd, 2019-01-27:
With the newest version of gphoto2 (2.5.20) and libgphoto2 (2.5.22) I finally managed to get my Canon Powershot SX540 HS recognized. The system is a Lubuntu 18

I called gphoto2 --capture-image
The camera driver is loaded but stops at 98.4%. Then a long time nothing happens. After some time I do get the message "Canon EOS M Full-Press failed" plus 0x02ff. 

After some time then also a error about a event with 0x2002

A picture is not taken and the camera hangs. I need to remove the battery to get the camera reacting again. 

As read somewhere I needed to unmount the camera first in order to get not I/O Error. 

I attached a debug file. I hope it helps. 

Thanks in advance!
[debug.txt](https://github.com/gphoto/libgphoto2/files/2800451/debug.txt)

Creat1veM1nd, 2019-01-30:
Same problem occurs when trying to set e.g. the ISO settings via --set-config. A --list-config or --get-config works.

msmeissn, 2019-02-02:
in both debug logs the communication with the camera seems already dead (it only gets timeouts)

Can you get one after a fresh attach?

I had reports that SX540HS works from someone else :/

Creat1veM1nd, 2019-02-04:
No problem, will do so - as soon as possible. 

I read some where that this camera works , too. That's why I am wondering why I am not able to get it working. 

Thanks for looking into that issue!

Creat1veM1nd, 2019-02-04:
Hopefully I did everything right: 

- System start
- plugged in my camera
- open terminal
- open filemanager
- turn on camera
- linux recognized camera and mounted it
- unmount camera in filemanager
- closed filemanager
- run  gphoto2 --debug --set-config /main/imgsettings/iso=2  2>&1 | tee gphoto.log

[gphoto.log](https://github.com/gphoto/libgphoto2/files/2829204/gphoto.log)

Creat1veM1nd, 2019-02-18:
Please let me know if I can do anything more to give you as much information as possible.

jam35L, 2019-02-19:
Encountered a similar issue with the following:
LibGphoto version 2.5.22:
"cameramodel": "Canon EOS M6",
"deviceversion": "3-15.0.1.0",
"manufacturer": "Canon Inc.",
"model": "67567616",
Changing date time works.

Problem: Changing camera caputre settings (aperture or shutterspeed) or imgage settings (iso) did not cause and error. But when an attempt to trigger the camera to take photo, it causes a camera error: "Canon EOS M Full-Press failed (0x2019: PTP Device Busy)"

Creat1veM1nd, 2019-03-08:
Any news about it ?

Creat1veM1nd, 2019-12-12:
I gave it a try again with the latest release, still the same. Any news on it ?

mechrekt, 2020-08-12:
> 
> 
> Same problem occurs when trying to set e.g. the ISO settings via --set-config. A --list-config or --get-config works.
> 
> I will reproduce it within the next days and post another debug file.

Hi, I have the same camera and same problems with gphoto2. Have you managed to solve the problem?

Creat1veM1nd, 2020-08-12:
@mechrekt : Sadly not. I gave up trying it since nothing helped and no one seems to have a solution or a hint.

mechrekt, 2020-08-12:
Thank you for your reply. The sad thing is that on gphoto web page the camera is declared as supported (Image Capture, Liveview, Configuration).
I purposely bought this camera knowing it was supported....now i hope i can make a return...
Have you tested any other cameras, so to be on the safe side?

Thank you

Creat1veM1nd, 2020-08-12:
I do not have any other camera which is supported, so sadly not other testing possible.

hfiguiere, 2020-08-12:
> Thank you for your reply. The sad thing is that on gphoto web page the camera is declared as supported (Image Capture, Liveview, Configuration).

It seemed to work until it didn't. It is users like you that report and try camera, and that's what the information is based. Sometime users also help debugging and the trial and error it takes. Unless Marcus has the camera to actually debug it, and even then it's not a done deal ; it remain a difficult task.

Remember: **none of the camera vendors help**. No doc, no support, not even hardware (loans). Marcus do this on his free time.

> I purposely bought this camera knowing it was supported....now i hope i can make a return...

Don't forget to tell the manufacturer why you return it. While bearing no hope, it surely has more impact than not telling anything. Given how the market / revenue is going downhill they might have to change.

mechrekt, 2020-08-12:
Ok, I understand, I agree with you.

If someone has some time to try I can give debug files if it helps. I've already sent debug file to gphoto-devel@lists.sourceforge.net hoping to do the right thing.

Thank you again guys.

msmeissn, 2020-08-12:
Sorry for losing track of this issue.

In the gphoto.log from @Creat1veM1nd the camera firmware is crashing:

1.381341 ptp_usb_senddata            (2): Sending PTP_OC 0x9110 (PTP_OC_CANON_EOS_SetDevicePropValueEx) data...

0000  18 00 00 00 02 00 10 91-18 00 00 00 0c 00 00 00  ................
0010  b0 d1 00 00 08 00 00 00-                         ........        

1.576747 camera_prepare_canon_eos_capture [config.c:458](0): 'ptp_canon_eos_setdevicepropvalue (params, PTP_DPC_CANON_EOS_EVFOutputDevice, &ct_val, PTP_DTC_UINT16)' failed: PTP I/O Error (0x02ff)

We can try to do adjustments to this code.

But in general, the new generation PowerShot SX series cameras ... while they support EOS style capture, it is very fragile and I have not been able to make all versions work. (e.g. my parents SX600HS does not work, my SX720HS works) ...

mechrekt, 2020-08-13:
Hi, thank you for your reply!
I bought this camera (Canon Powershot SX540 HS) on Amazon, so I can do a
return for a few more weeks. Please let me know if you intend to test and
if I can provide some logs and do some debug with you before returning it,
otherwise I will return it immediately and search for another camera which
is most definitely supported by gphoto.

Thank you very much for your availability and for your support!
Michele

Creat1veM1nd, 2020-08-13:
If needed I can try and give it another shot. Have to set up the whole thing again, so might take a while. But let me know.

msmeissn, 2020-08-13:
are you able to update libgphoto2 current git?

Creat1veM1nd, 2020-08-13:
Will try tonight, when off work. 

Anything particular I shall test then ?

mechrekt, 2020-08-13:
> 
> 
> are you able to update libgphoto2 current git?

Hi,
I've a raspberry pi fresh installed with raspbian and gphoto2 so I think everything is yet up to date. Please tell me if I can test something in particular and give you a feedback. Thank you!

msmeissn, 2020-08-13:
So I have a change spoecific to the SX540 HS  in current git

in camlibs/ptp2/config.c  
if (!strcmp(params->deviceinfo.Model,"Canon PowerShot SX540 HS")) mode = 0x11;

Sio you can try first the current libgphoto2 GIT as-is, if that does not change, adjust the above to "mode = 0x1;" perhaps

mechrekt, 2020-08-13:
>  
> Sio you can try first the current libgphoto2 GIT as-is, if that does not change, adjust the above to "mode = 0x1;" perhaps

Sorry I did not understand your last comment. I'm not so expert so if you need me to do something, please try to guide me step by step. Sorry and thank you again.

msmeissn, 2020-08-13:
I am afraid you really should be at the knowledge level of "git checkout and recompilation of libgphoto2", otherwise this kind of remote debugging does not make much sense. I currently do not have time to offer help on the git / recompilation parts :/

mechrekt, 2020-08-13:
No problem, please let me know if I can read from somewhere how to "git checkout and recompilation" (I like to learn).
Meanwhile I hope that also @Creat1veM1nd will be able to give us some feedback in time.
Thank you all

msmeissn, 2020-08-23:
very good to hear it works for capture.  does preview work too?

The hang is on closing, hard to say why this specific call hangs

Creat1veM1nd, 2020-08-31:
Live preview with Entangle is working
gphoto2 --capture-preview gives an image, too.