gphoto/libgphoto2 #367: The speed limit for capture-and-download-image? (Canon EOS 1300D)

zhzhang2018, 2019-02-02:
Hi,

I'm currently on a project where we try to use Raspberry Pi to control multiple cameras, and take pictures of an object from different angles. We plan to at least reach the speed of 20 images per minute for each camera, but recent testing with gphoto2 made me wonder if reaching the 3 seconds per image speed is possible. It would be really helpful if we could get the judgement from the expert of this library. 

Our current approach is to use Python subprocess to tell cameras (connected with mini USB) to take pictures in parallel for each angle. We tried to use os.system() and subprocess.Popen() to execute command-line gphoto2 program with the flag "--capture-image-and-download", but the cameras spent from 7 seconds to 15 seconds for each picture from trigger to download. I've seen other issues where people find out there's a 3s wait time, but even if we remove that wait successfully, it won't be fast enough. We're curious what is the current speed limit of this command, and if there are other faster commands we should try with. 

Also, would it help if we decide to only trigger capture on each camera without downloading, and in the end download all images saved on the camera? Hopefully combining 20 separate downloads and deletes into one can reduce some overheads and make it faster on average. But again, we don't know about the inner workings. 
(We only settled on using --capture-and-download-image for now because other commands returned multiple errors, but we're happy to switch if --capture-and-download-image is not the most efficient command in our case.) 

I'm also attaching a debug log here:
https://1drv.ms/t/s!AqPJBzNcLtowbVgQraYTFKuGgyo
Hopefully this length (15s) is inflated because of the debug mode. I believe the time difference between line 1250720 and 1250721 is the manifestation of the 3-second-wait. (Is there a way to suppress the debug log from printing the entire RAW file in hex?)

Thanks a lot.

msmeissn, 2019-02-03:
4.858203 gp_file_copy                (2): Copying '' onto 'capt0000.cr2'...
7.857475 gp_filesystem_get_file_impl (2): LRU cache used for type 1!
11.840501 gp_camera_file_delete       (2): Deleting file 'capt0000.cr2' in folder '/'...

This is most of the time spent, and its not camera related, but this is the time it needs to save the file on your local storage.

msmeissn, 2019-02-03:
But it took at least 4.8 seconds at least for this raw image capture. that might not be able to be done faster in your setup

zhzhang2018, 2019-02-03:
Thanks for the quick reply. 

I did another test today. Ran with 2 cameras, using Python subprocess with shell=False to execute he debug command. Out of the 8 attempts of taking pictures (4 per camera), 5 failed due to "0x2019: PTP device busy" (which also happened when I tried to use debug mode in CLI, but didn't ever happen without using debug mode -- I really wonder why). Among the remaining 3 success cases:

-- 1st time both camera succeeded together. Each spent 13~14 seconds until gp_file_copy (with around 8 to 10 seconds spent in downloading the 0x80000 byte chunks).

-- 2nd time one camera succeeded while the other failed with the 0x2019 message. The successful camera took 10s until gp_file_copy, spending 5 seconds in downloading the 0x80000 chunks.

So I guess that might be the best that we can expect. It's worth noting that this time neither case spent too much time between gp_file_copy, gp_filesystem_get_file_impl, and gp_camera_file_delete. All were done within one second, unlike the log that I uploaded that took 3+4=7 seconds. 

We'll still experiment with other methods (such as capturing photos without downloading during the cycle, and downloading all the files together when it's done), or try to determine if images of a lesser quality (like jpeg) can still satisfy our goals.

zhzhang2018, 2019-02-08:
> But it took at least 4.8 seconds at least for this raw image capture. that might not be able to be done faster in your setup

Hi Markus,

Recently I have tried other methods:
1) Using --capture-image directly from terminal
2) Entering shell mode, and then use capture-image
3) Following the suggestion in [Issue #70 of gphoto2](https://github.com/gphoto/gphoto2/issues/70), by removing the wait time in libgphoto_2.5.22/gphoto2/main.c, as well as ignoring the foundevents variable in gphoto_2.5.20/camlib/ptp2/library.c, followed by re-compiling the entire library through running ./configure, running "make", and then "sudo make install" on first libgphoto2 and then gphoto2. 

It turned out that all three methods took almost the exact amount of time (~7-8s) to finish -- 2 seconds before hearing the capture sound, 3 seconds before terminal informs me image has been saved to camera SD card (not RAM), and then another 2-3 seconds before the process ends. I'll assume that this is the best that a Raspberry Pi can do, but I still wonder if there are other sources of delay... 

Thanks for your support.

msmeissn, 2019-02-16:
sorry that it did not get better. thanks for the feedback!

closed 2019-02-16 by msmeissn

v-balbino, 2023-01-16:
> Hi Markus,
>
> Recently I have tried other methods:
>
> 1. Using --capture-image directly from terminal
> 2. Entering shell mode, and then use capture-image
> 3. Following the suggestion in [Issue #70 of [...]

How did you modify the gphoto2 library?