gphoto/libgphoto2 #368: Not all camera properties have names with a Sony Alpha-A7r II

abrock, 2019-02-05:
**Describe the bug**

The mapping from camera property id to human readable name is incomplete.

For example my Camera has the option to restrict sensor readout to a part of the sensor which has APC-size, the option is called "APS-C/Super 35mm" and I used sample-autodetect to find that the property ID is 0xd219:(readwrite) (type=0x2) Enumeration [2,1],
a value of "2" means "On", a value of "1" means either "Off" or "Auto". With "Auto" the camera behaves the same as with "Off", at least for stills.

I changed the lines 883 and 2102 in camlibs/ptp2/cameras/sony-a7r.txt to "Label: APS-C/Super 35mm" and ran make clean; make -j8; sudo make install but the sample-autodetect still prints "Property 0xd219", do I have to change more files?

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs

Sony Alpha-A7r II (Control)

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version

Current git versions:
gphoto2 2.5.20.1

Copyright (c) 2000-2019 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.20.1       gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

**To Reproduce**
Steps to reproduce the behavior:

Connect an A7RII and run examples/sample-autodetect.

msmeissn, 2019-02-06:
you need to change the c code, you only adjusted debug output ;)

I added a config variable called "sensorcrop" ... your name was a bit too complicated

abrock, 2019-02-06:
I pulled your changes and built libgphoto and gphoto but the property still has no name:
 
<pre>
./sample-autodetect 
Summary:
Manufacturer: Sony Corporation
Model: ILCE-7RM2
  Version: 4.0
  Serial Number: 00000000000000003282803003803630
Vendor Extension ID: 0x11 (1.0)
Vendor Extension Description: Sony PTP Extensions

Capture Formats: 
Display Formats: JPEG, Unknown(b301), ARW

Device Capabilities:
	File Download, No File Deletion, No File Upload
	No Image Capture, No Open Capture, Sony Capture

Storage Devices Summary:

Device Property Summary:
Compression Setting(0x5004):(readwrite) (type=0x2) Enumeration [2,3,4,16,19] value: 16
White Balance(0x5005):(readwrite) (type=0x4) Enumeration [2,4,32785,32784,6,32769,32770,32771,32772,7,32816,32786,32800,32801,32802] value: Automatic (2)
F-Number(0x5007):(readwrite) (type=0x4) Range [0 - 65535, step 1] value: f/14 (1400)
Focus Mode(0x500a):(readwrite) (type=0x4) Enumeration [1,2,32772,32773,32774] value: Manual Focus (1)
Exposure Metering Mode(0x500b):(readwrite) (type=0x4) Enumeration [4,1,32770] value: Average (1)
Flash Mode(0x500c):(readwrite) (type=0x4) Enumeration [2,1,4,3,5,32769,32771,32772,32817,32818] value: Undefined (0)
Exposure Program Mode(0x500e):(readwrite) (type=0x4) Enumeration [32768,32769,2,3,4,1,32848,32849,32850,32851,32852,32896,32897,32898,32899,32833,7,32785,32789,32788,32786,32787,32790,32791,32792] value: M (1)
Exposure Bias Compensation(0x5010):(readwrite) (type=0x3) Enumeration [0,1,2,5000,4700,4500,4300,4000,3700,3500,3300,3000,2700,2500,2300,2000,1700,1500,1300,1000,700,500,300,-300,-500,-700,-1000,-1300,-1500,-1700,-2000,-2300,-2500,-2700,-3000,-3300,-3500,-3700,-4000,-4300,-4500,-4700,-5000] value: 0.0 stops (0)
Still Capture Mode(0x5013):(readwrite) (type=0x4) Enumeration [1,32773,32771,32772] value: Single Shot (1)
DOC Compensation(0xd200):(readwrite) (type=0x3) Enumeration [0,1,2,3000,2700,2500,2300,2000,1700,1500,1300,1000,700,500,300,-300,-500,-700,-1000,-1300,-1500,-1700,-2000,-2300,-2500,-2700,-3000] value: 0
DRangeOptimize(0xd201):(readwrite) (type=0x2) Enumeration [1,31,17,18,19,20,21] value: 31
Image size(0xd203):(readwrite) (type=0x2) Enumeration [4,14,15,5,6,7,16,17,18,8,19,9,10,11,20,21,12,23,24,13,1,2,3] value: 1
Shutter speed(0xd20d):(readwrite) (type=0x6) Range [0 - 4294967295, step 1] value: 262154
Property 0xd20e:(readwrite) (type=0x2) Enumeration [1,2,3,11,8,9,10,4,12,5,13,6,14,7,15] value: 15
Color temperature(0xd20f):(readwrite) (type=0x4) Range [2500 - 9900, step 100] value: 0
Aspect Ratio(0xd211):(readwrite) (type=0x2) Enumeration [1,2] value: 1
Focus status(0xd213):(readwrite) (type=0x2) Enumeration [1,2,3,5,6,7] value: 1
ISO(0xd21e):(readwrite) (type=0x6) Enumeration [16777215,25,50,64,80,100,125,160,200,250,320,400,500,640,800,1000,1250,1600,2000,2500,3200,4000,5000,6400,8000,10000,12800,16000,20000,25600,32000,40000,51200,64000,80000,102400,128000,160000,204800,256000,320000,409600,33554431,16777241,16777266,16777280,16777296,16777316,16777341,16777376,16777416,16777466,16777536,16777616,16777716,16777856,16778016,16778216,16778466,16778816,16779216,16779716,16780416,16781216,16782216,16783616,16785216,16787216,16790016,16793216,16802816,16828416,16879616,16982016,17186816,50331647,33554532,33554632,33554832,33555232,33556032,33557632,33560832,33567232,33580032,33605632] value: 100
Picture Effect(0xd21b):(readwrite) (type=0x4) Enumeration [32768] value: 32768
Property 0xd21d:(readwrite) (type=0x2) Range [0 - 2, step 1] value: 0
Property 0xd21f:(readwrite) (type=0x2) Enumeration [2,1] value: 1
Property 0xd217:(readwrite) (type=0x2) Enumeration [2,1] value: 1
Battery Level(0xd218):(readwrite) (type=0x1) Range [-1 - 100, step 1] value: 100
Property 0xd219:(readwrite) (type=0x2) Enumeration [2,1] value: 1
Property 0xd212:(readwrite) (type=0x2) Range [0 - 15, step 1] value: 0
CC Filter(0xd210):(readwrite) (type=0x2) Range [164 - 220, step 1] value: 192
AB Filter(0xd21c):(readwrite) (type=0x2) Range [164 - 220, step 2] value: 192
Property 0xd221:(readwrite) (type=0x2) Enumeration [0,1,2] value: 1
Property 0xd214:(readwrite) (type=0x6) Range [0 - 4294967295, step 1] value: 57464832
Objects in memory(0xd215):(readwrite) (type=0x4) Range [0 - 65535, step 1] value: 0
Property 0xd220: error 2002 on query.
Property 0xd2c1:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Property 0xd2c2:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Property 0xd2c3:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Property 0xd2c9:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Movie(0xd2c8):(readwrite) (type=0x4) Enumeration [1,2] value: 1
Property 0xd2c5:(readwrite) (type=0x4) Enumeration [1,2] value: 1
Still Image(0xd2c7):(readwrite) (type=0x4) Enumeration [1,2] value: 1

lookup widget failed: -2
</pre>

And Gphoto2:

<pre>
gphoto2 --list-all-config --port=usb:001,018
/main/actions/autofocus                                                        
Label: Auto-Focus
[... 1128 lines cut ...]
/main/other/d2c8
Label: Movie
Readonly: 0
Type: MENU
Current: 1
Choice: 0 1
Choice: 1 2
END
/main/other/d2c5
Label: PTP Property 0xd2c5
Readonly: 0
Type: MENU
Current: 1
Choice: 0 1
Choice: 1 2
END
/main/other/d2c7
Label: Still Image
Readonly: 0
Type: MENU
Current: 1
Choice: 0 1
Choice: 1 2
END
</pre>

msmeissn, 2019-02-07:
I added the name decoding to git now, and if you look close in the --list-all-config you will see "sensorcrop" having appeared