gphoto/libgphoto2 #373: GPhoto2Error: [-7] I/O problem

uvedhe, 2019-02-11:
**Describe the bug**

I'm trying to get my Sony A850 to work with `libgphoto2`, through Python 3.7.1 using `python-gphoto2`. I can initiate the camera, get the summary, trigger a photo, but transferring said photo results in `gphoto2.GPhoto2Error: [-7] I/O problem` and the camera hanging (I need to unplug and plug back in for it to respond again).

**Name the camera**
Sony A850

**libgphoto2 and gphoto2 version**
libgphoto2 v2.5.20
gphoto2 v1.9.0

**To Reproduce**
````
# Python 3.7.1
import gphoto2 as gp
gp.error_severity[gp.GP_ERROR] = logging.WARNING
context = gp.Context()
camera = gp.Camera()

while True:
    error = gp.gp_camera_init(camera, context)
    if error >= gp.GP_OK:
        print('Camera connected')  # returns 'Camera connected'
        break
    if error != gp.GP_ERROR_MODEL_NOT_FOUND:
        raise gp.GPhoto2Error(error)
    # no camera, try again in 2 seconds
    time.sleep(2)

# Here's the problem:
print('Capturing image')  # Prints correctly
file_path = gp.check_result(gp.gp_camera_capture(camera, gp.GP_CAPTURE_IMAGE))  # Triggers camera but then hangs
print('Camera file path: {0}/{1}'.format(file_path.folder, file_path.name))  # Is never printed

# This is never executed (script errors above)
target = os.path.join('/tmp', file_path.name)
print('Copying image to ', target)
camera_file = gp.gp_camera_file_get(camera, file_path.folder, file_path.name, gp.GP_FILE_TYPE_NORMAL)
gp.gp_file_save(camera_file, target)
gp.GP_CAPTURE_IMAGE(camera)

Sija, 2019-02-11:
Try updating to the latest libgphoto2 library.

uvedhe, 2019-02-11:
> Try updating to the latest libgphoto2 library version.

Thanks, unfortunately doesn't solve the issue..

uvedhe, 2019-02-12:
FWIW: when I run the command `gphoto2 --capture-image` directly from Terminal, the camera is triggered successfully, but in Terminal I get the following message after ~5 seconds:

```
ERROR: Could not capture image.                                                
ERROR: Could not capture.
```

Adding `--debug` produces the following output:

```
0.000024 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000063 main                        (2): gphoto2 2.5.20
0.000070 main                        (2): gphoto2 has been compiled with the following options:
0.000075 main                        (2):  + clang (C compiler used)
0.000080 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000085 main                        (2):  + no exif (for displaying EXIF information)
0.000090 main                        (2):  + no cdk (for accessing configuration options)
0.000094 main                        (2):  + no aa (for displaying live previews)
0.000100 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000105 main                        (2):  + readline (for easy navigation in the shell)
0.000116 main                        (2): libgphoto2 2.5.22
0.000122 main                        (2): libgphoto2 has been compiled with the following options:
0.000126 main                        (2):  + all camlibs
0.000131 main                        (2):  + clang (C compiler used)
0.000136 main                        (2):  + ltdl (for portable loading of camlibs)
0.000141 main                        (2):  + no EXIF (for special handling of EXIF files)
0.000146 main                        (2): libgphoto2_port 0.12.0
0.000151 main                        (2): libgphoto2_port has been compiled with the following options:
0.000156 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.000160 main                        (2):  + clang (C compiler used)
0.000165 main                        (2):  + ltdl (for portable loading of iolibs)
0.000169 main                        (2):  + USB (libusb1, for USB cameras)
0.000174 main                        (2):  + serial (for serial cameras)
0.000178 main                        (2):  + no resmgr (serial port access and locking)
0.000183 main                        (2):  + no ttylock (serial port locking)
0.000187 main                        (2):  + no lockdev (serial port locking)
0.000192 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000197 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000202 main                        (2): invoked with following arguments:
0.000207 main                        (2):   --capture-image
0.000212 main                        (2):   --debug
0.000252 load_settings               (2): Creating gphoto config directory ('/Users/rvx/.gphoto')
0.000362 load_settings               (2): Loading settings from file '/Users/rvx/.gphoto/settings'.
0.000457 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.000479 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0'...
0.000693 foreach_func                (2): Called for filename '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/disk'.
0.001142 foreach_func                (2): Loaded '' ('^disk:') from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/disk'.
0.001152 foreach_func                (2): Called for filename '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/ptpip'.
0.001519 foreach_func                (2): Loaded 'PTP/IP Connection' ('ptpip:') from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/ptpip'.
0.001529 foreach_func                (2): Loaded '' ('^ptpip:') from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/ptpip'.
0.001535 foreach_func                (2): Called for filename '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/serial'.
0.006407 foreach_func                (2): Loaded 'Serial Port Device' ('serial:') from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/serial'.
0.006418 foreach_func                (2): Loaded '' ('^serial:') from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/serial'.
0.006423 foreach_func                (2): Called for filename '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/usb1'.
0.046898 foreach_func                (2): Loaded '' ('^usb:') from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/usb1'.
0.046931 foreach_func                (2): Loaded 'Universal Serial Bus' ('usb:020,017') from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/usb1'.
0.046938 foreach_func                (2): Loaded 'Universal Serial Bus' ('usb:026,003') from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/usb1'.
0.046944 foreach_func                (2): Loaded 'Universal Serial Bus' ('usb:029,007') from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/usb1'.
0.046950 foreach_func                (2): Called for filename '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/usbdiskdirect'.
[... cut ...]
40.524230 gp_filesystem_reset         (2): resetting filesystem
40.524243 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
40.524249 gp_filesystem_lru_clear     (2): fscache LRU list already empty
40.524254 delete_all_folders          (2): Internally deleting all folders from '/'...
40.524259 lookup_folder               (2): Lookup folder '/'...
40.524265 lookup_folder               (2): Found! / is 0x7fe7efc031d0
40.524271 recurse_delete_folder       (2): Recurse delete folder 0x7fe7efc031d0//

msmeissn, 2019-02-12:
This older Sonys dont have the 9209 opcode, which we emitted unconditionally. 

In current GIT I added some protection for this, it might work better ... can you try current libgphoto2 GIT?

uvedhe, 2019-02-12:
Thanks, just having a n00b moment here: after `git clone https://github.com/gphoto/libgphoto2.git` how do I ensure that `gphoto2` uses the new GIT?

uvedhe, 2019-02-15:
FWIW, here the `--debug` info using the new GIT `2.5.22.1` (I think here it actually shows that it is NOT using the new GIT `2.5.22.1`?)

```
0.000055 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000087 main                        (2): gphoto2 2.5.20
0.000093 main                        (2): gphoto2 has been compiled with the following options:
0.000098 main                        (2):  + clang (C compiler used)
0.000102 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000106 main                        (2):  + no exif (for displaying EXIF information)
0.000109 main                        (2):  + no cdk (for accessing configuration options)
0.000113 main                        (2):  + no aa (for displaying live previews)
0.000118 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000122 main                        (2):  + readline (for easy navigation in the shell)
0.000132 main                        (2): libgphoto2 2.5.22
0.000137 main                        (2): libgphoto2 has been compiled with the following options:
0.000141 main                        (2):  + all camlibs
0.000145 main                        (2):  + clang (C compiler used)
0.000149 main                        (2):  + ltdl (for portable loading of camlibs)
0.000153 main                        (2):  + no EXIF (for special handling of EXIF files)
0.000157 main                        (2): libgphoto2_port 0.12.0
0.000161 main                        (2): libgphoto2_port has been compiled with the following options:
0.000165 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.000169 main                        (2):  + clang (C compiler used)
0.000173 main                        (2):  + ltdl (for portable loading of iolibs)
0.000176 main                        (2):  + USB (libusb1, for USB cameras)
0.000180 main                        (2):  + serial (for serial cameras)
0.000184 main                        (2):  + no resmgr (serial port access and locking)
0.000188 main                        (2):  + no ttylock (serial port locking)
0.000191 main                        (2):  + no lockdev (serial port locking)
0.000195 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000200 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000204 main                        (2): invoked with following arguments:
0.000208 main                        (2):   --capture-image
0.000212 main                        (2):   --debug
0.000248 load_settings               (2): Creating gphoto config directory ('/Users/rvx/.gphoto')
0.000633 load_settings               (2): Loading settings from file '/Users/rvx/.gphoto/settings'.
0.000735 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.000757 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0'...
0.000954 foreach_func                (2): Called for filename '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/disk'.
0.002037 foreach_func                (2): Loaded '' ('^disk:') from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/disk'.
0.002052 foreach_func                (2): Called for filename '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/ptpip'.
0.002892 foreach_func                (2): Loaded 'PTP/IP Connection' ('ptpip:') from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/ptpip'.
0.002902 foreach_func                (2): Loaded '' ('^ptpip:') from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/ptpip'.
0.002907 foreach_func                (2): Called for filename '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/serial'.
0.008391 foreach_func                (2): Loaded 'Serial Port Device' ('serial:') from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/serial'.
0.008403 foreach_func                (2): Loaded '' ('^serial:') from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/serial'.
0.008408 foreach_func                (2): Called for filename '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/usb1'.
0.052593 foreach_func                (2): Loaded '' ('^usb:') from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/usb1'.
0.052643 foreach_func                (2): Loaded 'Universal Serial Bus' ('usb:020,025') from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/usb1'.
0.052658 foreach_func                (2): Loaded 'Universal Serial Bus' ('usb:026,003') from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/usb1'.
0.052671 foreach_func                (2): Loaded 'Universal Serial Bus' ('usb:029,007') from '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/usb1'.
0.052683 foreach_func                (2): Called for filename '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/usbdiskdirect'.
0.054069 foreach_func                (2): Called for filename '/usr/local/Cellar/libgphoto2/2.5.22/lib/libgphoto2_port/0.12.0/usbscsi'.
0.054986 gp_port_info_list_count     (2): Counting entries (9 available)...
0.055000 gp_port_info_list_count     (2): 5 regular entries available.
[... cut ...]
40.575323 gp_filesystem_reset         (2): resetting filesystem
40.575332 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
40.575336 gp_filesystem_lru_clear     (2): fscache LRU list already empty
40.575340 delete_all_folders          (2): Internally deleting all folders from '/'...
40.575344 lookup_folder               (2): Lookup folder '/'...
40.575348 lookup_folder               (2): Found! / is 0x7f9e35c031d0
40.575352 recurse_delete_folder       (2): Recurse delete folder 0x7f9e35c031d0//

msmeissn, 2019-02-15:
Good that you got it working ... 

Now does capture work ;)

uvedhe, 2019-02-15:
Like a charm! :)

msmeissn, 2019-02-16:
can you also check if
gphoto2 --capture-tethered

works (press the shutter button)?

uvedhe, 2019-02-24:
That one doesn't seem to work (sorry for the delay): 
[debug_tethered.txt](https://github.com/gphoto/libgphoto2/files/2898485/debug.tethered.txt)

Basically what happens is that it asks to push the button (`Waiting for events from camera. Press Ctrl-C to abort.`), and when I do the camera takes a photo, but Terminal shows:

```
UNKNOWN PTP Property d20d changed
UNKNOWN PTP Event c201, Param1 ffffc001
```

And gets stuck...