gphoto/libgphoto2 #375: Please support file deletion on Sony ILCE-5000

wferi, 2019-02-15:
`$ gphoto2 --summary` reports:
~~~
Camera summary:                                                                
Manufacturer: Sony Corporation
Model: ILCE-5000
  Version: 1.0
  Serial Number: 00000000000000003282653003868103
Vendor Extension ID: 0x11 (1.0)

Capture Formats: 
Display Formats: Association/Directory, MPEG, JPEG, Unknown(b982), Unknown(b301), Undefined Type, ARW, Unknown(b310)

Device Capabilities:
	File Download, No File Deletion, No File Upload
	No Image Capture, No Open Capture, No vendor specific capture

Storage Devices Summary:
store_00010001:
	StorageDescription: Storage Media
	VolumeLabel: None
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Generic Hierarchical
	Access Capability: Read-Only
	Maximum Capability: 15811477504 (15079 MB)
	Free Space (Bytes): 7838924800 (7475 MB)
	Free Space (Images): -1

Device Property Summary:
Battery Level(0x5001):(read only) (type=0x2) Enumeration [0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,100] value: 100% (100)
Property 0xd402:(read only) (type=0x0) Undefined
Property 0xd406:(readwrite) (type=0x0) Undefined
Property 0xd407:(read only) (type=0x6) 1
Property 0xd303:(read only) (type=0x2) 1
~~~
And trying to delete a file from the storage (SD card) indeed reports `*** Error (-6: 'Unsupported operation') ***`.  This hinders the GVFS operations too, of course.  Removing pictures using the camera buttons after copying them to the computer is not only inconvenient but also risky.  Is adding read-write support for this model feasible?  Is there anything I could help with?

msmeissn, 2019-02-16:
It seems delete is not supported here. Neither the Delete command is avialable, nor does the "store" report deletability (it is set read-only)

Not sure if was set on camera, or if the SD Card was marked readonly perhaps.

wferi, 2019-02-16:
Taking and deleting pictures via the camera controls is possible, so the SD card is writable.
I'm also pretty sure MTP write access can't be disabled in the camera menus.
I thought delete over MTP is such a fundamental function that it must be supported by the camera, but it may not be the case after all...