gphoto/libgphoto2 #378: Bulb mode issues with my Pentax K-70 and Olympus OM-D E-M5 Mark ii

Phattire72, 2019-02-23:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Phattire72, 2019-02-23:
I'm using gphoto2 in Ekos/Stellarmate.  I have a Pentax K-70 and an Olympus OM-D E-M5 Mark ii. Neither work correctly with Bulb mode correctly.  They both take a short exposure, then the image is uploaded at the end of the programmed shutter time in Ekos.  I think both would work if there was a way to use the what is called Live Time mode in Olympus or Bulb2 Mode for Pentax. Both work the same way, they just have a different name.  The mode works by initiating the long exposure with a short shutter release, then a second short shutter release is sent to stop the exposure.  So, if ghoto/Ekos could have a bulb2 mode added, then would allow the cameras that have bulb mode issues to work correctly if they have a similar mode.  I realize the mode would most likely need to be manually enable on the cameras, but it would accomplish the job.

Is there any way to add a Live Time/Bulb2 option in the Gphoto camera interface for gphoto/INDI/Ekos?

There seems to be a number of cameras that have Bulb Mode issues and this could possibly solve the problem with these cameras, at least with Pentax and Olympus Cameras, but possibly others if they have a similar mode.

Thank you for looking into this.

berylius, 2019-03-15:
Hi, 
I have started to make some modifications to manage the Bulb mode for Olympus.
I created a "buld widget" the olympus, now i need to manage the download of the image in library.c::camera_wait_for_event.
I'll look for the Pentax after that.