gphoto/libgphoto2 #379: Can't set zoom on Canon PowerShot G9 X Mark II

luffecn, 2019-02-26:
Hi msmeissn,

**Describe the bug**
The camera seems crashed while zooming. 

**Name the camera**
Canon PowerShot G9 X Mark II

**libgphoto2 and gphoto2 version**
gphoto2         2.5.20.1       gcc, popt(m), no exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.22.1       all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

**To Reproduce**
gphoto2 --set-config zoom=60
[zoom.log](https://github.com/gphoto/libgphoto2/files/2905203/zoom.log)

msmeissn, 2019-03-01:
unrelated, does --capture-image and --capture-preview work?

msmeissn, 2019-03-01:
Hmm. setting it works... is the camera not usable? Does it move the lens in some form at all?

luffecn, 2019-03-04:
Both  --capture-image and  --capture-preview work.
The zoom command's reply is ok, but the lens does not move.

angraca, 2020-08-14:
I have the same problem with a Canon SX720 HS. Everything else works fine but as soon I set to zoom the program hangs for a while. When the program becomes responsive again the connection to the camera is lost and I have to restart it.

erco77, 2021-12-07:
Same problem setting zoom with a Canon G5X + gphoto2 2.5.27.
Attaching [gphoto-set-zoom-hangs--canon-g5x.txt](https://github.com/gphoto/libgphoto2/files/7665623/gphoto-set-zoom-hangs--canon-g5x.txt) for the hung command `gphoto2 --set-config /main/capturesettings/zoom=60`

Before setting the zoom I can do image capture and download to the computer, --summary, --get-config, etc. In the case of zoom, getting the config shows:
```
gphoto2 --get-config /main/capturesettings/zoom
Label: Zoom                                                                    
Readonly: 0
Type: RADIO
Current: 0
Choice: 0 101
END
```
..which I take to mean any integer value from 0 to 101 should be allowed, but if I run e.g. `gphoto2 --set-config /main/capturesettings/zoom=60` it hangs for about 45 seconds, returns no messages and an exit code of 0. But from then on, gphoto2 can no longer communicate with the camera; all commands throw a `PTP Timeout` error.
 
To get things working again, I have to unplug/replug the USB cable for the camera, then it works again.

erco77, 2021-12-07:
One more data point: I switched to a different camera, an old Canon S3 IS, and that seems to work fine w/gphoto2; capture, get/set zoom.. no hangs.
Both cameras in Manual mode during these tests.